dresden-elektronik/deconz-rest-plugin #4494: Add coolsetpoint to Bitron 902010/32 thermostat

solvisit, 2021-02-28:
## Feature request type

Add to REST API config the coolsetpoint for Bitron 902010/32 thermostat and automatically set "Control Sequence of Operation" to "Cooling and Heating 4-pipes" when mode switch to "cool".

## Description

It is possible to add the same coolingsetpoint to Bitron 902010/32? 

It has the ablity to switch mode from "heat" to "cool". The device change to "cool" but it is not possible to set the coolsetpoint.

Besides, for enabling both modes it is necessary to set 0x001B (Control Sequence of Operation) of thermostat cluster to 0x04 (Cooling and Heating 4-pipes) before changing 0x001C (System Mode) to 0x02 (Cool). It is possible to set them to the correct values if the mode is changed to "cool"?

## Considered alternatives

No alternatives found

## Additional context

[image]
[image]

Mimiix, 2021-03-01:
@SwoopX I think you did this one?

SwoopX, 2021-03-01:
No, but regardless. Exposing the cooling setpoint is easy, fiddling the other thing in not so much and I'm a bit hesitant since it's only required once.

Maybe there's some place where we got that one already in...

solvisit, 2021-03-06:
> Exposing the cooling setpoint is easy

This would be ok for me. I can set the "Control Sequence of Operation" manually at installation time.

solvisit, 2021-03-14:
Any news about exposing the cooling setpoint? As I said the other thing is not so important. Thank you

solvisit, 2021-03-19:
@SwoopX Any news? Thx

closed 2021-04-22 by manup