dresden-elektronik/deconz-rest-plugin #4497: Full /var/log/messages of ConBee II

S474N, 2021-03-01:
## Describe the question or issue you are having

Have full /var/log/messages of these:

```
Mar  1 11:03:54 raspberrypi kernel: [   65.887563] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
Mar  1 11:03:54 raspberrypi mtp-probe: checking bus 1, device 7: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:03:54 raspberrypi mtp-probe: bus: 1, device: 7 was not an MTP device
Mar  1 11:03:54 raspberrypi mtp-probe: checking bus 1, device 7: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:03:54 raspberrypi mtp-probe: bus: 1, device: 7 was not an MTP device
Mar  1 11:04:24 raspberrypi kernel: [   95.659580] usb 1-1.3: USB disconnect, device number 7
Mar  1 11:04:24 raspberrypi kernel: [   95.954065] usb 1-1.3: new full-speed USB device number 8 using xhci_hcd
Mar  1 11:04:24 raspberrypi kernel: [   96.092707] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Mar  1 11:04:24 raspberrypi kernel: [   96.092717] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar  1 11:04:24 raspberrypi kernel: [   96.092724] usb 1-1.3: Product: ConBee II
Mar  1 11:04:24 raspberrypi kernel: [   96.092731] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Mar  1 11:04:24 raspberrypi kernel: [   96.092738] usb 1-1.3: SerialNumber: DE269666634
Mar  1 11:04:24 raspberrypi kernel: [   96.095403] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
Mar  1 11:04:24 raspberrypi mtp-probe: checking bus 1, device 8: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:04:24 raspberrypi mtp-probe: bus: 1, device: 8 was not an MTP device
Mar  1 11:04:24 raspberrypi mtp-probe: checking bus 1, device 8: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:04:24 raspberrypi mtp-probe: bus: 1, device: 8 was not an MTP device
Mar  1 11:04:28 raspberrypi kernel: [   99.500448] usb 1-1.3: USB disconnect, device number 8
Mar  1 11:04:28 raspberrypi kernel: [   99.794131] usb 1-1.3: new full-speed USB device number 9 using xhci_hcd
Mar  1 11:04:28 raspberrypi kernel: [   99.933221] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Mar  1 11:04:28 raspberrypi kernel: [   99.933245] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar  1 11:04:28 raspberrypi kernel: [   99.933263] usb 1-1.3: Product: ConBee II
Mar  1 11:04:28 raspberrypi kernel: [   99.933280] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Mar  1 11:04:28 raspberrypi kernel: [   99.933297] usb 1-1.3: SerialNumber: DE269666634
Mar  1 11:04:28 raspberrypi kernel: [   99.938147] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
Mar  1 11:04:28 raspberrypi mtp-probe: checking bus 1, device 9: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:04:28 raspberrypi mtp-probe: bus: 1, device: 9 was not an MTP device
Mar  1 11:04:28 raspberrypi mtp-probe: checking bus 1, device 9: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:04:28 raspberrypi mtp-probe: bus: 1, device: 9 was not an MTP device
Mar  1 11:04:54 raspberrypi kernel: [  125.872909] usb 1-1.3: USB disconnect, device number 9
Mar  1 11:04:54 raspberrypi kernel: [  126.194168] usb 1-1.3: new full-speed USB device number 10 using xhci_hcd
Mar  1 11:04:55 raspberrypi kernel: [  126.332943] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Mar  1 11:04:55 raspberrypi kernel: [  126.332953] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar  1 11:04:55 raspberrypi kernel: [  126.332960] usb 1-1.3: Product: ConBee II
Mar  1 11:04:55 raspberrypi kernel: [  126.332968] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Mar  1 11:04:55 raspberrypi kernel: [  126.332975] usb 1-1.3: SerialNumber: DE269666634
Mar  1 11:04:55 raspberrypi kernel: [  126.335635] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
Mar  1 11:04:55 raspberrypi mtp-probe: checking bus 1, device 10: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:04:55 raspberrypi mtp-probe: bus: 1, device: 10 was not an MTP device
Mar  1 11:04:55 raspberrypi mtp-probe: checking bus 1, device 10: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:04:55 raspberrypi mtp-probe: bus: 1, device: 10 was not an MTP device
Mar  1 11:04:58 raspberrypi kernel: [  129.713242] usb 1-1.3: USB disconnect, device number 10
Mar  1 11:04:58 raspberrypi kernel: [  130.014216] usb 1-1.3: new full-speed USB device number 11 using xhci_hcd
Mar  1 11:04:58 raspberrypi kernel: [  130.153205] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Mar  1 11:04:58 raspberrypi kernel: [  130.153228] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar  1 11:04:58 raspberrypi kernel: [  130.153246] usb 1-1.3: Product: ConBee II
Mar  1 11:04:58 raspberrypi kernel: [  130.153264] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Mar  1 11:04:58 raspberrypi kernel: [  130.153281] usb 1-1.3: SerialNumber: DE269666634
Mar  1 11:04:58 raspberrypi kernel: [  130.158121] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
Mar  1 11:04:58 raspberrypi mtp-probe: checking bus 1, device 11: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:04:58 raspberrypi mtp-probe: bus: 1, device: 11 was not an MTP device
Mar  1 11:04:58 raspberrypi mtp-probe: checking bus 1, device 11: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:04:58 raspberrypi mtp-probe: bus: 1, device: 11 was not an MTP device
Mar  1 11:05:29 raspberrypi kernel: [  160.950291] usb 1-1.3: USB disconnect, device number 11
Mar  1 11:05:30 raspberrypi kernel: [  161.255399] usb 1-1.3: new full-speed USB device number 12 using xhci_hcd
Mar  1 11:05:30 raspberrypi kernel: [  161.394473] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Mar  1 11:05:30 raspberrypi kernel: [  161.394497] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar  1 11:05:30 raspberrypi kernel: [  161.394515] usb 1-1.3: Product: ConBee II
Mar  1 11:05:30 raspberrypi kernel: [  161.394533] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Mar  1 11:05:30 raspberrypi kernel: [  161.394551] usb 1-1.3: SerialNumber: DE269666634
Mar  1 11:05:30 raspberrypi kernel: [  161.399390] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
Mar  1 11:05:30 raspberrypi mtp-probe: checking bus 1, device 12: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:05:30 raspberrypi mtp-probe: bus: 1, device: 12 was not an MTP device
Mar  1 11:05:30 raspberrypi mtp-probe: checking bus 1, device 12: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:05:30 raspberrypi mtp-probe: bus: 1, device: 12 was not an MTP device
[... 148 lines cut ...]
Mar  1 11:08:43 raspberrypi kernel: [  355.057978] usb 1-1.3: Product: ConBee II
Mar  1 11:08:43 raspberrypi kernel: [  355.057996] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Mar  1 11:08:43 raspberrypi kernel: [  355.058013] usb 1-1.3: SerialNumber: DE269666634
Mar  1 11:08:43 raspberrypi kernel: [  355.062914] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
Mar  1 11:08:43 raspberrypi mtp-probe: checking bus 1, device 25: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:08:43 raspberrypi mtp-probe: bus: 1, device: 25 was not an MTP device
Mar  1 11:08:43 raspberrypi mtp-probe: checking bus 1, device 25: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar  1 11:08:43 raspberrypi mtp-probe: bus: 1, device: 25 was not an MTP device
```

## Screenshots

## Environment

 - Host system: Raspberry Pi 4B
 - Running method: Raspbian 
 - Firmware version: 26680700
 - deCONZ version: 2.10.00
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? SSD HDD on USB3

## deCONZ Logs

## Additional context

Mimiix, 2021-03-01:
Hi,

Does it work properly?

S474N, 2021-03-01:
@Mimiix so far yes, but the device ID is growing incrementally

Mimiix, 2021-03-01:
Can you provide some logs from deCONZ?

To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

S474N, 2021-03-01:
Tested another USB port of RPi and no annoying messages in logs. I'll watch it.