dresden-elektronik/deconz-rest-plugin #4499: Tuya Zigbee Smart Siren Alarm With temperature and Humidity Sensor

aidbish, 2021-03-01:
Please could i request the following device be supported at your earliest opportunity

- Product name: Tuya Zigbee Smart Siren Alarm With temperature and Humidity Sensor
- 
- Manufacturer:  _TZE200_d0yu2xgi
- Model identifier:  TS0601
- Device type : 
  - Sensor
  - Siren

https://www.aliexpress.com/item/4000813903996.html

## Screenshots
[image]
[image]

  If available add screenshots of other clusters.

  Relevant clusters are: Simple Metering, Electrical Measurement, Power Configuration, Thermostat, etc. You can typically spare Identify, Alarms, Device Temperature, On/Off. Please ensure data has been read prior to taking any screenshots.
-->

Smanar, 2021-03-01:
Already in PR > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4379

Smanar, 2021-03-01:
Done.
do you know how to test the code ?

sudo apt install deconz-dev
```
git clone --branch tuya_siren_clean https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

aidbish, 2021-03-01:
i run deconz in docker, is there a dev tag i can pull?

Smanar, 2021-03-01:
No it s too much complicated on docker, will see on future version.

ikke-zelf, 2021-03-08:
Does not recognise the device for me

Smanar, 2021-03-08:
@ikke-zelf remember you need 2 inclusion without deleting the device.
Wich one deconz version a re you using ?
Can you share the manaufacture name to check the device ?

Mimiix, 2021-03-10:
@aidbish That is supposed to happen. Phoscon needs to add support on their own. The RestAPI should expose it.

Smanar, 2021-03-10:
You are not using third app ?

aidbish, 2021-03-10:
hi smanar, 
not sure i follow, 3rd app?

Smanar, 2021-03-10:
An application that use the deconz API, like home automation/openHAB/Domoticz/FHEM/nodeRed, ect ...

aidbish, 2021-03-10:
Sorry, Home assistant

Smanar, 2021-03-11:
And nothing is visible on HA ?

Probably same issue than here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4549  even it s not the same device.

aidbish, 2021-03-12:
No nothing in HA

Smanar, 2021-03-12:
You don't have linux machine to make test with code modification ?

Smanar, 2021-03-20:
Probably solved by https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4617

nukleuz80, 2021-04-07:
Hi all,

i have same Tuya NEO Sirene Device on Deconz 2.10-4 beta - Phoscon doesn´t show the device - pairing is not successfull:

[image]

What i have to do now?

Thanks

nukleuz80, 2021-04-07:
I´m working with iobroker and the deconz-adapter...

Branko-Wilhelm, 2021-04-07:
i also can't pair this device on 2.10.4

b2un0, 2021-04-07:
I also can’t pair this device on 2.10.4

Mimiix, 2021-04-07:
The hex code indicates it isn't whitelisted in the Rest plugin. Therefore, not added/visible in API.

@Smanar can you confirm?

Smanar, 2021-04-07:
Yep.
I think your manufacture number is 0x1002 ?
This PR correct this issue https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4617 but it was included in v2.10.5.

nukleuz80, 2021-04-11:
Which date will come the new beta?

Smanar, 2021-04-11:
Honestly the schedule is not perfect ^^.
But you have at least 1 beta and 1 stable by month.

kmplngj, 2021-04-17:
@Smanar Should this work in 2.11.0-beta? Thx!

Smanar, 2021-04-17:
Yep https://github.com/dresden-elektronik/deconz-rest-plugin/releases

kmplngj, 2021-04-17:
> Yep https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.11.0-beta

Is that correct that I only have binary sensor, humidity and temperature in HA?
How to use the alarm in HA?

[image]

I tried multiple inclusions without deleting something between.

Many thanks for the help!

Smanar, 2021-04-17:
Can you take a look direclty in the api ?
`http://IP:PORT/api/key/lights`

IP and PORT are the same than in phoscon
KEY is an apikey https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

Or logs just at start, to find something about the lights entries....

clipse2004, 2021-04-18:
Hi, so with the new beta the Tuya siren is detected as alarm 75 in Deconz, nothing is detected at all in Phoscon. 

I wanted to check the API for you, but the device is not listed there either.
If I look in the Deconz-Gui then it looks like this:

Can it be that it is currently not yet possible to control the siren somehow? Do you need anything else so that you can implement this completely?
[image]
many thanks for your help!

Smanar, 2021-04-19:
You need to use third app to control it, or direclty the API, not supported by phoscon yet.

clipse2004, 2021-04-19:
Homebridge-hue didnt work too :/ 
Do you know which third Party App works?

Smanar, 2021-04-19:
Homebridge-hue need to work.
You can try first directly with api to be sure all is working.
`
http://IP:PORT/api/KEY/sensors`
and
`http://IP:PORT/api/KEY/lights`

IP and PORT are the same than for phoscon
KEY is an API key

https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

Try first to see if you can see the device in the API.

Smanar, 2021-04-20:
Ok, so can you try to include it again, and share the logs during the inclusion (with "info" and "info_l2")
Remember never delete the previous entries, just re-include it.

Deconz show only the node, so you have only one, phoscon show the sensor entry, so you need to have 1 lights and + sensors.

Heichele, 2021-05-06:
Hallo, wann kann man damit rechnen das dieses Gerät Supportet wird?

Mimiix, 2021-05-06:
@Heichele Someone needs to test with smanar :) For further posts: please stick to english 😄

clipse2004, 2021-05-06:
I would, however my Deconz is now running over docker. I had huge problems with the beta and my HUE motion sensors.
Now I'm back on stable :/ 
And there the Tuya is recognized as no sensor at all.

Heichele, 2021-05-06:
@Mimiix Okay

Smanar, 2021-05-06:
Oup, have forget.
Have just made the PR for the removed part https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4844

But it have no impact on the device detection.

Heichele, 2021-05-06:
@Smanar can I help with testing?

Smanar, 2021-05-06:
Sure, if you are using a recent deconz version, and have a device not reconised, I will be interested by your logs (with "info" and "info_l2") during the inclusion.

Tuya stuff need generaly 2 inclusions, without deleting anything between both.

nukleuz80, 2021-05-06:
FYI: i‘m using latest beta and the ioBroker adapter as third party component. After implementation i get three sensors - Alarm, humidity and temperature but i can‘t trigger them... How to do this?

I read that a light is missing to trigger... but there is nothing.

Thanks

Heichele, 2021-05-06:
@Smanar I'm not very good at these things, would you need some help executing ... "Info" and "info_I2"?

Smanar, 2021-05-07:
@Heichele depend if you have headless or GUI deconz version ?

@nukleuz80 have found the problem, are you able to compile code to make test ?

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_siren_7 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Smanar, 2021-05-09:
Ok so the PR is already validated, you will be able to test it on the next beta version, tell me if it's good with the correction, to close this PR.

nukleuz80, 2021-05-10:
Ok, i will test in next beta.

A question to the dev-version of deConz: 
Is it possible to run it on the same system as the beta runs? Then i will try it...

Smanar, 2021-05-10:
Depend of what you are calling the dev-version ? and depend too your system.
PR stay in waiting list, then manup valid some of them, and immediatly after make a beta or stable version with them.

But if you are taking about compiling yoursef the code to test, the procedure just replace a file, you can keep the old one on another folder to make a rollback after your tests.

But all works fine if you have a full OS, more complicated on docker.

nukleuz80, 2021-05-10:
I mean with dev-Version the link you have posted - and i understand what you wrote and how to do it .

My OS is Raspbian Buster and i will give a try to compile it myself...

Thanks...

Smanar, 2021-05-10:
ha ok, so it s not realy "the" dev Version, but more "my" dev version ^^, nothing official.
But yes, you just need to swap files to swap versions.

nukleuz80, 2021-05-11:
Ok, i compiled it and now i get three sensors + one light. In Phoscon i see the light but cant switch or toggle it. 
W
How should be the workflow?

Thanks

nukleuz80, 2021-05-11:
And yes, i added it two times without deleting. 😎

Smanar, 2021-05-11:
For alert/lselect it s something standard, all sirens are working same, it s a security, too dangerous use state/on for exemple because of group.

I don't remember all valus but you can try

- "volume" from 0>100
- "melody" from 1>18
- "preset" with "humidity" "temperature" or "both
- some threshold to configure min/max value for  both detections

gio82c, 2021-05-18:
Hi, with the latest release I managed to integrate it into deconz and homeassistant, I also managed to change melody with REST API, the only problem is that when I activate the switch after a few seconds it turns off automatically, there is an option to increase this thing?

Smanar, 2021-05-18:
There is a setting duration on the hardware but not used on the API.
I can add it if you are able to test the code ?

gio82c, 2021-05-18:
Excuse me but the method to test the code is still not very clear to me, although I would like to understand more ... anyway if with HA Supervised I can test the code would I like it or does it take a particular installation to do it?

Smanar, 2021-05-18:
> does it take a particular installation to do it

If you are using HA, I think yes. The easier way is using a linux machine with full OS, can work with virtual machine, but never succed on my side with them.

gio82c, 2021-05-18:
ok thanks, is there any guide so i can read it to understand something more?

Smanar, 2021-05-18:
Yep here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

But it s the global description, if you have a linux machine, I will give you a link to my fork with modifications, and I will update the command to use.

If you can make the procedure on the first link on your machine, I will give you the command to test a new code.

gio82c, 2021-05-19:
unfortunately i don't have a machine with linux, i have homeassistant OS on rasperry, it's not on rasbian so i can't test it. Also I realized that the device does not even report the battery, I hope that someone better than me will be able to test the code ..

nukleuz80, 2021-06-06:
Hi again, i tested it and if i change the volume or melody nothing changed...?

Smanar, 2021-06-06:
Do you have some logs to share during the command ? (with "info" and "info_l2")

nukleuz80, 2021-06-06:
i will give it to you if you tell how to log the command :)

[47 more replies]

closed 2021-07-10 by github-actions[bot]