dresden-elektronik/deconz-rest-plugin #4503: could not sync with bootloader

SuperSchelli, 2021-03-01:
## Describe the question or issue you are having

## Screenshots

## Environment
 - Host system:  Raspberry Pi 4B+ -- gpio -v shows v2.52
 - Running method: Raspbian 
 - Firmware version: (not connected)
 - deCONZ version: (2.10.01)
 - Device:  RaspBee I 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

SuperSchelli, 2021-03-01:
[image]
[image]
[image]

SuperSchelli, 2021-03-01:
ok, here are the screens, dont know why they failed in my first post

Smanar, 2021-03-02:
Just by curisity, have you tried this method ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2904

SuperSchelli, 2021-03-02:
nope havent tried that, will try older gcfflasher versions, thx for the tip, thought they were downward compatible.

SuperSchelli, 2021-03-04:
hmm, i grabbed the 2.11 gcfflasher deb, the newest one and tried it and bam it flashed without problems, although it also just said GCFFlasher V3_17, same as yesterday, everything in the log looks exactly the same, except this time it did not fail to sync ... well, i will not complain although i do not understand it.
Thx for the help, problem solved

closed 2021-03-04 by SuperSchelli