dresden-elektronik/deconz-rest-plugin #4505: Xiaomi Multisensor no air pressure and humidity

Schattenruf, 2021-03-02:
## Describe the question or issue you are having
 After the last deconz update my Xiaomi Multisensor or Temperature Sensor does not show any air pressure or humidity

## Screenshots

This is how it should look like:
[image]

And how it actually looks:
[image]

Is there anything I can do to enable those values again?

## Environment

 - Host system: Beelink BT3
 - Running method: Debian with iobroker
 - Firmware version: (264A0700)
 - deCONZ version: (2.05.74)
 - Device: ConBee II 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

fribse, 2021-03-04:
I see problems with the xiaomi sensor as well, as described [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4517)

Schattenruf, 2021-03-05:
@fribse but here is the humidity and air pressure visible... For me it is not.

Mimiix, 2021-03-08:
Would be useful if we could sniff the traffic here.

Schattenruf, 2021-03-08:
@Mimiix Is there a manual or anything like this? I could try to get those logs.

Mimiix, 2021-03-30:
Sorry I missed this one. The traffic sniffing is something you need a cc device for or a ConBee /raspbee 1

Schattenruf, 2021-03-31:
As I mentioned I have the conbee Stick. I also have a CC Stick that got replaced by the conbee one

Smanar, 2021-03-31:
And if you try to set phoscon in permit join, and read again basic attribute on deconz ?
Pressing shortly the button every 2/3s after inclusion (just after the 5 + 3 blink) to prevent the device go to sleep ?

Schattenruf, 2021-04-03:
@Smanar I reseted the device and rejoined it to my Network but no change. No Air pressure or humidity. Just temperature. I did not needed to Push the pair Button twice because He found it immediatly. 

The other values just do not get requested after the latest deconz Upgrade.

I do Not know what to do now. I think it is a bug. But I cannot find any logs or how to sniff the Traffic between these two.

Schattenruf, 2021-04-06:
Hey again,
seems it works now again:
[image]

I do not know what has happened here or what caused the issue but I will close this post, if it stays like that for the next 3 days.

closed 2021-04-10 by Schattenruf