dresden-elektronik/deconz-rest-plugin #4507: Issue: Duplicate Websocket Events

bencrazy, 2021-03-02:
## Describe the bug

After upgrading from v2.08.xx to 2.09.03, there appear to be more than one websocket message sent for at least some events. I noticed this in particular with several of my 'Button' devices, causing lights to toggle multiple times due to my light toggle automations being tied to the '1002' (short button press) event of the buttons.

In addition to verifying multiple websocket messages in the deCONZ logs, I also confimed that Home Assistant is receiving multiple same events using the Developer Tools --> Events.

I'm running the virgin Raspberry Pi deCONZ image, but have manually upgraded to v2.09.03 per the instructions for Raspian.

I will add to this bug report, hopefully this evening. I want to roll back to the previous version again and retest to make sure this wasn't also occurring in that version.

## Steps to reproduce the behavior

  1. Manual upgrade of deCONZ for Raspian per instructions at https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

## Expected behavior

Single websocket event per device event

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi 3+
 - Running method: deCONZ image running on Pi 3+ as deCONZ/Phoscon Gateway, Home Assistant with deCONZ Add-on running on separate Pi 4
 - Firmware version: unknown (will edit later)
 - deCONZ version: 2.09.03
 - Device: RaspBee II
 - Do you use an USB extension cable: (n/a)
 - Is there any other USB or serial devices connected to the host system? No, and the Pi 3+ has been isolated from other 2.4 GHz and USB 3.0 devices.

## deCONZ Logs

```
19:48:15:524 discard sensor state push for 2: state/lastupdated (already pushed)
19:48:15:731 enqueue event state/lastupdated for /sensors/11
19:48:15:733 enqueue event state/buttonevent for /sensors/11
19:48:15:735 enqueue event state/lastupdated for /sensors/11
19:48:15:775 Websocket 192.168.0.196:46348 send message: {"e":"changed","id":"11","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-03-02T14:48:15.735"},"t":"event","uniqueid":"28:6d:97:00:01:0a:32:d9-01-0500"} (ret = 165)
19:48:15:781 Node data 0x286d9700010a32d9 profileId: 0x0104, clusterId: 0x0500
19:48:15:786 enqueue event state/lastupdated for /sensors/11
19:48:15:789 	payload: 0200190100
19:48:15:791 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
19:48:15:794 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
19:48:15:895 discard sensor state push for 11: state/buttonevent (already pushed)
19:48:15:900 poll node 00:12:4b:00:22:6a:76:05-01
19:48:15:903 Poll light node Router - North Lean-to
19:48:15:954 Websocket 192.168.0.196:46348 send message: {"e":"changed","id":"11","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-03-02T14:48:15.786"},"t":"event","uniqueid":"28:6d:97:00:01:0a:32:d9-01-0500"} (ret = 165)
19:48:15:961 Poll APS request to 0x00124B00226A7605 cluster: 0x0006 dropped, values are fresh enough
19:48:15:994 discard sensor state push for 11: state/lastupdated (already pushed)
19:48:16:028 Node data 0x286d970001035d34 profileId: 0x0104, clusterId: 0x0B04
19:48:16:032 0x286D970001035D34: update ZCL value 0x01/0x0B04/0x0508 after 0 s
19:48:16:036 enqueue event state/current for /sensors/7
19:48:16:038 enqueue event state/lastupdated for /sensors/7
19:48:16:041 	payload: 0805212900
19:48:16:045 binding for attribute reporting of ep: 0x01 cluster 0x0B04 seems to be active

```

## Additional context

Smanar, 2021-03-02:
You have it only for button sensor ?
There is a similar issue, but I don't see why you haven't this problem on previous version ...

bencrazy, 2021-03-02:
> You have it only for button sensor ?
> There is a similar issue, but I don't see why you haven't this problem on previous version ...

Only that I've noticed so far. I don't have automations set up for any other devices, so perhaps that is why I just haven't noticed...

Smanar, 2021-03-02:
Probably nothing to see with you, but the PR about the issue I m speaking is here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4265

Is more specific to tuya switch.

bencrazy, 2021-03-02:
I also noticed #4272 and I checked time settings on both the Pi and the RaspBee. I did have two different time zones (due to carelessness when I set up the Phoscon Gateway). Fixed that but have no idea if that could have any effect. I don't know if events are 'cached' somewhere. I do know I'm still getting the seemingly random 'ghost' events coming in (the web server is pushing out messages to Home Assistant even though there was no button activity).

SwoopX, 2021-03-03:
This might be related to the `websocketnotifyall` setting, which is set to `true` by default.

Smanar, 2021-03-03:
I m not agree, setting websocketnotifyall to false is a workaround for me, not a solution.
If there is no changes, the notification don't need to be send.

bencrazy, 2021-03-03:
Just to update, I've been running the "old" image (the November 26 image with v2.05.88 installed) since yesterday and I am able to confirm that it does not seem to have this problem.

I (re) recreated my network configuration in the same manner each time I've either loaded an image or upgraded deCONZ, that is by importing my backed up configuration via the Phoscon Gateway. So that *shouldn't* be a factor.

I've tried upgrading to the new 2.09.03 version twice (using the published manual upgrade steps) and have had the same result both times, i.e. multiple websocket messages are sent on button press event, and "random" (maybe periodic, I haven't timed them) ghost events (websocket message sent out even though no button press event).

Smanar, 2021-03-03:
Just by curiosity, you can't try with more recent deconz version ?

manup, 2021-03-03:
Which switch/product is that or is 'Button' device referring to SmartThings Button switch?

bencrazy, 2021-03-03:
> Which switch/product is that or is 'Button' device referring to SmartThings Button switch?

You are correct, my buttons are Smartthings (Samjin), part number GP-U999SJVLEAA.

bencrazy, 2021-03-03:
> Just by curiosity, you can't try with more recent deconz version ?

I t

bencrazy, 2021-03-03:
Sorry guys, accidentally hit the close issue.

Smanar, I'll verify this evening but I was under the impression that 2.09.03 was the latest version (which is the version I'm having issues with)...

Smanar, 2021-03-03:
ok, but in fact there is already an issue for Smartthings (Samjin), and I don't think the previous one was solved too, so can be linked.

bencrazy, 2021-03-03:
Thanks Smanar. I see some very similar issues doing a quick search for "Samjin". Don't know why I didn't think to search first, was just excited to participate here for the first time, I guess!

v2.05.88 seems stable for now, so I'll continue using this version and will monitor progress on the other related issues. If there's anything I can provide that helps, I'll do the best I can.

Smanar, 2021-03-04:
Np ^^
On the other issue, the device was working on v2.8.0-beta but have issue on v2.9.1-beta https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4232

closed 2021-04-02 by github-actions[bot]