dresden-elektronik/deconz-rest-plugin #4510: Tuya siren(TS0216) doesn't expose options to Home Assistant

aidbish, 2021-03-03:
## Describe the bug

 i have paired a tuya siren but the options to set volume and siren time and other options are not exposed to Home Assistant, in phoscon app it only has the option to turn on/off the siren

Home Assistant only shows the siren to be and on/off device (works) there are no other options

## Steps to reproduce the behavior

i have reset and repaired but again the options are not available. in the deconz app i can see the clusters and the options and able to set/change but these are not available to home assistant

## Expected behavior

Options to be controllable in Home automation software

## Screenshots

  If applicable, add screenshots to help explain the problem.
[image]
[image]

## Environment

Host system: PC
Running method: Marthoc Docker container
Firmware version: 26680700
deCONZ version: 2.09.01
Device: ConBee II
Do you use an USB extension cable: yes
Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

aidbish, 2021-03-03:
I have just noticed that in Home assistant if i select to turn on the device the siren will work but the toggle in HA return to off position
To turn the alarm off, i have to select on then off in the toggle

Mimiix, 2021-03-03:
I think this is intended behavior. Options should be set via the rest api or deconz gui.

Smanar, 2021-03-03:
This device is not yet in deconz, still in PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4379

aidbish, 2021-03-03:
Hi Smanar
This is a different model to that pr
manufacturer name : _TYZB01_a5tf0bpl
modelTS0216

Smanar, 2021-03-03:
Ha yes, right, this device have the tuya cluster 0xEF01 but it seem using only the IAS cluster.

aidbish, 2021-03-25:
Hi smanar

is there anything more you need from me regarding your last comment

Mimiix, 2021-03-26:
@Smanar :)?

Smanar, 2021-03-26:
I think this device work like all siren
- You have predefined mode
- You can change the duration using state.ontime

If it don't work, can you show your device JSON ?

You have the predefined mode here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4432#issuecomment-782841021

closed 2021-04-24 by github-actions[bot]