dresden-elektronik/deconz-rest-plugin #4511: Tuya ZigBee Multi-Sensor 4 in 1 Smart PIR Motion Feuchtigkeit Licht Temperatur Sensor / TS0202

easybeat, 2021-03-03:
## Device

- Product name: Tuya ZigBee Multi-Sensor 4 in 1 Smart PIR Motion Feuchtigkeit Licht Temperatur Sensor
- Manufacturer: _TZ3210_zmy9hjay
- Model identifier: TS0202
- Device type : Please remove all unrelated device types. 
   - Sensor
   - Other: PIR, Temperature, Humidity, Luminance 4in1 sensor
   - Link to [product page](https://de.aliexpress.com/item/1005001878974427.html?spm=a2g0s.9042311.0.0.255f4c4dj8zNLu)

[image]
 
## Screenshots

[image]

[image]

______

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Thank you very much for your help.

Mimiix, 2021-03-03:
@Smanar

Smanar, 2021-03-03:
This device have the tuya cluster, but not sure it use it.

@easybeat do you have a real linux machine to test code modification ?

easybeat, 2021-03-03:
Yes I run deconz on a ubuntu workstation

Smanar, 2021-03-03:
So here you have the procedure to test code https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

```
sudo apt install deconz-dev
git clone --branch multisensor https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

I have just added the device in white list, it use ikea mac adress range ...
There is already a motion sensor with the same model id, so I think your device will react same.

easybeat, 2021-03-04:
Good morning,

thank you for your immediate support. I will test today at lunch time!

easybeat, 2021-03-04:
@Smanar 
Sorry really beginner on this....I get an error message if I try qmake...

[image]

Do I have to install something else before?

easybeat, 2021-03-04:
@Smanar 

found the problem. Did "sudo apt install build-essential" and then it works

easybeat, 2021-03-04:
[image]
[image]

I think this did work!!

easybeat, 2021-03-04:
@Smanar 
I did reset the device and joined it in Phoscon and it was successful.

Do I have to check something else?

Smanar, 2021-03-04:
If you can see it in phoson (and not only deconz), can you just check if all is working ?
Temperature/humidity/motion and luminance ?

It s possible phoscon don't support all device feature, if you can check with third app or directly with the API if needed ?

easybeat, 2021-03-04:
@Smanar 

No I can't see it in Phoscon but did check inside ioBroker with hueExtended Adapter and there all the datapoints are in. A bit strange as it did create 4 devices but that's maybe more a problem from the Adapter.

I also triggered an update on the device and new temperature und humidity etc. came through.

So I really think it's fine.

Thank you very much.

Smanar, 2021-03-04:
Yep it create 1 devices by endpoint, and 1 by cluster, it s the normal working mode for deconz.
So if all is fine, I can make the PR ?

easybeat, 2021-04-28:
> Yep it create 1 devices by endpoint, and 1 by cluster, it s the normal working mode for deconz.
> So if all is fine, I can make the PR ?

Hi Smanar

Just updated to 2.11.01 and now finally paired TS0202 succesfully. Unfortunately small problem. Temperature and Humidity values do not update automatically. If I press read on cluster then the values are updated.

Is there something you can do about it?

Thank you

Smanar, 2021-04-28:
Can be reporting issue, but all seem fine in the code, it have worked on your first try ?

Just Humidity and Temperature ?
Can you share the Json from thoses 2 entries.

Perhaps a step haven't worked during the inclusion, the set reporting part.

easybeat, 2021-04-28:
Yes it is only Humidity and Temperature which is not working.

I will try to do the inclusion again.

Smanar, 2021-04-28:
Will be fine if can catch log during inclusion in same time with "info" and "info_l2" (but can be hudge, just need all about the cluster 0x0402 and 0x0405, log about reporting or bind)

easybeat, 2021-04-29:
Ok Smanar,

I deleted the device and restarted deconz and the joined it again. It looks wierd now as "lightlevel 71"

[image]

And also no temperature or humidity comes through.

here is the log-fiel during paring:
[deconz 4-1senor-log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6400014/deconz.4-1senor-log.txt)

Hope this is helping?

Smanar, 2021-04-29:
```
17:58:19:493 0x14B457FFFE0BDA59 (TS0202) create binding for attribute reporting of cluster 0x0001 on endpoint 0x01
17:58:19:493 0x14B457FFFE0BDA59 (TS0202) create binding for attribute reporting of cluster 0x0500 on endpoint 0x01
17:58:19:497 0x14B457FFFE0BDA59 (TS0202) create binding for attribute reporting of cluster 0x0001 on endpoint 0x02
17:58:19:498 0x14B457FFFE0BDA59 (TS0202) create binding for attribute reporting of cluster 0x0402 on endpoint 0x02
17:58:19:500 0x14B457FFFE0BDA59 (TS0202) create binding for attribute reporting of cluster 0x0405 on endpoint 0x02
17:58:19:502 0x14B457FFFE0BDA59 (TS0202) create binding for attribute reporting of cluster 0x0001 on endpoint 0x03
17:58:19:503 0x14B457FFFE0BDA59 (TS0202) create binding for attribute reporting of cluster 0x0400 on endpoint 0x03
```

At least bind seem good ...
Can you check setting for temperature or humidity ?

Endpoint 0x02
Cluster 0402 or 0405
Attribute 0x0000
double clic on the value, and press" read config" on the new panel (probably need to awake the device.

Value used by deconz are
- For humidity
```
        rq.minInterval = 10;
        rq.maxInterval = 300;
        rq.reportableChange16bit = 100; // resolution: 1%
```

- For temperature
```
            rq.minInterval = 10;           // value used by Hue bridge
            rq.maxInterval = 300;          // value used by Hue bridge
            rq.reportableChange16bit = 20; // value used by Hue bridge
```

Smanar, 2021-04-29:
Edit:

Ok, I have found the problem, those cluster are not used, the device use tuya cluster instead, need to re write code.

Smanar, 2021-04-29:
Can you try this new code
```
sudo apt install deconz-dev
git clone --branch tuya_sensor https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

With this code you will have new log starting with "tuya debug", if the code don't work, logs can be usefull.

There is too a command to enable reporting, but idk if this setting is usefull yet, what is the utility to disable a reporting ....

easybeat, 2021-04-30:
Hi

it looks different now but still no luck with temperature and humidity

[image]

[image]

Luminance is working but it should also have presence.

Here the log file:
[deconz 4-1senor-log2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6405150/deconz.4-1senor-log2.txt)

I did connect this sensor with zigbee2mqtt and attributes created there were as follows:

[image]

Does this help?

Thanks for help

Smanar, 2021-04-30:
You don't need to re-include it, the inclusion is fine.
It s just this device don't use classic cluster
Something you can do is looking logs (with "info" and "info-l2"), and try to change the temperature.

You need to see some logs about a cluster 0xFC00 or better logs starting by
```
Tuya debug 4 : ............
Tuya debug 5 : ............
```

easybeat, 2021-05-01:
OK started the dev instance again and now it is showing like this:

[image]

and the log is here when I read temperature:
[deconz 4-1senor-log3.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6409424/deconz.4-1senor-log3.txt)

if you search for /sensors/61 then you find the event.

Does this help?

Thanks

Smanar, 2021-05-01:
Ok so still nothing about the tuya cluster ..
Have inluded the "reporting" command, on the temperature sensor and (only this one), you will be able to see it in his json, to activate it, need to use the API

`curl -H 'Content-Type: application/json' -X PUT -d '{"reporting": true}' http://IP:PORT/api/KEY/sensors/61/config`

easybeat, 2021-05-01:
Hi

Do I have to rebuild deconz? As I get this error:

[{"error":{"address":"/sensors/62/config/reporting","description":"parameter, reporting, not available","type":6}}]

Thanks

Smanar, 2021-05-01:
Yep, like for previous tests, but no need to re-include the device (I think the restart will be enought)

easybeat, 2021-05-01:
Sorry Smanar...not very good at this. I have to do all this:

sudo apt install deconz-dev
git clone --branch tuya_sensor https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins

and then run the command?

Smanar, 2021-05-01:
So you have never tested the modified code ?

Yes, but you need a linux machine with full OS (not a docker), thoses command will install the dev package, compile the code and replace the file.
Then after just run deconz, better to close it before the last command, else it will close up itself.

easybeat, 2021-05-01:
yes I have a full linux ubuntu

but If I run this again, I get an error message:
fatal: Zielpfad 'deconz-rest-plugin' existiert bereits und ist kein leeres Verzeichnis.

sorry it is in German and says that the path already exists

Smanar, 2021-05-01:
ha yep sorry

So you can just switch branch

```
cd deconz-rest-plugin
git pull
git checkout tuya_sensor 
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

easybeat, 2021-05-01:
Still get this error:
{"error":{"address":"/sensors/63/config/reporting","description":"parameter, reporting, not available","type":6}}]

without including the device...everytime I restart deconz it gets a new ID....by the way.

easybeat, 2021-05-01:
Sorry my fault...it did work with Id 61

here the log after temperature update:
[deconz 4-1senor-log4.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6409954/deconz.4-1senor-log4.txt)

Does that help?

easybeat, 2021-05-01:
And here also the log inlcuding a read on temperature:

[deconz 4-1senor-log5.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6410009/deconz.4-1senor-log5.txt)

Smanar, 2021-05-01:
Yes, nice, thx a lot, lot of usefull data.

```
18:41:25:565 Tuya debug 4 : Address 0x14B457FFFE0BDA59 Payload 00006e02000400000064
18:41:25:565 Tuya debug 5 : Status: 0 Transid: 0 Dp: 622 (0x02,0x6E) Fn: 0 Data 100
```

This one is the battery

```
18:41:27:966 Tuya debug 4 : Address 0x14B457FFFE0BDA59 Payload 00006b020004000000da
18:41:27:966 Tuya debug 5 : Status: 0 Transid: 0 Dp: 619 (0x02,0x6B) Fn: 0 Data 218
```

Yeah, this one is the temperature ^^

```
18:41:29:167 Tuya debug 4 : Address 0x14B457FFFE0BDA59 Payload 00006c0200040000002e
18:41:29:167 Tuya debug 5 : Status: 0 Transid: 0 Dp: 620 (0x02,0x6C) Fn: 0 Data 46
```

Humidity

And all in first logs, I think I will disable the cluster 0x0402 and 0405, as they don't look usefull.

So what you have done, You have enabled the reporting or not ?
It realy don't work if you disable the reporting ?
I think if you heat the sensor, you will have in log
>18:41:27:966 Tuya debug 5 : Status: 0 Transid: 0 Dp: 619 (0x02,0x6B) Fn: 0 Data 218

0x02 0x6C is for the temperature, the value is 218 = 21.8 degrés

If I m right now the temperature is working, but not with the cluster 0x0402 but the 0xEF00 and with the same log than the previous line.

easybeat, 2021-05-03:
Good morning Smanar

I have done nothing with reporting.

it is not enabled:
[image]

and if I try to set reporting, I get this:
[image]

what can I do next?

Smanar, 2021-05-03:
Nope, forget "normal working mode", this device don't use this "stuff", It work only with tuya one.

So to enable reporting (the tuya one, not the one on your previous capture) you need to use the API

`curl -H 'Content-Type: application/json' -X PUT -d '{"reporting": true}' http://IP:PORT/api/KEY/sensors/61/config`

But you already have the good log
>18:41:29:167 Tuya debug 5 : Status: 0 Transid: 0 Dp: 620 (0x02,0x6C) Fn: 0 Data 46

So 2 solutions.
- the reporting is not needed, and you will have this logs every time you heat the sensor (and it will update the json too, if working)
- the reporting is needed and you need to enable it using the API.

easybeat, 2021-05-04:
Good morning Smanar

I'm not sure if I fully understand what you want to achieve. But I did run the curl command and it was successful.

Should the temperature now automatically be reported into the node?

Because it doesn't look that it is...

[image]

Thanks

Smanar, 2021-05-04:
>Should the temperature now automatically be reported into the node?

Yes , but not where you are looking.
This device don't use the attribute 0x0402, so it s useless to take a look here.

The only place you can see if it s working, it s in the log (and only in log, nowhere in the GUI) with the line
>18:41:29:167 Tuya debug 5 : Status: 0 Transid: 0 Dp: 620 (0x02,0x6C) Fn: 0 Data 46

I just need to know if you have this line in log if you heat the sensor (to make a temperature change), with or without the tuya reporting enabled (the one you have set using the api)

easybeat, 2021-05-04:
OK. With enabled:

16:53:50:554 Tuya debug 4 : Address 0x14B457FFFE0BDA59 Payload 00006b020004000000ea
16:53:50:556 Tuya debug 5 : Status: 0 Transid: 0 Dp: 619 (0x02,0x6B) Fn: 0 Data 234

16:55:37:127 Tuya debug 4 : Address 0x14B457FFFE0BDA59 Payload 00006b020004000000eb
16:55:37:128 Tuya debug 5 : Status: 0 Transid: 0 Dp: 619 (0x02,0x6B) Fn: 0 Data 235

I think this means temp. went up from 23.4 to 23.5

Then I restarted deconz. So reporting should be off?

Now I can't find entries about Tuya debug.

If I press the wake up button then the entry is in the log. With or without reporting enabled.

OK?

easybeat, 2021-05-04:
Sorry Smanar, not correct.

I repeated the test. Restarted deconz, run curl command and started debugging. Started heating up thermostat for about 2 Mins.

BUT NOT ENTRIES at all!

Only if I press the wake up button the entry is in the log.

Smanar, 2021-05-04:
Arf :(

I don't think it s from the "reporting time" setting, I think the defaut value is ok to make the device working.

>Then I restarted deconz. So reporting should be off?

Nope this setting is memorised on the device itself, at restart the value is "null" and this value is updated when the device send it.

Have added one more debug line, can you update your code

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

And just try to heat again the device to see if you have something new in log ? (still with "info", no need to re-include)

easybeat, 2021-05-05:
Had to enable with curl otherwise no entries but now this:

08:36:09:052 Tuya debug 3 : Address 0x14B457FFFE0BDA59, Command 0x06, Payload 00006b020004000000ff
08:36:09:053 Tuya debug 4 : Address 0x14B457FFFE0BDA59 Payload 00006b020004000000ff
08:36:09:054 Tuya debug 5 : Status: 0 Transid: 0 Dp: 619 (0x02,0x6B) Fn: 0 Data 255

08:36:10:251 Tuya debug 3 : Address 0x14B457FFFE0BDA59, Command 0x06, Payload 00006c0200040000002b
08:36:10:252 Tuya debug 4 : Address 0x14B457FFFE0BDA59 Payload 00006c0200040000002b
08:36:10:252 Tuya debug 5 : Status: 0 Transid: 0 Dp: 620 (0x02,0x6C) Fn: 0 Data 43

Does that help?

Thanks

Smanar, 2021-05-05:
>Had to enable with curl otherwise no entries but now this:

So you need to enable the tuya reporting to have values ?

>Does that help?

Yep, thoses one are the good one to check but you need to
- have them everytime the temperature change, or at least every X minuts of reporting
- They need to update the value in the JSON in same time

easybeat, 2021-05-05:
Yes I did have to enable tuya reporting. Otherwise no entries at all.

Is there a chance that we get the device fully to work? As there is also luminance and present which are nowhere in the log at all...

Smanar, 2021-05-05:
>Is there a chance that we get the device fully to work? As there is also luminance and present which are nowhere in the log at all

All value that not use the "classic way" can be used with the "tuya way", but on your first post you have said only temperature and humidity was missing ?

>Yes I did have to enable tuya reporting. Otherwise no entries at all.

And now all is working or you still have only 1 value when you wake up the device ?

easybeat, 2021-05-06:
> All value that not use the "classic way" can be used with the "tuya way", but on your first post you have said only temperature and humidity was missing ?
> 
Yes. This is correct. Only the temperature and humidity did not report automatically. And I checked again. This is still the case. All other values are reported automatically.

[image]
 
> And now all is working or you still have only 1 value when you wake up the device ?

Should the temperature and humidity now be reported to ioBroker (as it does not)? Or only create an entry in the log?

easybeat, 2021-05-06:
Just did run the log now for 10 Mins. and the tuya log entry is in every 5 Mins.

So I think reporting is working when enabled.

OK?

Smanar, 2021-05-06:
>Just did run the log now for 10 Mins. and the tuya log entry is in every 5 Mins.

But the value is not updated on the device json ?

>Should the temperature and humidity now be reported to ioBroker (as it does not)? Or only create an entry in the log?

If I m right , it create the entry on log and update the json.

Smanar, 2021-05-08:
> "lastseen":"2021-05-08T08:08Z"
> "lastupdated":"2021-05-05T06:37:14.120"

Yes there is a problem, have found an issue.
Thoses 2 sensors are created for the endpoint 02. So in the code they are on bad endpoint and bad cluster (code is updated with endpoint 01 cluster EF00).

Need to make change on the code.

Smanar, 2021-05-08:
Ok, can test the new code

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

No need to re-include the device. Now if I m right the json will be updated in same time you are seing the log.

easybeat, 2021-05-09:
Hi Smanar

I did the rebuild and restarted deconz, enabled reporting with curl but json still looks like:

{"config":{"battery":100,"offset":0,"on":true,"reachable":true,"reporting":true},"ep":2,"etag":"15fa624b02456fd4808c485f3d539e13","lastseen":"2021-05-09T17:42Z","manufacturername":"_TZ3210_zmy9hjay","modelid":"TS0202","name":"Temperature 61","state":{"lastupdated":"2021-05-05T06:37:14.120","temperature":2556},"type":"ZHATemperature","uniqueid":"14:b4:57:ff:fe:0b:da:59-02-0402"}

Still not working.

easybeat, 2021-05-09:
Sorry Smanar...just send the message and went back to get json again...and now it seems to work:

{"config":{"battery":100,"offset":0,"on":true,"reachable":true,"reporting":true},"ep":2,"etag":"802b4b46193575d9b09a1d06a9a31ae8","lastseen":"2021-05-09T17:44Z","manufacturername":"_TZ3210_zmy9hjay","modelid":"TS0202","name":"Temperature 61","state":{"lastupdated":"2021-05-09T17:44:53.963","temperature":2460},"type":"ZHATemperature","uniqueid":"14:b4:57:ff:fe:0b:da:59-02-0402"}b

easybeat, 2021-05-09:
I will enable value recording on ioBroker now to see if it is reported every 5 minutes and let you know tomorrow morning.

Thanks

Smanar, 2021-05-10:
>but when I press "read" in cluster the value comes through correctly. Has this something to do with reporting?

Nope, this command read directly the attribute, the reporting is something "natural" it s the device that send the value itself, when you use this command you force a read attribute on temperature cluster endpoint 0x02, but the device make a report on endpoint 0x01 cluster 0xEF00

Have updated the code for the humiditiy, don't need to re include the device.

And BTW you still need to "enable the reporting" at every load ?

easybeat, 2021-05-11:
Hi Smanar

For me this all looks good. Had value recording running for more than 24 hours and they are reported very regular and correct.

You can make the PR.

I will close this now.

Thank you for your help...I also learnt a lot :-)

Kind regards
Beat

closed 2021-05-11 by easybeat

Smanar, 2021-05-19:
@easybeat I have a question !
If you disable the reporting using the API {"reporting":false}, it realy disable the reporting ? what can be the utility of this ?

easybeat, 2021-05-19:
@Smanar No it doesn't make sense to do "reporting:false". Why do you think I would do this?

No need for me.

Smanar, 2021-05-19:
it s exactly the manup reaction ^^.

But I don't see the utility of this command, it realy disable TOTALY the reporting for temperature and humidity ?

Smanar, 2021-05-22:
@easybeat if you still have your dev machine ?

I have a new branch to test
```
cd deconz-rest-plugin
git pull
git checkout tuya_sensor_test
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

On this one I have remove the field "config/reporting" as this setting is probably useless, so now the code send itself the command to enable it

To test it, you just need to re-include your device, you will have in log (with "info")
> Send Tuya request: Dp_type .................
> Tuya device ....... reporting status state ........

easybeat, 2021-05-27:
Do I have to install latest release to do that?

[19 more replies]