dresden-elektronik/deconz-rest-plugin #4512: Can anyone explain how I can find out from which device these entries are coming:

oywino, 2021-03-03:
## Describe the question or issue you are having

## Screenshots

[image]

## Environment

 - Host system: (Raspberry Pi / PC / **NAS**)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / **Marthoc Docker container** / Windows / Virtual Machine)
 - Firmware version: (26660700)
 - deCONZ version: (2.10.00 / 12/23/2020)
 - Device: (ConBee I / **ConBee II** / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (**yes** / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

See screenshot above.

## Additional context

oywino, 2021-04-01:
Bump! Anyone?

Smanar, 2021-04-01:
Try enabling more flag, like APS
You will be able to see previous logs, with more information like the MAC address

It s a device with the commissioning cluster.

oywino, 2021-04-01:
Hi, and thank you for stepping in.
However,I have no idea how to enable a flag.  Could you please explain?

Smanar, 2021-04-01:
Where are you looking logs ?
in console ? > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags
in deconz/help/gui ? > flags are the checkbox.

oywino, 2021-04-01:
The messages were captured from the console.
I'm not sure if they also appear in the log, but I'll check that. Thank you for guiding me 👍

Mimiix, 2021-04-01:
@oywino If you open another issue later, please add more info in the description on creation.

oywino, 2021-04-01:
> @oywino If you open another issue later, please add more info in the description on creation.

Sorry, of course - I have updated. I was hoping that whoever wrote this software would know the meaning of these messages.

closed 2021-05-01 by github-actions[bot]