dresden-elektronik/deconz-rest-plugin #4513: Moes House / EAdimmer EDM-1ZAA-EU / TS0601 dimmer

andys73, 2021-03-03:
This is a Zigbee dimmer identified by deconz as smart plug under lights  so I can only turn it on and off there is no dimming functionality
https://zigbee.blakadder.com/Earda_EDM-1ZAA-EU.html

Many thanks for your great work!

## Device

- Product name: EDM-1ZAA-EU
- Manufacturer: _TZE200_9i9dt8is
- Model identifier: TS0601
- Device type :
  - Light

## Screenshots
[image]
[image]
[image]
[image]
[image]

Smanar, 2021-03-03:
You are lucky, we have just finished the tuya dimmer https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4414

Do you have a linux machine to test code modification ?

andys73, 2021-03-03:
really lucky :)
I have a raspberry pi to my Home assistant but if you give me instructions what do I have other Linux systems that I can use to test.

Smanar, 2021-03-03:
I have added your device to the code, so to test it

You need to instal deconz on a linux machine then take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch tuya_dimmer https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

And re-include the device.

I don't remember if it will work in phoscon, else you can make HA use the second deconz gateway, but don't ask me how ....

andys73, 2021-03-03:
I 'll try to test and report back . Is night here so it might be in few days

choss38, 2021-03-04:
Hi, there are some make errors:

```
pi@phoscon:~/deconz-rest-plugin` $ git pull
remote: Enumerating objects: 28, done.
remote: Counting objects: 100% (28/28), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 28 (delta 21), reused 28 (delta 21), pack-reused 0
Dépaquetage des objets: 100% (28/28), fait.
Depuis https://github.com/Smanar/deconz-rest-plugin
   5bce1a22..dde87584  tuya_dimmer      -> origin/tuya_dimmer
 * [nouvelle branche]  multisensor      -> origin/multisensor
   7e8cec47..7e14a9b3  orvibo           -> origin/orvibo
   b39550a9..5b87847e  tuya_extender    -> origin/tuya_extender
   3f64f6ed..5048db07  tuya_siren_clean -> origin/tuya_siren_clean
Mise à jour 5bce1a22..dde87584
Fast-forward
 resource.cpp    |  3 ++-
 rest_lights.cpp | 20 ++++++++++----------
 tuya.cpp        | 44 ++++++++++++++++++++++++++++++++------------
 tuya.h          |  4 ++--
 4 files changed, 46 insertions(+), 25 deletions(-)
pi@phoscon:~/deconz-rest-plugin $ qmake && make -j2
make -f Makefile.Release
make[1] : on entre dans le répertoire « /home/pi/deconz-rest-plugin »
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.10.00\" -DGW_SW_DATE=1614802421 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"dde87584dea7c86afc1fc52a20a36c88721717a9\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/air_quality.o air_quality.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.10.00\" -DGW_SW_DATE=1614802421 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"dde87584dea7c86afc1fc52a20a36c88721717a9\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/authorisation.o authorisation.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.10.00\" -DGW_SW_DATE=1614802421 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"dde87584dea7c86afc1fc52a20a36c88721717a9\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/bindings.o bindings.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.10.00\" -DGW_SW_DATE=1614802421 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"dde87584dea7c86afc1fc52a20a36c88721717a9\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/inc [... cut ...] Port -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/appliances.o appliances.cpp
make[1]: *** [Makefile.Release:1457: release/tuya.o] Error 1
make[1]: *** Attente des tâches non terminées....
make[1] : on quitte le répertoire « /home/pi/deconz-rest-plugin »
make: *** [Makefile:40: release] Error 2
```

andys73, 2021-03-04:
Hi,

I have used a raspberry pi and it produced an object
-rwxr-xr-x 1 pi pi 2494524 Mar  4 19:11 ../libde_rest_plugin.so
pi@adypi:~/deconz-rest-plugin $ ls -all ../libde_rest_plugin.so

testing it now…

Regards

andys73, 2021-03-04:
It compiled fine for me on a raspbery pi 3 but I cannot seems to connect to deCONZ to the USB stick.. hope I do not need to reset the conbee?

andys73, 2021-03-04:
ok managed to get it started.

It does not properly work.
I get the property now in Phoscon for dimming.
Problems noted:
 1. Dimming works from time to time, i.e. if I increase the brightness it I cannot decrease it anymore. 
 2. I can turn off the light but I cannot turn it back on.

Smanar, 2021-03-05:
I think, you was used an old version for your first try, you can update the code just with

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

>     Dimming works from time to time, i.e. if I increase the brightness it I cannot decrease it anymore.

You can only increase and never decrease ? Can you take a look in log, on deconz/help/log, with "info" and "info_l2".
When you try to change the brightness with the software, you will see in log "debug tuya, send request XXXXXXXXXXXXXXXXxxx".
Then change it manualy using the device, you will see "debug tuya request XXXXXXXXXXXXXXXXXXXX", can you share this logs to compare them ?

>    I can turn off the light but I cannot turn it back on.

I think it s same issue, can you make the same test than previously but with on/off ?

On the other device it seem you can send only brightness command, but it seem your device need the use of the 2 commands. And on my code I m making

```
if (bri) 
{
}
else if (on)
{
}
```

So you can't send 2 command in same time, I think the problem is from here, phoscon use 2 command in same time.

andys73, 2021-03-06:
It seems that it used the latest...

I will try to figure out how to get the logs.

pi@adypi:~ $ cd deconz-rest-plugin/
pi@adypi:~/deconz-rest-plugin $ git pull
remote: Enumerating objects: 165, done.
remote: Counting objects: 100% (165/165), done.
remote: Compressing objects: 100% (73/73), done.
remote: Total 177 (delta 124), reused 117 (delta 92), pack-reused 12
Receiving objects: 100% (177/177), 389.52 KiB | 1.53 MiB/s, done.
Resolving deltas: 100% (125/125), completed with 33 local objects.
From https://github.com/Smanar/deconz-rest-plugin
 * [new branch]        DS82       -> origin/DS82
   57a62742..920f81ce  doorlock   -> origin/doorlock
   5d7e81fa..51586b47  master     -> origin/master
 * [new branch]        tmpbranch7 -> origin/tmpbranch7
Already up to date.
pi@adypi:~/deconz-rest-plugin $ qmake && make -j2
make -f Makefile.Release
make[1]: Entering directory '/home/pi/deconz-rest-plugin'
make[1]: Nothing to be done for 'first'.
make[1]: Leaving directory '/home/pi/deconz-rest-plugin'

Smanar, 2021-03-06:
For logs, as you have the GUI, will be easy.
Just go in deconz/help/log, check "info" and "info_l2"

andys73, 2021-03-08:
thanks, will try to get the logs later today

andys73, 2021-03-09:
logs attached.
[info-logs.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6107633/info-logs.zip)
Thanks!

Smanar, 2021-03-09:
> 20:38:54:454 Send Tuya request: Dp_type: 0x02, Dp_identifier 0x02, data: 000001f4
> 20:38:54:563 Tuya debug 5 : Status: 94 Transid: 61 Dp: 514 (0x02,0x02) Fn: 0 Data 500
> 20:38:54:605 Websocket 192.168.0.128:49406 send message: {"e":"changed","id":"6","r":"lights","state":{"alert":null,"bri":127,"on":true,"reachable":true},"t":"event","uniqueid":"bc:33:ac:ff:fe:f4:ce:12-01"}

> 20:38:55:432 Send Tuya request: Dp_type: 0x02, Dp_identifier 0x02, data: 00000379
> 20:38:55:741 Tuya debug 5 : Status: 94 Transid: 61 Dp: 514 (0x02,0x02) Fn: 0 Data 889
> 20:38:55:765 Websocket 192.168.0.128:49406 send message: {"e":"changed","id":"6","r":"lights","state":{"alert":null,"bri":225,"on":true,"reachable":true},"t":"event","uniqueid":"bc:33:ac:ff:fe:f4:ce:12-01"} (ret = 149)

> 20:38:57:481 Send Tuya request: Dp_type: 0x02, Dp_identifier 0x02, data: 00000023
> 20:38:57:580 Tuya debug 5 : Status: 94 Transid: 61 Dp: 514 (0x02,0x02) Fn: 0 Data 35
> 20:38:57:609 Websocket 192.168.0.128:49406 send message: {"e":"changed","id":"6","r":"lights","state":{"alert":null,"bri":8,"on":true,"reachable":true},"t":"event","uniqueid":"bc:33:ac:ff:fe:f4:ce:12-01"}

I don't see problem for dimmer ?
But I have found the problem for "off" and yes was because you are using "bri" and "on" in same time, can you try the new code pls ?

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

andys73, 2021-03-10:
I have tested with the new code and it works all ok.

- on / off - OK
- dimming slide up and down - OK
- remembering the dimming level - OK

Many thanks for the great work and looking forward to have it in the release.

Smanar, 2021-03-10:
Ha nice, will ask to other use, now I have modified the code to be sure it s still working for him.

andys73, 2021-03-24:
Hi @Smanar I have just installed the deconz update 2.10.4 that was made available. Did this ticket did not make it in 2.10.4 or do I need to remove it and add it again? Thank you!

Smanar, 2021-03-24:
No it s not included yet.
This issue will be closed when the PR will be validated. You can check direclty the PR too.

If you need the last version possible I can update the PR with last version from official github if you want ?

andys73, 2021-03-25:
I am using HA / HASSIO on a pi 3 so whenever it could get in that official version it will help me :)
Many thanks,
Adrian

andys73, 2021-04-16:
Hi @Smanar can this issue be closed and be sure that the code ends up in the next release? Many thanks

Smanar, 2021-04-16:
Still in waiting list ...
Hoping for it will be validated on next session.

But don't worry, all will be automatic, the issue will be closed in same time that the PR will be validated.

closed 2021-04-18 by manup