dresden-elektronik/deconz-rest-plugin #4514: Xiaomi Smart Plug issues (consumed energy, icons, switching on/off)

raidenlal, 2021-03-03:
## Describe the bug
I have paired two EU Xiaomi Smart plugs with conbee2. They behave weirdly:

- they are reported as light bulbs in "Light group"
- in "connected lights" menu when adding lights/sockets the icon is correct (plug)
- total consumed energy (kWh) is reported on one plug and not on the other one
- I can switch on/off by clicking on the "bulb icon" in the lights group
- I can not switch on/off by using the "light" in home assistant: "Failed to call service switch/turn_off. /lights/55 resource, /lights/55, not available"

## Steps to reproduce the behavior
1. Pair Xiaomi Smart plug
2. Add to phoscon light group
3. Check icon in group
4. Check device in HA and entity for switching and energy consumption metering

## Expected behavior
Correct icons, ability to switch on/off plug in HA, consumed energy correctly reporting.

## Screenshots
Icons ok:
[image]
Icons not ok:
[image]
Energy consumption missing:
[image]

## Environment
 - Host system: Raspberry Pi 4
 - Running method: Raspbian? (rpi image on sd card)
 - Firmware version: 26660700
 - deCONZ version: 2.09.03 / 12/23/2020
 - Device: ConBee II
 - Do you use an USB extension cable: no, but other devices are fine
 - Is there any other USB or serial devices connected to the host system? Z-Stick Gen5+

## deCONZ Logs
These logs seem really irrelevant but I will paste them anyway:
22:12:16:946 ZCL attribute report 0x0017880103A52F73 for cluster: 0x0001, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
22:12:37:652 remove dead link for (775F, 4E3D)
22:12:37:892 reuse dead link (dead link container size now 0)
22:12:47:295 Websocket disconnected 127.0.0.1:60960, state: 0, close-code: 1000, reason: 
22:12:48:451 New websocket 127.0.0.1:33074 (state: 3) 
22:12:49:087 ZCL attribute report 0x00158D0001EF43C5 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
22:12:49:626 ZCL attribute report 0x00158D0001712FAA for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
22:12:50:757 [INFO] - No button map for: RWL021, unicast to: 0x0000, endpoint: 0x02, cluster: 0x0001, command: 0x0A, payload: 210020BD, zclSeq: 59
22:12:50:759 ZCL attribute report 0x0017880103C98DBE for cluster: 0x0001, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
22:12:51:585 Current channel 25
22:12:51:589 CTRL got nwk update id 1
22:12:51:593 Device TTL 3177 s flags: 0x6
22:12:53:907 [INFO] - No button map for: RWL020, unicast to: 0x0000, endpoint: 0x02, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq: 112
22:12:53:909 ZCL attribute report 0x001788011033D7B9 for cluster: 0x0001, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
22:13:08:814 ZCL attribute report 0x04CF8CDF3C8E25F2 for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
22:13:08:815 0x04CF8CDF3C8E25F2 extract Xiaomi special attribute 0x00F7
22:13:08:816 	64 on/off 1
22:13:08:817 	03 Device temperature 24 °C
22:13:08:818 	98 power 1.175000 (1)
22:13:08:819 	95 consumption 0.066287 (66)
22:13:08:820 	96 voltage 2390.000000 (239)
22:13:08:822 	97 current 4.916318 (5)
22:13:08:823 	05 RSSI dB (?) 1 (0x0001)
22:13:08:824 	9a unknown 0 (0x00)
22:13:08:825 	08 unknown 278 (0x0116)
22:13:08:826 	07 unknown 0 (0x0000000000000000)
22:13:08:827 	09 unknown 2816 (0x0B00)
22:13:08:828 	0b unknown 0 (0x00)
22:13:08:829 	9b Consumer connected (yes/no) 1
22:13:15:572 remove dead link for (C8D9, E9C1)
22:13:24:452 reuse dead link (dead link container size now 0)
22:13:35:070 binding for cluster 0x0006 of 0x84182600000E4211 exists (verified by reporting)
22:13:35:071 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x84182600000E4211 (seems to be active)
22:13:43:157 ZCL attribute report 0x84182600000E4211 for cluster: 0x0006, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
22:13:49:085 ZCL attribute report 0x00158D0001EF43C5 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
22:13:49:626 ZCL attribute report 0x00158D0001712FAA for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
22:13:51:584 Current channel 25
22:13:51:588 CTRL got nwk update id 1
22:13:51:592 Device TTL 3117 s flags: 0x6
22:13:56:118 Bind response success for 0x04cf8cdf3c8e25f2 ep: 0x01 cluster: 0x0006
22:14:05:678 ZCL attribute report 0x84182600000E5F0B for cluster: 0x0006, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
22:14:06:412 [INFO] - No button map for: RWL021, unicast to: 0x0000, endpoint: 0x02, cluster: 0x0001, command: 0x0A, payload: 210020AD, zclSeq: 200
22:14:06:413 ZCL attribute report 0x0017880103E57334 for cluster: 0x0001, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
22:14:10:640 dev /dev/ttyAMA0
22:14:10:641 GW firmware version: 0x26660700
22:14:10:641 GW firmware version is up to date: 0x26660700
22:14:10:918 [INFO] - No button map for: RWL021, unicast to: 0x0000, endpoint: 0x02, cluster: 0x0001, command: 0x0A, payload: 21002089, zclSeq: 151
22:14:10:919 ZCL attribute report 0x0017880103E57484 for cluster: 0x0001, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
22:14:19:118 Bind response success for 0x04cf8cdf3c8e25f2 ep: 0x01 cluster: 0x0006
22:14:26:686 Websocket disconnected 127.0.0.1:33074, state: 0, close-code: 1000, reason: 
22:14:27:986 ZCL attribute report 0x54EF441000078447 for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F

Mimiix, 2021-03-04:
What smart plug are we talking about? There's numerous of them around. Can you prvide a screen of the basic cluster?

raidenlal, 2021-03-04:
EU socket type Xiaomi Smart Plug that looks like this:
[image]

Might I ask what is "basic cluster"?
[image]
[image]

Mimiix, 2021-03-04:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

raidenlal, 2021-03-04:
Sorry I didn't find that tutorial.
"Dryer" plug reports total consumed kWh, but can't turn on/off in hass:
[image]
"Washing machine" plug that does not report total consumed and can't be turned on/off in hass:
[image]

raidenlal, 2021-03-08:
Is there anything I can try to resolve this issue?

Mimiix, 2021-03-09:
@SwoopX Can you help out? I'm a bit lost

raidenlal, 2021-03-10:
Today I have rescanned cluster info basic and new fields appeared:
[image]
also now both sensors report wattage. Weird. It "fixed itself" after unplugging/plugging both smart plugs. Do not know if this helped or there was something else happening so I could leave a note here how it was fixed for the future.

Cannot reproduce this issue anymore.

closed 2021-03-10 by raidenlal