dresden-elektronik/deconz-rest-plugin #4517: Tuya Temperature Sensor (Zigbee) not visible in Phoscon / Hue

Zazou89, 2021-03-03:
## Describe the bug
I added a Tuya Temperature sensor in my installation, it's visible on Deconz but not on Phoscon / Hue-homebridge. Isn't compatible ?

## Steps to reproduce the behavior
1. Add in Phoscon

## Expected behavior
Visible in Phoscon / Hue

## Screenshots
[image]
[image]

## Environment
 - Host system: Nas
 - Running method: Marthoc Docker container
 - Firmware version: 0x26670700
 - deCONZ version: 2.10.01
 - Device: ConBee II
 - Do you use an USB extension cable: Yes

## deCONZ Logs
Nothing wrong

## Additional context
No.

ebaauw, 2021-03-03:
Looks like the API plugin hasn’t created any resource; typically the label on the node changes when it has.

Mimiix, 2021-03-04:
Is this supported?  What does the Basic cluster say?

fribse, 2021-03-04:
I'm seeing this excact problem here. I had a multipurpose sensor that lost connection, so I've removed it, and now it shows up as TS201 in deconz, but not in phoscon. I have one more of these already  in the network, but that was added before upgrading to 2.09.03

 - Host system: PC (NUC with PVE)
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26670700)
 - deCONZ version: (2.09.03)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? AEOTEC Gen5 z-wave

[image]

[image]

[image]

[image]

This is the other one I have, it was shown as a Multipurpose Sensor, and I just added the room in front of it for naming.

[image]

fribse, 2021-03-05:
Ok, in my case it's really weird, I see the TS0201 in HomeAssistant, but not in Phoscon???

Mimiix, 2021-03-05:
@fribse That could be the case if Phoscon didn't add support yet. deCONZ != Phoscon.

fribse, 2021-03-05:
@Mimiix Ok, I know deconz is not the same as phoscon, but what bothering me is that I have two of these, and they were enlisted before going to 2.09.03, and they were recognized. And now I've removed it, and readded it, and now it doesn't recognize it??
This is the one that is shown in Deconz:
[image]

Mimiix, 2021-03-05:
@fribse That's a interesting one. Does reading the basic cluster in deCONZ change anything? 

Otherwise: I suggest opening a bug report at the phoscon repo.

fribse, 2021-03-05:
No, I tried reading all the clusters, and also do a read on descriptors, no change 😭 

😆

fribse, 2021-03-05:
@Mimiix Where do I find the right phoscon repo?

Mimiix, 2021-03-05:
https://github.com/dresden-elektronik/phoscon-app-beta < That one :)

fribse, 2021-03-05:
Oh, the 'beta' one, gotcha!

closed 2021-04-03 by github-actions[bot]

Iceman808, 2021-08-13:
I got 2 of the Tuya Temperature Sensors. 

Connection with the Phoscon App is possible but the Sensors does not appears in the Sensor List.

On the deconz App I see both Sensors, but one of it returns battery Level an the other not. 

I tried to delete the Sensor in deconz but evertime I connect it again, I get the same result (without battery Level...)

How can I fix it?

[image]
[image]
[image]
[image]

I used:

    Host system: Nas
    Running method: marthoc/deconz:latest
    Firmware version: 0x26680700
    deCONZ version: v2.12.3-beta
    Device: ConBee II
    Do you use an USB extension cable: No

Mimiix, 2021-08-13:
They are different.

Can you read the basic cluster and provide screens of these @Iceman808 ?

Iceman808, 2021-08-13:
Yes, I can:

[image]
[image]

And thx for your quick reply.

Mimiix, 2021-08-13:
@Iceman808 The above one isn't read. We need them both to show the manufacutrer en model id

Iceman808, 2021-08-13:
@Mimiix If I press the read Button in the Basic Cluster Info, the Informations are not changes and the sensor 28 is the one which works...

Mimiix, 2021-08-13:
Doesn't matter, i want to see if my suspicion is correct. They don't have the same Mac Adress prefix, so i suspect the devices are not the same.

By comparing the basic cluster, we can make sure.

Click the button on the device a few times while clicking read, that should make it apear.

Iceman808, 2021-08-13:
Ok, now I get the Basic Cluster Information form the working device.

[image]

Mimiix, 2021-08-13:
@Smanar do you know why?

Iceman808, 2021-08-13:
After I read the other Devie again with the Button Click, I get a different Application Version.

[image]

Smanar, 2021-08-13:
But the battery level appear in the API ?
The appearance in the deconz app don't depend of the API, and can take somes hours to display the icon.

Iceman808, 2021-08-13:
It s you that have set the name "Humidity 34" ?

No the Name(s) "Himudity 34" and "28" are automaticly added.

Here are the API Informations:
`"33": {
            "config": {
                "battery": null,
                "offset": 0,
                "on": true,
                "reachable": true
            },
            "ep": 1,
            "etag": "138ac49bd6e5c15e1865719cbaf62ac7",
            "lastseen": "2021-08-13T16:06Z",
            "manufacturername": "_TZ2000_a476raq2",
            "modelid": "TS0201",
            "name": "Temperature 33",
            "state": {
                "lastupdated": "2021-08-13T16:06:24.039",
                "temperature": 2420
            },
            "type": "ZHATemperature",
            "uniqueid": "80:4b:50:ff:fe:1a:46:3e-01-0402"
        },
        "34": {
            "config": {
                "battery": null,
                "offset": 0,
                "on": true,
                "reachable": true
            },
            "ep": 1,
            "etag": "b59dd7fc56ea87703a607bb8cd8670d2",
            "lastseen": "2021-08-13T16:06Z",
            "manufacturername": "_TZ2000_a476raq2",
            "modelid": "TS0201",
            "name": "Humidity 34",
            "state": {
                "humidity": 5940,
                "lastupdated": "2021-08-13T16:06:24.158"
            },
            "type": "ZHAHumidity",
            "uniqueid": "80:4b:50:ff:fe:1a:46:3e-01-0405"
        }
    }
}`

And this is the "Humidity 28":

`        "27": {
            "config": {
                "battery": 100,
                "offset": 0,
                "on": true,
                "reachable": true
            },
            "ep": 1,
            "etag": "902f074a58fb204b896d60c31635acac",
            "lastseen": "2021-08-13T16:09Z",
            "manufacturername": "_TZ2000_a476raq2",
            "modelid": "TS0201",
            "name": "Temperature 27",
            "state": {
                "lastupdated": "2021-08-13T16:09:28.847",
                "temperature": 2410
            },
            "type": "ZHATemperature",
            "uniqueid": "5c:02:72:ff:fe:da:bb:04-01-0402"
        },
        "28": {
            "config": {
                "battery": 100,
                "offset": 0,
                "on": true,
                "reachable": true
            },
            "ep": 1,
            "etag": "69e4132f30fe50d3f18bb29c7068dd87",
            "lastseen": "2021-08-13T16:09Z",
            "manufacturername": "_TZ2000_a476raq2",
            "modelid": "TS0201",
            "name": "Humidity 28",
            "state": {
                "humidity": 5930,
                "lastupdated": "2021-08-13T16:09:29.378"
            },
            "type": "ZHAHumidity",
            "uniqueid": "5c:02:72:ff:fe:da:bb:04-01-0405"
        },
`

How can I read the attributes in the cluster 0x0001?

Smanar, 2021-08-13:
Ha yes, effectively there is something strange.

On your previous screenshoot
- Select the cluster 0x0001 "power configuration
- Press the "read" button, it will fill the table

You can need to wake up the device, in same time you press the button.

But you can just need to wait somes hours, the device can need time to send battery value.

Mimiix, 2021-08-13:
@Smanar you notice the different Mac prefix?

Smanar, 2021-08-13:
Yep, but from my memory it impact only the entry creation, after that only the model id and manufacture name are used for checks in code.

But this device can have something special in the power configuration cluster, I don't think the time cluster have an impact on it too.

Iceman808, 2021-08-14:
Hi all,

@Smanar: After ein do the Cluster Read in te Power configuration Section with clicks on the sensor the Battery symbol appears in the deconz App and I geht the BatteryPercent in my fhem installation. 

@Smanar @Mimiix :Thanks for your quick and brillant help.

One question I had left, why the Sensors not appaers in my phoscon App?

Smanar, 2021-08-14:
>After ein do the Cluster Read in te Power configuration Section with clicks on the sensor the Battery symbol appears in the deconz App and I geht the BatteryPercent in my fhem installation.

But you need to have it naturally after somes hours, when the device send the report, so not sure yet if it s something normal, will see on usage.

>One question I had left, why the Sensors not appaers in my phoscon App?

Phoscon is just an app that use the API, like FHEM, it s not a deconz part, so it use white/black list to display only somes sensors, but you can have the full list with checking the API in the phoscon beta (or future official version)

Iceman808, 2021-08-16:
Ok, here are an update to the previous Issue.
The Sensor with called "Humidity 28" sends battery level after some Hours autmaticlly. The other Sensor "Humidity 34" does not send any battery level until the last Cluster read by clicking with the Sensor Button.

Ist it possible that the Sensor only sends a battery level when it changed? But this not explained the different behaviour between both sensors.

Smanar, 2021-08-16:
>Ist it possible that the Sensor only sends a battery level when it changed? But this not explained the different behaviour between both sensors.

But even if the device send a notification about battery, you don't see it ? deconz send websocket notification only if the value change.

And yes, that's what I thought, if you need to read value yourself, it mean there is a problem.
Can you compare the attribute reporting on both device ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2196#issuecomment-695843587 But you probably need to wake up the device in same time you press "read config"

supersebbo, 2021-11-16:
I have got two of these sensors with same model number but different manufacturer. 
[image]

Can these be added?

quikfast, 2022-12-16:
I'm facing currently the same issue, I have an temperature sensor of Tuya on zigbee 3.0, the Deconz can read it, however from Phoscon I cannot detect the device. 

There is also no binding to the device.. 
Any tricks possible?

Smanar, 2022-12-16:
Hello https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support
And check if your device is not laready in PR, there are so much of them in waiting list.