dresden-elektronik/deconz-rest-plugin #4519: Some IKEA Tradfri Lights are unreachable and disabled

rene-mueller, 2021-03-04:
## Describe the question or issue you are having

Currently I setup my system for the whole household. So I have around 20 blubs. Randomly after a view days some of them are not reachable so the can't control via homebridge and Siri.

Is there an easier way than adding each bulb again?

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26690700)
 - deCONZ version: (2.09.03)
 - Device: RaspBee II

## deCONZ Logs

## Additional context

Mimiix, 2021-03-04:
Can you provide some logs?

msex67, 2021-03-06:
I have the same issue, after installing the latest homeassistant updates. 

[image]

closed 2021-04-04 by github-actions[bot]