dresden-elektronik/deconz-rest-plugin #4520: Lonsonho Tuya Zigbee 3 way switch (TS0043 / _TZ3000_a7ouggvs)

Big4SMK, 2021-03-04:
## Device

- Product name: Lonsonho Tuya Zigbee 3 way switch
- Manufacturer:_TZ3000_a7ouggvs
- Model identifier: TS0043
- Device type : Switch

Note that I already added the manufacturer in my [tuya_testing](https://github.com/Big4SMK/deconz-rest-plugin/tree/tuya_testing) to get to the below screenshots. However, I'm a bit at a loss on how to continue. From other threads I understood that I could use the API to see button presses, but per the below json the state returns null for the buttonevent.
Any help to move things forward would be helpful.

## Screenshots
[image]
[image]
[image]

`{
"config":{
"battery": 100,
"on": true,
"reachable": true
},
"ep": 1,
"etag": "4f50140a18929c74b4d9d4788763a07e",
"lastseen": "2021-03-04T21:44Z",
"manufacturername": "_TZ3000_a7ouggvs",
"mode": 1,
"modelid": "TS0043",
"name": "Switch",
"state":{
"buttonevent": null,
"lastupdated": "none"
},
"type": "ZHASwitch",
"uniqueid": "5c:02:72:ff:fe:d7:66:c3-01-0006"
}`

Smanar, 2021-03-05:
Your code seem good, but remember you need to edit the json file too.
This file is in /usr/share/deCONZ/device or something like that.

Big4SMK, 2021-03-05:
Oh thanks! Your comment made me realize that the json is not being used to create the shared object but needed to be copied as a separate file. It seems to be working fine now.

PR #4522 (2021-03-25): Add Lonsonho Tuya Zigbee 3 gang switch (TS0043 / _TZ3000_a7ouggvs)
Fixes #4520
closed 2021-03-25 by manup