dresden-elektronik/deconz-rest-plugin #4521: DS82-Tuya-Zigbee Curtain Drive

Kllrv, 2021-03-05:
This is supposedly a Zigbee 3.0 compliant curtain rail motor meant to be used with rails, released in 2020 or Jan 2021. 

https://www.aliexpress.com/item/1005001874380608.html?

## Device

- Product name: DS82-Tuya-Zigbee Curtain Drive
- Manufacturer: _TZE200_zpzndjez
- Model identifier: TS0601
- Device type :
  - Other: Cover

## Screenshots

Somewhat misidentified as a smart plug
[image]

basic
[image]

[image]

tuya specific

[image]

otau
[image]

Smanar, 2021-03-05:
Hello, do you have a linux machine to test code change ?

Kllrv, 2021-03-05:
Yes - with following considerations: I run DeconZ community container in Debian Buster. If testing is required and existing setup isn't sufficient, I can make another virtual machine.

Smanar, 2021-03-05:
Yes, I think it will be better, hard to do on docker, easier on VM.
You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you

```
sudo apt install deconz-dev
git clone --branch DS82 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

You will need 2 inclusions without deleting the device for integration.

I haven't tested the code on my side, so you can have error on compilation.

Kllrv, 2021-03-05:
Will report back on this

Kllrv, 2021-03-05:
It compiled, but:

```
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/deconz.service.d
           └─override.conf
   Active: activating (auto-restart) (Result: exit-code) since Fri 2021-03-05 23:12:56 EET; 25s ago
 Main PID: 9674 (code=exited, status=1/FAILURE)
      CPU: 4ms

Mar 05 23:12:56 Docker-Host-1 systemd[1]: deconz.service: Unit entered failed state.
Mar 05 23:12:56 Docker-Host-1 systemd[1]: deconz.service: Failed with result 'exit-code'.
```

Not sure if trying on a fresh(er) VM would help, I used an old one. This error is cryptic.

Smanar, 2021-03-05:
Will try on my side, but need some time.

Kllrv, 2021-03-05:
On a fresh install, it compiled and starts up properly after swapping the Conbee II to that VM.

Unfortunately the controls in Phoscon appear to do nothing, possibly due to an oversight on my part. Thus, I can't test whether the device connects properly now. I will attempt to add GUI to the VM during the weekend to have a look at it.

Smanar, 2021-03-05:
Hu, it will be more complicated ^^
You are not using third app ?

The device is probably not visible in phoscon, and tuya cluster is not usable in the GUI.

You need to use third app or directly the API.

Kllrv, 2021-03-06:
Okay, I had time to take this further. Got the API calls working, what am I supposed to request from the server?

Smanar, 2021-03-06:
For exemple with curl

`curl -H 'Content-Type: application/json' -X PUT -d '{"open": true}' http://IP:PORT/api/KEY/lights/ID/state`

IP and PORT : same than for phoscon
KEY is an api key
ID the device ID

To get an api key
`curl http://IP:PORT/api/ -X POST -d '{"devicetype": "TestKey"}'`

Kllrv, 2021-03-06:
requesting /lights/ID returns

`{"etag":"e1fb00eb1b21059124536a16abc0e42e","hascolor":false,"lastannounced":"2021-03-06T13:11:59Z","lastseen":"2021-03-06T13:12Z","manufacturername":"_TZE200_zpzndjez","modelid":"TS0601","name":"Smart plug 18","state":{"alert":"none","bri":0,"lift":0,"on":false,"open":false,"reachable":true},"swversion":null,"type":"Window covering device","uniqueid":"84:71:27:ff:fe:a9:79:35-01"}`

requesting /lights/ID/state returns

`[{"error":{"address":"/lights/18/state","description":"resource, /lights/18/state, not available","type":3}}]`

Kllrv, 2021-03-06:
Oh nevermind, using CURL and not REST client made something happen. The motor started moving. `[{"success":{"/lights/18/state/open":true}}]`

Kllrv, 2021-03-06:
I have tested 
open = true/false - these spin in different directions
stop = true - stops the motor
lift = 0 .. 100 spins in one way, maybe motor should be calibrated - I was unable to guess commands for this

Smanar, 2021-03-06:
Read the manual for calibration before, there is a command to trigger a zigbee calibration, but IDK what it do on the device.

`curl -H 'Content-Type: application/json' -X PUT -d '{"calibration": true}' http://IP:PORT/api/KEY/lights/ID`

To change side, it s possible with code, but try before

`curl -H 'Content-Type: application/json' -X PUT -d '{"reverse": true}' http://IP:PORT/api/KEY/lights/ID`

Kllrv, 2021-03-06:
Unfortunately the manual is for the wifi version and only tells how to reverse motor direction or attach remote (also puts to zigbee pairing mode). I have however learned the device should auto calibrate during first run.

Smanar, 2021-03-06:
>I was able to force auto calibration by long-pressing pairing key to reset position data, then starting the motor in each direction in sequence and stopping the motor shaft with pliers

You mean to calibrate the device, you have re-included it, and used it immediately ? I m realy interested by the method, lot of user ask for it.

>Motor now responds to "lift": 1..100 as expected, and opens/closes to positions "calibrated" by stopping the motor.

So all is working ? no need to reserve a side or a return ?
To be sure you need to have in the JSON
```
shutter closed
on= true
bri = 255
open = false
lift = 100%
```

or ALL values reversed.

Kllrv, 2021-03-06:
To reset calibration of the device, I simply long-pressed the include key. To put into pairing mode, one would triple-press and long press in sequence.

To reverse motor, it could be done with RF remote according to manual, or supposedly thru Zigbee command as described in zigbee2mqtt device info. https://www.zigbee2mqtt.io/devices/TS0601_cover.html

For `'{"reverse": true}' ` I get "parameter, reverse, not available"

Where do I see this JSON info?

Smanar, 2021-03-06:
To have the JSON just go `http://IP:PORT/api/KEY/lights/ID`

For the reverse it s the same command than for zigbee2mqtt, can you show the curl command you are using ? (no "state" inside)

Kllrv, 2021-03-06:
`{
  "etag": "a46345e8d9b96af1c390161477dabb82",
  "hascolor": false,
  "lastannounced": "2021-03-06T13:11:59Z",
  "lastseen": "2021-03-06T14:29Z",
  "manufacturername": "_TZE200_zpzndjez",
  "modelid": "TS0601",
  "name": "Smart plug 18",
  "state": {
    "alert": "none",
    "bri": 254,
    "lift": 100,
    "on": true,
    "open": false,
    "reachable": true
  },
  "swversion": "0xFFFFFFFF",
  "type": "Window covering device",
  "uniqueid": "84:71:27:ff:fe:a9:79:35-01"
}`

Something happened with the motor with this:

`curl -X PUT -i 'http://ip/api/key/lights/18/' --data '{"reverse": true}''`

The open/close commands still move it the same way, but it reports reversed state.

`{
  "etag": "22953385545fd4583067f2c403473610",
  "hascolor": false,
  "lastannounced": "2021-03-06T13:11:59Z",
  "lastseen": "2021-03-06T14:40Z",
  "manufacturername": "_TZE200_zpzndjez",
  "modelid": "TS0601",
  "name": "Smart plug 18",
  "state": {
    "alert": "none",
    "bri": 248,
    "lift": 98,
    "on": true,
    "open": true,
    "reachable": true
  },
  "swversion": "0xFFFFFFFF",
  "type": "Window covering device",
  "uniqueid": "84:71:27:ff:fe:a9:79:35-01"
}`

Smanar, 2021-03-06:
Ha ?

I think you have find a new issue.
>"lift": 98

For deconz if lift < 100 , the covering is not closed, It's because the covering is not full closed or because of the calibration ?
I think because of the calibration the min and max are different than 0 and 100 ?

>The open/close commands still move it the same way

Are you sure the command have worked (because for me the difference in JSON is not from that) ? I don't think it change something, but try with removing the last "/"

Kllrv, 2021-03-06:
It was by coincidence that the motor value changed. The motor moves a little bit back and forth when the reverse command is issued.  

I'm now using Mozilla RESTClient extension to make it easier.  All commands report success in response field.

Tests: 
`
PUT http://server/api/key/lights/18 { "reverse": false }`

```
{
  "etag": "182cd29636a160c481fd63c6598a9703",
  "hascolor": false,
  "lastannounced": "2021-03-06T13:11:59Z",
  "lastseen": "2021-03-06T18:00Z",
  "manufacturername": "_TZE200_zpzndjez",
  "modelid": "TS0601",
  "name": "Smart plug 18",
  "state": {
    "alert": "none",
    "bri": 254,
    "lift": 100,
    "on": false,
    "open": true,
    "reachable": true
  },
  "swversion": "0xFFFFFFFF",
  "type": "Window covering device",
  "uniqueid": "84:71:27:ff:fe:a9:79:35-01"
}
```
`
PUT http://server/api/key/lights/18/status { "open": true }`

```
{
  "etag": "7a443033d859e14d39b3f4ce530ad159",
  "hascolor": false,
  "lastannounced": "2021-03-06T13:11:59Z",
  "lastseen": "2021-03-06T18:03Z",
  "manufacturername": "_TZE200_zpzndjez",
  "modelid": "TS0601",
  "name": "Smart plug 18",
  "state": {
    "alert": "none",
    "bri": 254,
    "lift": 100,
    "on": false,
    "open": true,
    "reachable": true
  },
  "swversion": "0xFFFFFFFF",
  "type": "Window covering device",
  "uniqueid": "84:71:27:ff:fe:a9:79:35-01"
}
```

`
PUT http://server/api/key/lights/18 { "reverse": true }`

```
{
  "etag": "16ccdbee5d7062ccdd1b0ea9181525f4",
  "hascolor": false,
  "lastannounced": "2021-03-06T13:11:59Z",
  "lastseen": "2021-03-06T18:03Z",
  "manufacturername": "_TZE200_zpzndjez",
  "modelid": "TS0601",
  "name": "Smart plug 18",
  "state": {
    "alert": "none",
    "bri": 254,
    "lift": 100,
    "on": true,
    "open": false,
    "reachable": true
  },
  "swversion": "0xFFFFFFFF",
  "type": "Window covering device",
  "uniqueid": "84:71:27:ff:fe:a9:79:35-01"
}

```
`
PUT http://server/api/key/lights/18/status { "open": true }`

```
{
  "etag": "96e6b8bc9fcfc380382e7ac751da1e94",
  "hascolor": false,
  "lastannounced": "2021-03-06T13:11:59Z",
  "lastseen": "2021-03-06T18:03Z",
  "manufacturername": "_TZE200_zpzndjez",
  "modelid": "TS0601",
  "name": "Smart plug 18",
  "state": {
    "alert": "none",
    "bri": 254,
    "lift": 100,
    "on": false,
    "open": true,
    "reachable": true
  },
  "swversion": "0xFFFFFFFF",
  "type": "Window covering device",
  "uniqueid": "84:71:27:ff:fe:a9:79:35-01"
}
```

Note how the status was "not open" until issuing the open command. The motor didn't spin except for about a full rotation back and forth when attempting to apply reverse mode.

Smanar, 2021-03-06:
So need some cycle to stabilise the "machine" ?

But I don't see difference for { "open": true } with "reverse" = true or false.
It s perfect for me, but can you see a difference on the hardware ?

Kllrv, 2021-03-06:
I found the hardware didn't function in a different manner no matter the setting.

Smanar, 2021-03-06:
Ok, I have found a problem in the "reverse" command, can you try with the new code.

```
cd deconz-rest-plugin
git pull
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

No need to re-include the device, just close deconz before changing file, adn restart it.

Kllrv, 2021-03-06:
Tested multiple times, consistent results immediately after reaching position:
reverse = true - opens counter clockwise, closes clockwise. 0 when told to close. Lift 0 is closed, lift 100 is open.
reverse = false - opens counter clockwise, closes clockwise. 0 when told to close. Lift 0 is open, lift 100 is closed.

Issuing '{"open": true}'  always moves to position 100, and '{"open": false}'   always moves to position 0. These change the open status accordingly even if motor doesn't move, overriding the reverse setting behavior.

Smanar, 2021-03-07:
So the "reverse" command now have an impact ?
It reverse the "lift" value (so I think all other values) but have no impact on working mode ?

But how you can reverse the motor side ? We can't switch wire on this device.
Nothing on the manual ? perhaps need to calibrate after reverse ?

>Issuing '{"open": true}' always moves to position 100, and '{"open": false}' always moves to position 0

So if the "reverse" command switch lift position, it mean in a situation 100 is up and the other 100 is down ?

Kllrv, 2021-03-07:
> So if the "reverse" command switch lift position, it mean in a situation 100 is up and the other 100 is down ?

Yes, with current behavior this appears to be the case.

> But how you can reverse the motor side ? We can't switch wire on this device.

Attaching double curtain to different sides of the rail. Single curtain, attaching it on the other side. Movement direction would then change. Because doing this is possible, reversing shouldn't be neccessary in a real world scenario so it may only exist for convenience, or because this might be shared firmware for roller curtains - those have a valid use case for a mirrored installation.

Smanar, 2021-03-07:
Ha ok.

So it s device dependant, for your device, to reverse side, we need
- change the motor position
- use the command

so the motor still run in the same side, but the automation have the return reversed.

Thx  a lot for the "reverse" issue, I realy have skipped this bug, can explain why there is so much user with side direction problem ....

And BTW, so your device works correctly finaly ? nothing to change ?

Kllrv, 2021-03-07:
Didn't find any functional issues with the device last night.

Btw, when manually pulling the curtain (or rotating the motor without anything attached), it will move to either min or max point, and report the state change once the move completes.

Smanar, 2021-03-07:
>Btw, when manually pulling the curtain (or rotating the motor without anything attached), it will move to either min or max point, and report the state change once the move completes.

Nice feature for me, can make the motor move without using remote, just with moving a little the covering ?

Pr done

Kllrv, 2021-03-07:
> Nice feature for me, can make the motor move without using remote, just with moving a little the covering 

Yes, feels very premium. Or would, with the real rail instead of 3D printed adapters...

Smanar, 2021-03-07:
Ha ?
So the motor seem robust but the rail cheaper ?

Kllrv, 2021-03-07:
> So the motor seem robust but the rail cheaper ?

Rather, I had to DIY a rail because it was hard to find the real one without outrageous shipping cost.

Smanar, 2021-03-07:
Ha ok ^^

marithpl, 2021-03-12:
Guys what you did to have it working? My motor is not a curtain but for blinds/shutters. Model it's the same.
Is it possible to manually change a type from `Smart Plug` to `Window Covering Motor`?

```
{
"etag": "9a19cee1d0f2ca83d8f3e7442b9301f3",
"hascolor": false,
"lastannounced": null,
"lastseen": "2021-03-12T09:22Z",
"manufacturername": "_TZE200_nueqqe6k",
"modelid": "TS0601",
"name": "Smart plug 9",
-"state": {
"alert": "none",
"on": false,
"reachable": true
},
"swversion": null,
"type": "Smart plug",
"uniqueid": "60:a4:23:ff:fe:ec:49:ac-01"
},
```

Kllrv, 2021-03-12:
>Guys what you did to have it working

Your manufacturer name is different. @Smanar could perhaps add that, I'm certain it has the exact same Tuya firmware. This however wouldn't make it to the beta on 15th this month according to release cycle.

marithpl, 2021-03-12:
Should I make a new request for support this device?

Smanar, 2021-03-12:
Have added your device here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4529

If you can test this PR ?

marithpl, 2021-03-12:
Sure if you can tell me how :)

Smanar, 2021-03-12:
You need a linux machine (not docker)

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you

```
sudo apt install deconz-dev
git clone --branch ds82 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

micke, 2021-03-12:
I've built a docker container with the mentioned PR changes, available at `lisinge/deconz:ds82` if needed.

imammedo, 2021-03-17:
Hi Smanar,

I just got DS82, and build/installed your DS82 branch.
However it isn't recognized as "Window Covering Device", but as "Smart Plug"
`
{
    "etag": "f444a3e28e92d55b89c0a51820974899",
    "hascolor": false,
    "lastannounced": "2021-03-17T12:42:28Z",
    "lastseen": "2021-03-17T12:52Z",
    "manufacturername": "_TZE200_zpzndjez",
    "modelid": "TS0601",
    "name": "Smart plug 29",
    "state": {
        "alert": "none",
        "on": true,
        "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "84:71:27:ff:fe:a9:7c:a8-01"
}

I have tried to include it 2 times and more, it still insist on being "Smart Plug"

is there anything I can try to find out where is problem?

PS: other notes
  * it turns motor on only in one direction when I set state { 'on': false } and turns it off on  'true'
  * 'reverse' makes motor go back and forth a turn, but it doesn't affect direction
  * sending status { "open": true}, errors out with '404 Not Found'

Smanar, 2021-03-17:
You haven't deleted it between the 2 inclusions ?

- Try to set phoscon in permit join (add new light)
- Read again the basic attribute in deconz, Cluster 0x0000 button read.

imammedo, 2021-03-17:
I haven't deleted it (just did rejoin) using (add new light) and join/reset squence on motor (i.e.  3 short presses + 1 long till light flashes).

I also reread basic cluster, it returns exactly the same values as in the first post screenshots.
Is there some tracing/debugging to turn on to see where it goes off expected code path?

btw: I use 2.10.3 + plugin built from your DS82 branch

Smanar, 2021-03-17:
Yep, you can try to take a look in deconz log with "info_l2", at start you will see line starting by "Tuya debug 7"
If I m right you will see the code finding the manufacture name.

imammedo, 2021-03-17:
It shows as a router (if yellow node is router).

log has:
22:21:43:189 Tuya debug 7 : Missing manufacture name for 0x847127fffea97ca8

full log for device: (egrep -i "tuya|a97ca8|a9:7c:a8")
 https://paste.mozilla.org/rNsy2pER

imammedo, 2021-03-18:
sorry, set log expiration to only 1hr, here is updated link for a week http://paste.mozilla.org/xPkOizZF

PS:
 it seems that device always connects via a router so reading attributes doesn't happen fast.

marithpl, 2021-03-18:
> You need a linux machine (not docker)
> 
> You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
> So for you
> 
> ```
> sudo apt install deconz-dev
> git clone --branch ds82 https://github.com/Smanar/deconz-rest-plugin.git
> cd deconz-rest-plugin
> qmake && make
> sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
> ```
> 
> You probably need to include the device 2 time, without deleting it, the first time will be included as plug, the second one as covering.

I have already a linux device.
Can I unplug Conbee from NAS and use for test on other device and after that connect again to NA?
I don't want to test in on my "production" and lost current sensors. I'm not sure if any connections are stored on Conbee or is just an antena?

Smanar, 2021-03-18:
@marithpl the network setting is stored on the gateway, but the device data are stored on the host, so you can move the conbee 10 mn for some test if you want.

@imammedo all seem fine, I have added some debug line if you can remake the same log but with the new code ? (witn "info" and "info_l2", have make a mistake)

```
cd deconz-rest-plugin
git pull
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

imammedo, 2021-03-18:
@Smanar

Apparently I was not able to copy built plugin  properly the first time
It seems to be working now i.e. it detects as window covering device.
Sorry for noise, I'll continue with testing it.

imammedo, 2021-03-18:
Confirming Kllrv finding about 'reserse' command has no effect.
However one can reserve motor by pressing button on it, 2 short 1 long press, did the trick.
Also his method to put motor in calibration mode (1 long press) also worked for me.

Kllrv, 2021-03-19:
To reverse direction with motor button, one short press and one long press did it for me. Works the same as the command, as described above.

Smanar, 2021-03-19:
>Confirming Kllrv finding about 'reserse' command has no effect.

Can you share logs ("info") when using the command, with the last code.

And BTW I have find why it can be hard to include the device some time >https://github.com/Smanar/deconz-rest-plugin/commit/31eb4b6995139cb842ef4aa65296021df48e0689

Kllrv, 2021-03-19:
>Can you share logs ("info") when using the command, with the last code.

Unable to compile the changes I pulled just now:

```
{standard input}: Assembler messages:
{standard input}:101184: Warning: end of file not at end of a line; newline inserted
{standard input}:102846: Error: unknown pseudo-op: `.lls'
{standard input}:102845: Error: invalid operands (*UND* and *UND* sections) for `-'
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
```

Smanar, 2021-03-19:
Just now ? you are trying wich one branch ? The tuya_14 ? or the DS82 ?

Kllrv, 2021-03-19:
DS82. Did you want me to try the other one?

Smanar, 2021-03-19:
Depend if the device is reconized or not ?
If the device is not reconized I need to make a branch with both mixed for you (the tuya_14 improve reconization, but not support DS82)

Kllrv, 2021-03-19:
In the DeconZ UI, it's appearing as the wrong device type - but otherwise functions as expected through API. Not sure if this would be fixed by removing and re-adding, not eager to break a working setup right now.

imammedo, 2021-03-19:
cherry-picked 31eb4b6 into DS28 branch,
* It seems that it finds motor from the first time (tried 3 times).
   In web gui it appears as light and named as 'smart plug' (as it is without that commit), query via REST API returns Window covering.
* 'reverse' command does have effect but only in one direction, more precisely motor does about a turn back&forward 'and' direction reverts to the factory default, regardless of true/false value or how many times I repeat command
* 5 short, 1 long (till constant light comes up) presses on button toggles direction on both of my motors

>Can you share logs ("info") when using the command, with the last code

I captured log, what I should grep from it of interest?

Smanar, 2021-03-20:
>     * It seems that it finds motor from the first time (tried 3 times).

So perhpas a solution ...

>       In web gui it appears as light and named as 'smart plug' (as it is without that commit), query via REST API returns Window covering.

Yep, this is normal, the device present itself as a smart plug, but I can cheat for the API

>     * 'reverse' command does have effect but only in one direction, more precisely motor does about a turn back&forward 'and' direction reverts to the factory default, regardless of true/false value or how many times I repeat command
>       
>       * PS: 'reverse' didn't have 'any' effect on new motor, only after I did manual reverse it started to react.

I don't know exactly how work the reverse command, I just mimic the command send by the gateway, the last users that have tested it have said it NOT change the motor direction, but the state return when it reach a position.
 
> I captured log, what I should grep from it of interest?

Yep the log starting by "tuya debug" when  you have do the reverse command pls.

[14 more replies]

closed 2021-03-27 by manup