dresden-elektronik/deconz-rest-plugin #4523: Manufacturer name and model (_TZ3000_rdtixbnu and TS0121)

fernandodelucca, 2021-03-05:
## Describe the question or issue you are having

This device of mine appears as that name _TZ3000_rdtixbnu, how do I make the name as friendly as it appears other devices on my network? 
#4457 

## Screenshots

[image]

## Environment

 - Host system: Ubuntu 20.04.2
 - Running method: Virtual Machine
 - Firmware version: 26680700
 - deCONZ version: 2.10.02
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Smanar, 2021-03-05:
Hello
_TZ3000_rdtixbnu is not the name, but the manufacture name, the name is "smartplug010" and you can edit it using the "pen" just close the name

fernandodelucca, 2021-03-05:
Hello Smanar,
Can I change the name of the manufacturer to be friendly like the others? 
[image]

Smanar, 2021-03-05:
Ha, no ^^, it s a long story, but no, for tuya this "no friendly" name is realy important to reconize the device, but all device with this kind of name will be tuya device.

fernandodelucca, 2021-03-05:
LoL, thanks for your answer, it was clear to me. I didn’t understand why some had friendly names and others didn’t, but I imagine it’s because there are particularities between Router and End Devices

[image]

Smanar, 2021-03-05:
No, it something possible, we can change this name direclty in the api, or keep it in the API but change it before print it, it happen for some tuya device from lidl.
But the problem is tuya device have same model id, so we need the manufacture name to reconize the device, and this is for all tuya device, router and end device.

fernandodelucca, 2021-03-05:
Really cool, does that make Tuya devices worse than others? just to know if I should keep buying from them.. lol

Smanar, 2021-03-06:
They are more exotic, there is some of one of them with some issues (not always their fault, it s new stuff), but I have generaly good return from them.
They are exotic because they are using a special cluster, but for me more zigbee compliant than some device using standard cluster.

For me better stuff than xiaomi, less mesh problem. There is som much tuya device, you can find the better like the worse.

Having a friendly name displayed, is something possible, the code will be able to use a product name soon, the application just need to display other thing than the manufacture name.
But not for the moment, and not possible to change it in the API.

fernandodelucca, 2021-03-07:
perfect the explanation @Smanar, thank you very much!

closed 2021-03-09 by fernandodelucca