dresden-elektronik/deconz-rest-plugin #4524: 4 button remote Nedis ZBRC10WT R7054

tunam007, 2021-03-05:
Hello,

I bought this Nedis remote control but I can't use it with deconz   2.10.2 . The other Zigbee Nedis products work well and are recognised via Jeedom if ever.

Device

    Product name: Nedis ZBRC10WT
    Manufacturer: TZ3000_fsiepnrh
    Model identifier: TS0215A
    Device type : Remote
    

Screenshots

[image]

Node Info
[image]

basic
[image]

Power Config
[image]

IAS Zone
[image]

[image]

Thank you 

Greg

Mimiix, 2021-03-08:
Can you make sure the basic cluster is read?

tunam007, 2021-03-08:
Hello,
Could you tell me how to do this? 

Thank you 

Greg

Mimiix, 2021-03-09:
Click "read" on the right top :)

tunam007, 2021-03-18:
Hello,

Sorry I had a lot of work, I'm not used to the software. I use deconz with Jeedom. Here is what I have read:

[image]

Tell me if I can help you more? I am in Belgium, I can even send it to you and you can send it back to me when you are done.

Thanks 

Greg

tunam007, 2021-03-31:
Hello,

I was able to read one or two more values

[image]

[image]

tunam007, 2021-03-31:
I was also able to extract this from the Zigbee plugin on Jeedom:

{
    "ieee": "bc:33:ac:ff:fe:5d:2a:bc",
    "nwk": 24323,
    "status": 2,
    "lqi": "None",
    "rssi": "None",
    "last_seen": "None",
    "node_descriptor": "02:40:80:02:10:52:52:00:00:2c:52:00:00",
    "endpoints": [
        {
            "id": 1,
            "status": 1,
            "device_type": 1025,
            "profile_id": 260,
            "manufacturer": "_TZ3000_fsiepnrh",
            "model": "TS0215A",
            "output_clusters": [
                {
                    "id": 25,
                    "name": "Ota",
                    "attributes": []
                },
                {
                    "id": 10,
                    "name": "Time",
                    "attributes": []
                }
            ],
            "input_clusters": [
                {
                    "id": 0,
                    "name": "Basic",
                    "attributes": [
                        {
                            "id": 0,
                            "name": "zcl_version",
                            "value": 3
                        },
                        {
                            "id": 1,
                            "name": "app_version",
                            "value": 67
                        },
                        {
                            "id": 2,
                            "name": "stack_version",
                            "value": 0
                        },
                        {
                            "id": 3,
                            "name": "hw_version",
                            "value": 1
                        },
                        {
                            "id": 4,
                            "name": "manufacturer",
                            "value": "_TZ3000_fsiepnrh"
                        },
                        {
                            "id": 5,
                            "name": "model",
                            "value": "TS0215A"
                        },
                        {
                            "id": 6,
                            "name": "date_code",
                            "value": ""
                        },
                        {
                            "id": 7,
                            "name": "power_source",
                            "value": 3
                        }
                    ]
                },
                {
                    "id": 1,
                    "name": "Power Configuration",
                    "attributes": []
                },
                {
                    "id": 1280,
                    "name": "IAS Zone",
                    "attributes": []
                },
                {
                    "id": 1281,
                    "name": "IAS Ancillary Control Equipment",
                    "attributes": []
                }
            ]
        }
    ],
    "signature": {
        "manufacturer": "_TZ3000_fsiepnrh",
        "model": "TS0215A",
        "node_desc": {
            "byte1": 2,
            "byte2": 64,
            "mac_capability_flags": 128,
            "manufacturer_code": 4098,
            "maximum_buffer_size": 82,
            "maximum_incoming_transfer_size": 82,
            "server_mask": 11264,
            "maximum_outgoing_transfer_size": 82,
            "descriptor_capability_field": 0
        },
        "endpoints": {
            "1": {
                "profile_id": 260,
                "device_type": 1025,
                "input_clusters": [
                    0,
                    1,
                    1280,
                    1281
                ],
                "output_clusters": [
                    25,
                    10
                ]
            }
        }
    },
    "class": "zigpy.device"
}

tunam007, 2021-04-22:
:(

closed 2021-05-21 by github-actions[bot]

jhubig, 2022-04-09:
@tunam007 Could you make it finally work? I'm looking since long time for a "keyfob" working with deconz. This (https://zigbee.blakadder.com/Immax_07046L.html) would be my preferred one and it should be compatible but I cannot find it anywhere to buy it. Also happy to hear about your alternatives.

jordanderoubaix, 2022-10-28:
Hello,
I would be very happy to know if it can be paired with deconz also :).
I try using the DDF from R7052 without luck