dresden-elektronik/deconz-rest-plugin #4525: Raspbee 2 firmware not connected

murphwiz, 2021-03-06:
## Describe the bug
 unable to connect or update firmware.

## Steps to reproduce the behavior
 update using SSH fails

## Expected behavior
 

## Screenshots

## Environment

 - Host system: Raspberry Pi 4b
 - Running method: Homebridge raspbian image add on
 - Firmware version: (not connected)
 - deCONZ version: (2.09.03)
 - Device: (RaspBee II)
 - Do you use an USB extension cable: (/no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?
No, running successfully: homebridge and pi hole.

## deCONZ Logs
 https://pastebin.com/FRR6JJxi

## Additional context

Mimiix, 2021-03-08:
Hi!

Can we close this as i think we fixed it trough discord already. Can you let me know?

murphwiz, 2021-03-08:
Yes, please close, thank you for your help!

closed 2021-03-08 by murphwiz