dresden-elektronik/deconz-rest-plugin #4526: Linkind 5-Key Remote Control - ZBT-RGBWSwitch-D0801 - ZS230002-IE-V1A02

IgnacioHR, 2021-03-06:
## Device

- Product name: Linkind 5-Key Remote Control
- Manufacturer: lk
- Model identifier: ZBT-RGBWSwitch-D0801
- Device type :  
  - Switch

## Screenshots

[image]

[image]

Diagnostics (might not be needed)
[image]

Picture
[image]

Mimiix, 2021-03-08:
Can you provide all screenshots?

IgnacioHR, 2021-03-08:
Hello, thank you for the quick response. 

Sure, I've been playing with the tools over the weekend and now I'm becoming more familiar with them

[image]

[image]

[image]

[image]

[image]

Hope this is all you need

I'm actually a developer (in a completely different area) so I'm familiar with xml, json, yaml, html, js, ... and more than other 100 four letters acronyms so feel free to give me technical details when needed.

IgnacioHR, 2021-03-30:
HI, I've been notified about lack of activity regarding this topic and I would like to revive it.

I would like to update the screen captures because I learned that the information changes when I click on "Read" more than once in deCONZ gui

[image]

[image]

Let me know if you have enough information to move this forward

IgnacioHR, 2021-03-30:
BTW,

In Home Assistant, when I subscribe to deconz_event, I see messages for other devices but this is NOT sending status changes. I also notice there is a page with code FD01 that is named "unknown" and maybe the information is there and in a proprietary extension?

IgnacioHR, 2021-04-23:
Still no progress, what else do you need?
Thanks!

Mimiix, 2021-04-23:
A Dev with time/willingness to integrate it :) Other than that there's nothing we can do.

IgnacioHR, 2021-04-23:
Great news!

I'm a developer and I would like to use it so I night help. But at this time I ignore the terminology as I'm not involved in this specific field. So let me describe, using my own words, what is the first and second goal, maybe someone could help me moving forward.

The first goal would be to allow Home Assistant to receive events when I press the buttons in the device.

The second goal would be to allow Phoscon Gateway to use the device and show a representation of it etc.

I'm happy if someone helps me trying to achieve goal 1.

At this time, I've

- RPI-4 with deCONZ installed and running.
- The device is visible in the graphical network on the right.
- With the "Cluster Info" panel active on the left, I click on the different rows in the device and I see rows groping information that belongs to the same "group"
- One of the rows has code "FD01", name is "unknown" and contains no records
- While pressing the buttons in the device I see no changes

What I guess is the next step is to figure out what is behind the FD01 "cluster info" page and hopefully there will be registers with that information.

I don't know if that information can be obtained, from where, and also I don't know the files where deCONZ stores that information.

I'm doing this for pleasure and hobby so I'll be spending some time over the weekend but any help will be appreciated.

Cheers
Ignacio

Smanar, 2021-04-23:
Can hope the cluster 0fFD01 is not used by the device for the moment
Something you can do is mimic the device "ZBT-CCTSwitch-D0001" on the code
Vendor is too VENDOR_LDS
Mac prefix is too silabs2MacPrefix

All the rest can be same

To compile code you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

The only thing you can miss is the color control (can use the exotic cluster) but will be visible on log if it s the case.

closed 2021-05-22 by github-actions[bot]