dresden-elektronik/deconz-rest-plugin #4527: Deleted devices still exist insivibly and are provided to Home Assistant deCONZ integration

bcutter, 2021-03-06:
## Describe the bug
I deleted formerly used devices from Phoscon/deCONZ. They do neither appear on the sensor/lights/switches lists (Phoscon) nor on the node list (deCONZ) anymore.

BUT: deCONZ (running in a Supervisor/HASS addon) still exports those devices to the Home Assistant deCONZ integration (https://www.home-assistant.io/integrations/deconz). Using the `deconz.remove_orphaned_entries` service does not remove those devices. Reloading the integration always (re)imports those deleted devices. Rrestarting deCONZ and Home Assistant does not change anything.

This happened for two "lumi.plug.mmeu01" devices so far (only ones being removed from the ZigBee network).

## Expected behavior
Deleted devices should not be provided anymore.

## Screenshots

[image]

## Environment

[image]

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: see screenshot
 - deCONZ version: see screenshot
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?: no

## deCONZ Logs

## Additional context

Questions:
1) How to check the internal devices database of deCONZ?
2) According to 1: how to completely delete/remove devices from deCONZ?

Note: I´m not sure where this issue belongs to (deCONZ side? deCONZ integration for Home Assistant side?). Seeing deleted devices still being exported to the deCONZ HA integration I assume the bug is on the deCONZ side.

bcutter, 2021-03-06:
Update:
Using https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/#get-all-sensors I discovered 8 sensor devices matching the one I had deleted before (`"modelid": "lumi.plug.mmeu01"`).

1) I deleted them using https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/#delete-sensor.
2) I ran `deconz.remove_orphaned_entries` service in Home Assistant.
3) I reloaded the deCONZ integration in Home Assistant.

So I "worked around" that issue, fine _for the moment_. Remaining question:

**Why do sensors still exist after they´ve been already deleted?**

Is the deletion / cleanup not done properly?

I keep this issue open until those questions have been answered - because I fear ending up in the same situation soon.

Smanar, 2021-03-06:
>I deleted formerly used devices from Phoscon/deCONZ. They do neither appear on the sensor/lights/switches lists (Phoscon) nor on the node list (deCONZ) anymore.

Hello, you have used phoscon or deconz ?

bcutter, 2021-03-06:
> 
> 
> > I deleted formerly used devices from Phoscon/deCONZ. They do neither appear on the sensor/lights/switches lists (Phoscon) nor on the node list (deCONZ) anymore.
> 
> Hello, you have used phoscon or deconz ?

Both. Usually Phoscon, sometimes deCONZ. When deleting it immediately in the other application (Phoscon/deCONZ) devices were removed from the GUI.

I tested it after having deleted all sensors using the API as mentioned few minutes before:
I added/paired one of those devices again and deleted it (using Phoscon). This time they also were removed from the internal deCONZ database (or whatever the storage is called) and were therefore not exported to deCONZ integration for HA.

Maybe this happens when deleting disconnected devices? Or it´s another cause, I have no idea.

Smanar, 2021-03-06:
From my memory this can happen if you delete the device in deconz instead of phoscon, but I can't find again the explanation.

closed 2021-04-04 by github-actions[bot]

mic4ael, 2023-02-01:
Any update on this since I am also curious why deletion of entities doesn't propagate to HA.

Rob65, 2023-09-13:
This still is an issue on HA 2023.09 with Deconz 6.20.0 addon.
This should be reopened and fixed