dresden-elektronik/deconz-rest-plugin #4528: 2.09.03-raspbian-buster-stable hangs on start

erikproper, 2021-03-06:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2021-03-06:
Hello, can you share at least some logs ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags

erikproper, 2021-03-06:
After about 15 minutes (...) it is up again.
Need it to function now (getting dark, so need control of the lights).

If I have time in the next days, I'll do a restart with the debug flags
added.

Smanar, 2021-03-06:
Can be a database problem (if you have a backup to try)
Or a device (if you can power off 50% of your device to make test)

Mimiix, 2021-03-06:
Closing this issue. This is not how ti works.

Mimiix, 2021-03-06:
@Smanar please withhold from replying to incomplete issues. Costs way to much time..

closed 2021-03-06 by Mimiix

erikproper, 2021-03-06:
Sorry, but maybe it should be clearer, and easier, to obtain log files.
Especially for users ...

Thanks to @Smanar i now know what to do next time! Getting log files ‘Costs
way too much time’ if it is not clear how to obtain these.

Mimiix, 2021-03-06:
> Sorry, but maybe it should be clearer, and easier, to obtain log files. Especially for users ... Thanks to @Smanar i now know what to do next time! Getting log files ‘Costs way too much time’ if it is not clear how to obtain these.
> -- Sent from my mobile device.

It is really easy. In deconz, click help and click log viewer. 

The template however is even more easy but you refused to fill in the basics.

erikproper, 2021-03-06:
The basics were not available, as the information regarding the log files
were not available ... as such, the template should contain the pointers on
how to get access to the logs. So, no ‘refusal’ on my side. Just lack of
data ...

Furthermore, the logs I can find using the GUI does not report much about
the first 15 minutes after the reboot while it was ‘stuck’

Note: will certainly do a manual restart tomorrow with commandline logging
turnend on. However as it is a production system, I need it in a working
state (which it is at the moment) now, since we need the lights ...

Mimiix, 2021-03-07:
To be fair on the logs part : the template clearly instructs how to obtain them.

Debug logs can help to investigate certain problems like device pairing and communication issues.
  Logging can be enabled on the command line as described in:
  https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags
  Secondly, for versions above 2.05.87: In the deCONZ GUI: click Help > debug view. Output of APS,ZDP, ZCL and INFO can be handy. 

Also: we want them to rule stuff out. You not seeing anything useful, doesn't mean we don't see anything useful either.

erikproper, 2021-03-07:
Based on: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/55 now doing a reboot with a modified deconz-gui.service (see attached).  

This generates the attached log file (recorded over the last 20 minutes ... still waiting for the GUI and/or webinterface to come up). I've also included a (Gzipped) copy of my zll.db

I do see a lot of repeating entries in the log file.  For instance:
$ grep 00:17:88:01:08:64:c5:27-02-0402 deCONZ-error.log  | wc -l
results in more than 2500 entries ...

[deCONZ-error.log.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6097778/deCONZ-error.log.gz)

[zll.db.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6097762/zll.db.gz)

[deconz-gui.service.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6097752/deconz-gui.service.txt)