dresden-elektronik/deconz-rest-plugin #4532: Apex smart plug - adjust reporting

svenove, 2021-03-07:
## Describe the question or issue you are having

In regards to #3943.
The plug is working great, but I'm struggeling to adjust the reporting intervals.

See screenshot no 1.
These are the default values. I've tried to change them, but it doesn't seem to have any effect. I am able to read the new configuration back, so it seems the config is indeed saved. But, after some hours/days (or perhaps a deCONZ restart?) the values are set back to default again.

I've been in contact with @haarvig in a Norwegian forum and he's confirmed with their developers that this should work.
He also uses the plug in their own hub, and the settings seems to work fine there.
I noticed that in their hub, the default values are different, but that could be the hub is set to change them on inclusion? Or is there some datatype-issue with deCONZ that makes the default values to read different?
Screenshot no 2 is a screenshot from their own hub.

## Screenshots

[image]

[image]
## Environment

 - Host system: PC
 - Running method: Virtual machine running Ubuntu 16.04 LTS
 - Firmware version: 26390500
 - deCONZ version:  2.10.01
 - Device: ConBee I
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? If so: Which? RFXtrx433E

## deCONZ Logs

## Additional context

About 15 IKEA-bulbs and 25 Xiaomi sensors. 2 Philips Hue light strips.

SwoopX, 2021-03-07:
Well, that indeed feels a bit odd. So firstly. what you see in your first screenshot is deconz' default configuration for smart plugs. As you already discovered, the settings are generally freely configurable.

Now, let me give you a better understanding how deconz handles this internally. During pairing, deconz typically creates the required sensors, does the cluster bindings (simplified, which cluster should talk) as well as the corresponding attribute reportings. Additionally, there's a safeguard which checks if reports are overdue and if they are, binding and reporting is re-initiated. If I recall correctly, this mechanism also kicks in when the device is rebooted due to whatever reason or deconz is restarted (as the settings are kept in memory).

Presumably, one of the last mentioned curcumstances occurred, so this could explain the "reset to default values". Interestingly, I never had any noticable issues with that. I prefer not to collect current and voltage from all my plugs and therefore disabled reporting, and that seemed to have survived all deconz restarts so far.

I might have overlooked some details here and the code is a bit scattered. For mid term, it is planned to also include reporting configuration in device specific configuration files to better fit that to ones needs. Adjusting the configuration in the meantime to the Datek defaults also does not seem the proper solution here, as it has potential to make other people unhappy, I presume.

SwoopX, 2021-03-14:
Would you be able to compile the plugin? There might be something to try out.

svenove, 2021-03-17:
Sorry for the delay, I believe I compiled the plugin myself a while ago, so I guess I could try doing it again? :)

SwoopX, 2021-04-13:
Well, I guess I figured something out which should at least ease the situation a bit, so you might want to try that out once I got the PR ready. Please not that I've just tested this in my dev network so far and that needs to be discussed with the other devs.

So what I cannot change is the initial reporting configuration upon deconz start. That would simply go too deep. However, I brought that topic up to be considered in a huge refactoring of the code to be considered.

Now, what I did is streamlining the process throughout the resource types (lights and sensors) to behave equally. That implies to fall pack to polling once a report from the measurement sensors has been missed and presumably causing the "reconfiguration" of the reporting in your case. That worked fine and reliable so far for a smart plug I have which doesn't access any reporting configuration.

As indicated, the corresponding PR needs to be set up and get the others buy in.. Regardless, once the PR is available, you can check it out.

sepodele, 2021-05-05:
Bumpetibump.

closed 2021-06-04 by github-actions[bot]