dresden-elektronik/deconz-rest-plugin #4534: Device name

jonmoerk, 2021-03-07:
LK FUGA® Wiser wireless battery 4 button switch
https://www.lk.dk/produkter?iid=13901

## Device

- Product name: 
LK FUGA® Wiser wireless battery 4 button switch
- Manufacturer: Schneider Electric
- Model identifier: FLS/AIRLINK/4
- Device type : 
  - Switch

## Screenshots
[image]

[image]

Mimiix, 2021-03-07:
Missing clusters screenshots

thiesens, 2021-03-08:
[image]
[image]
[image]
[image]

tomillr, 2021-03-16:
This products seems to be related to the Zigbee-based Wiser-range bei Merten. There's an older request for support that has yet to be addressed:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1615

mwaldhauer, 2021-03-22:
What is eta on adding device to Phoscon ?
Exited to add new gadgets 😁

scaarup, 2021-03-25:
> This products seems to be related to the Zigbee-based Wiser-range bei Merten. There's an older request for support that has yet to be addressed:
> 
> #1615

That request is alsost 2 years old. Wiser (Schneider) recently released a series of new devices in Denmark and we are a lot of users eager to get this working.
What's the eta and what can we do, as users, to help?

akafester, 2021-03-30:
It seems that Zigbee2mqtt has some support for the Merten line. But it overall seems to be a pretty unsupported product line. 
Looks forward to see some support for this!

palletodap, 2021-04-05:
This would be really great to get supported in Deconz, especially after the Danish marked has a huge focus on these new FUGA Wiser products right now😄 These are going to be the new standard in new smart-homes build in Denmark, since that's the only brand used for traditional switches here😅

Cubixzno1, 2021-04-15:
Any news on this topic? Would be greatly appreciated!

teslafreak1, 2021-04-17:
Any news? Really waiting for this to be integrated

scaarup, 2021-05-10:
This is a valid request and should not auto expire!

Noizexx, 2021-05-10:
ZHA has this somewhat working if you need pointers.

Noizexx, 2021-05-10:
> > ZHA has this somewhat working if you need pointers.
> 
> No it is not working well in ZHA.

Working somewhat. With some fiddling, it's fully functional.

scaarup, 2021-05-10:
Some fiddling? I couldn't detect any key-presses when listening for events. But yes, you can turn on/off, dimm up/down a light - that works. But I would like to do other automations based on pressing a key on the wiser.

Cubixzno1, 2021-05-10:
Is it possible to use the battery switch only or do I still need the other parts?

JonasKristiansen, 2021-05-21:
Also very eager to get the Wiser line working with Deconz.

Just purchased a motion sensor, but seems to have no luck getting it connected to Deconz.

mwaldhauer, 2021-06-03:
@SwoopX I see you have helped with the LK Wiser Temp and Water leak sensors. Can you help here as well ?

EjvindHald, 2021-06-23:
@SwoopX - if you have time to assist with this one, I would highly appreciate it as well. Thanks in advance.

teslafreak1, 2021-07-15:
Please keep this alive.

EjvindHald, 2021-07-15:
Schneider has provided this hardware directly to Dresden Electronics, who are now working on proper integration in Phoscon of this device.

EjvindHald, 2021-08-08:
This is still relevant, and Dresden Electronics has received the hardware.

mwaldhauer, 2021-08-10:
Is there an ETA on the integration ?

Mimiix, 2021-08-10:
Not sure. A dev needs to pick it up. If DE has the device, they need to pick it up. 

I'll ask around :)

perevers, 2021-08-23:
Looking forward to this working :)

EjvindHald, 2021-09-14:
It is expected to be included in next version 2.13 according to DE.

Cubixzno1, 2021-09-14:
That sounds Amazing!

perevers, 2021-09-16:
That sounds amazing. Any ETA on version 2.13?

EjvindHald, 2021-11-11:
It is still in progres...

BOB-DK, 2021-11-25:
@EjvindHald 
Do you know what the status is at the moment?

EjvindHald, 2021-11-26:
Schneider has provided a sample of many Wiser devices to DE, and I have tested these in beta version.

However, my mistake the 4 button battery device was not included. DE has received that device now, and they will support it in coming release. I do not when, as DE has been impacted by Covid-19.

BOB-DK, 2021-11-27:
@EjvindHald 
Thanks for the answer.
Hope DE recovers quickly.
Looking forward to the upcoming release.

mwaldhauer, 2021-12-19:
Still not fixed

EjvindHald, 2022-01-07:
I tested it on version 2.13.04 and the device cannot be adopted. The result of this test was given to DE support, and today they replied that it has been handed over the development team.

jh83, 2022-01-16:
Any updates on this?

EjvindHald, 2022-01-16:
I suggest you contact Dresden Electronics to get latest status on this.

Mimiix, 2022-01-16:
I ask manup to check

manup, 2022-01-16:
The Wiser devices will be checked soon, we have a full set of products now for testing and integration.
Plan is to get them all working cleanly with DDF :)

jh83, 2022-01-17:
> The Wiser devices will be checked soon, we have a full set of products now for testing and integration. Plan is to get them all working cleanly with DDF :)

Sounds great. Does that "full set" also include the "PowerTags"?

manup, 2022-01-18:
> > The Wiser devices will be checked soon, we have a full set of products now for testing and integration. Plan is to get them all working cleanly with DDF :)
> 
> Sounds great. Does that "full set" also include the "PowerTags"?

The set includes the power plug 16A and connected socket outlet, do you mean these?

jh83, 2022-01-18:
> The set includes the power plug 16A and connected socket outlet, do you mean these?

I was thinking about these: https://www.se.com/ww/en/product-range/63626-powertag/21517070574-powertag-energy/?N=2766576315

I have one of the three phase powerTags ordered. So if this isn't included in the above integration, then I will create a new ticket

EjvindHald, 2022-01-18:
Is this PowerTag a part of the Wiser product series?

EjvindHald, 2022-01-18:
I talked to a Product Manager at Schneider regarding this. The PowerTags are currently not a part of Wiser By SE. They are using ZigBee Green Power and not ZigBee 3.0 for communication.

closed 2022-01-22 by manup

jkbkster, 2022-02-20:
> The Wiser devices will be checked soon, we have a full set of products now for testing and integration. Plan is to get them all working cleanly with DDF :)

We are so much looking forward to these devices omboarded. Any new status, or should I just spend all the savings to onboard Wiser for the home ?

JesperJI, 2024-02-10:
Hi All
Did any of you get the Wiser wireless battery 4 button switch working?
Do you have a working DDF?

mwaldhauer, 2024-02-10:
Works fine with the built-in check this Node flow though to get the button
press
[image: image.png]

mwaldhauer, 2024-02-10:
[image]

JesperJI, 2024-02-10:
Thank you.
I will buy a few of the battery switches then.
Strange the switch is not on the compatibility list yet.