dresden-elektronik/deconz-rest-plugin #4535: IKEA Kadrilj looses connection randomly

fribse, 2021-03-07:
## Describe the bug

I have a few IKEA curtains, both FYRTUR and KADRILJ. One of them is randomly disconnecting.
Even if I've just used it, it will suddenly not respond to commands.
If I use several commands, sometimes it will 'wake up' all of the sudden, and perform it.
The other blinds does not go deaf and dumb.

## Steps to reproduce the behavior

  1. Move the blind on the buttons on the blind itself, this will definetely wake it up, or remove and leave the battery out for 15s before reinserting it.
  2. Use a command to move the blind
  3. Wait a while
  4. perform a command again, it will not react until several commands are sent

## Expected behavior

It should move, with or without a pause from the last command

## Screenshots

## Environment

 - Host system: PC
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26670700)
 - deCONZ version: (2.09.03)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? AEOTEC Gen 5 z-wave

## deCONZ Logs

## Additional context

Mimiix, 2021-03-07:
Can you provide some logs? That will be extremely helpful

fribse, 2021-03-07:
Hi @Mimiix 
I would love to, but exactly when do I need to capture the log (I gather it's from the debug view in Deconz)?
When it doesn't react, or when it reacts?

Oh, and how do I close the debug view after opening it?

Mimiix, 2021-03-07:
Both :) gather as many as you can.

Users reported that hitting esc helps

fribse, 2021-03-07:
This is log while one curtain is moving up:

```
timed off, payload: 0058020000, zclSeq: 36
19:38:16:535 remove outdated neighbor 0xF40B
19:38:16:536 remove outdated neighbor 0x6F32
19:38:16:537 remove outdated neighbor 0xD568
19:38:17:567 [INFO] - Button 1002 - TRADFRI motion sensor, broadcast to: 0x6D98, endpoint: 0x01, cluster: ONOFF (0x0006), action: On with timed off, payload: 0058020000, zclSeq: 37
19:38:20:180 [INFO] - No button map for: FYRTUR block-out roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002064, zclSeq: 30
19:38:20:181 ZCL attribute report 0x14B457FFFEFD49F1 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:38:27:495 [INFO] - No button map for: lumi.sensor_motion.aq2, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0400, command: 0x0A, payload: 0000210000, zclSeq: 189
19:38:27:497 ZCL attribute report 0x00158D0005A9E6C9 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
19:38:27:508 [INFO] - No button map for: lumi.sensor_motion.aq2, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0406, command: 0x0A, payload: 00001801, zclSeq: 190
19:38:27:509 ZCL attribute report 0x00158D0005A9E6C9 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
19:38:27:883 [INFO] - No button map for: KADRILJ roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 0800205F, zclSeq: 61
19:38:27:883 ZCL attribute report 0xD0CF5EFFFEEA2305 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:38:28:915 [INFO] - No button map for: KADRILJ roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002052, zclSeq: 62
19:38:28:916 ZCL attribute report 0xD0CF5EFFFEEA2305 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:38:29:946 [INFO] - No button map for: KADRILJ roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002045, zclSeq: 63
19:38:29:947 ZCL attribute report 0xD0CF5EFFFEEA2305 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:38:30:983 [INFO] - No button map for: KADRILJ roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002038, zclSeq: 64
19:38:30:984 ZCL attribute report 0xD0CF5EFFFEEA2305 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:38:32:018 [INFO] - No button map for: KADRILJ roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 0800202B, zclSeq: 65
19:38:32:020 ZCL attribute report 0xD0CF5EFFFEEA2305 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:38:33:049 [INFO] - No button map for: KADRILJ roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 0800201E, zclSeq: 66
19:38:33:050 ZCL attribute report 0xD0CF5EFFFEEA2305 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:38:34:099 [INFO] - No button map for: KADRILJ roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002011, zclSeq: 67
19:38:34:100 ZCL attribute report 0xD0CF5EFFFEEA2305 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:38:35:133 [INFO] - No button map for: KADRILJ roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002004, zclSeq: 68
19:38:35:133 ZCL attribute report 0xD0CF5EFFFEEA2305 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:38:36:160 [INFO] - No button map for: KADRILJ roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002000, zclSeq: 69
19:38:36:161 ZCL attribute report 0xD0CF5EFFFEEA2305 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:38:39:765 [INFO] - No button map for: SPZB0001, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0201, command: 0x0A, payload: 08002059, zclSeq: 198
19:38:39:766 ZCL attribute report 0x00158D0003C3FB71 for cluster: 0x0201, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
```
This is from the curtain that sometimes fails (this time it works):

```
9:40:12:143 [INFO] - No button map for: lumi.sensor_motion.aq2, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0400, command: 0x0A, payload: 0000210000, zclSeq: 103
19:40:12:144 ZCL attribute report 0x00158D0005A9E744 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
19:40:12:159 [INFO] - No button map for: lumi.sensor_motion.aq2, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0406, command: 0x0A, payload: 00001801, zclSeq: 104
19:40:12:160 ZCL attribute report 0x00158D0005A9E744 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
19:40:12:235 delay sending request 220 dt 0 ms to 0x000B57FFFE495FE3, ep: 0x01 cluster: 0x0006 onAir: 1
19:40:12:236 delay sending request 222 dt 0 ms to 0x000B57FFFE495FE3, ep: 0x01 cluster: 0x0008 onAir: 1
19:40:12:316 	0x000B57FFFE495FE3 force poll (2)
19:40:12:337 delay sending request 222 dt 0 ms to 0x000B57FFFE495FE3, ep: 0x01 cluster: 0x0008 onAir: 1
19:40:12:338 ZCL attribute report 0x000B57FFFE495FE3 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:40:12:384 	0x000B57FFFE495FE3 force poll (2)
19:40:12:422 	0x000B57FFFE495FE3 force poll (2)
[... cut ...]
19:40:22:013 ZCL attribute report 0xEC1BBDFFFEB50695 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
19:40:23:098 [INFO] - No button map for: FYRTUR block-out roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002035, zclSeq: 32
19:40:23:099 ZCL attribute report 0xEC1BBDFFFEB50695 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
```

Now I'm waiting for it to fail :-)
Oh, and pressing escape worked :-)

fribse, 2021-03-07:
So now it failed, I had to press the button several times before it reacted, I can't figure out if the log has any relevant info:

```
ZCL attribute report 0x00158D0001FFB0F2 for cluster: 0x0201, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
21:45:12:036 binding for cluster 0x0000 of 0xCCCCCCFFFEB2E9CB exists (verified by reporting)
21:45:13:461 [INFO] - No button map for: TS0121, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0702, command: 0x0A, payload: 000025C90000000000, zclSeq: 72
21:45:13:462 ZCL attribute report 0x842E14FFFE161E1C for cluster: 0x0702, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:45:21:491 ZCL attribute report 0xD0CF5EFFFE158554 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:45:44:250 0xEC1BBDFFFEB50695 error APSDE-DATA.confirm: 0xA7 on task
21:45:45:236 sensor 127 (lumi.sensor_motion.aq2): disable presence
21:45:50:787 ZCL attribute report 0x000D6FFFFE1F0954 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:45:54:993 [INFO] - No button map for: lumi.weather, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0402, command: 0x0A, payload: 0000292F03, zclSeq: 113
21:45:54:995 ZCL attribute report 0x00158D00045CB318 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
21:45:55:023 [INFO] - No button map for: lumi.weather, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0403, command: 0x0A, payload: 000029F903140028FF100029BB27, zclSeq: 115
21:45:55:023 ZCL attribute report 0x00158D00045CB318 for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
21:45:55:480 [INFO] - No button map for: FYRTUR block-out roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002064, zclSeq: 109
21:45:55:481 ZCL attribute report 0xEC1BBDFFFEB50695 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:45:55:553 Current channel 15
21:45:55:560 CTRL got nwk update id 0
21:45:55:591 Device TTL 3715 s flags: 0x7
21:46:00:808 0xEC1BBDFFFEB50695 error APSDE-DATA.confirm: 0xA7 on task
21:46:09:049 delay sending request 102 dt 1 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:09:136 delay sending request 102 dt 1 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:09:236 delay sending request 102 dt 1 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:09:330 delay sending request 105 dt 0 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:09:336 delay sending request 105 dt 0 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:09:436 delay sending request 105 dt 0 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:09:536 delay sending request 105 dt 0 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:09:635 delay sending request 105 dt 0 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:09:736 delay sending request 105 dt 0 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:09:835 delay sending request 105 dt 0 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:09:936 delay sending request 105 dt 0 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:10:036 delay sending request 105 dt 1 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:10:115 Replace task 2168428 type 36 in queue cluster 0x0102 with newer task 2168433 of same type. 1 runnig tasks
21:46:10:121 delay sending request 110 dt 1 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:10:137 delay sending request 110 dt 1 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:10:236 delay sending request 110 dt 1 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:10:336 delay sending request 110 dt 1 ms to 0xEC1BBDFFFEB50695, ep: 0x01 cluster: 0x0102 onAir: 1
21:46:10:347 [INFO] - No button map for: FYRTUR block-out roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002062, zclSeq: 110
21:46:10:348 ZCL attribute report 0xEC1BBDFFFEB50695 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:46:11:410 [INFO] - No button map for: FYRTUR block-out roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002060, zclSeq: 111
21:46:11:411 ZCL attribute report 0xEC1BBDFFFEB50695 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:46:13:426 [INFO] - No button map for: FYRTUR block-out roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 0800205F, zclSeq: 112
21:46:13:428 ZCL attribute report 0xEC1BBDFFFEB50695 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:46:14:575 [INFO] - No button map for: FYRTUR block-out roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 0800205A, zclSeq: 113
21:46:14:576 ZCL attribute report 0xEC1BBDFFFEB50695 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:46:14:672 Bind response success for 0x842e14fffe161e1c ep: 0x01 cluster: 0x0B04
21:46:15:507 [INFO] - No button map for: FYRTUR block-out roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 08002056, zclSeq: 114
21:46:15:508 ZCL attribute report 0xEC1BBDFFFEB50695 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
[... cut ...]
21:46:17:615 ZCL attribute report 0xEC1BBDFFFEB50695 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:46:18:368 ZCL attribute report 0xD0CF5EFFFEF2939A for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:46:18:435 Set sensor check interval to 1000 milliseconds
21:46:18:606 [INFO] - No button map for: FYRTUR block-out roller blind, unicast to: 0x0000, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), command: 0x0A, payload: 0800204A, zclSeq: 117
21:46:18:610 ZCL attribute report 0xEC1BBDFFFEB50695 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
```

Mimiix, 2021-03-07:
There's error code 0xA7. 

How does the map look like?

fribse, 2021-03-07:
It looks like this.
But I don't recall seeing the connection to the IKEA range extender earlier today, ie. before pressing the 'action button' multiple times until it reacted.
[image]

fribse, 2021-03-08:
Looking at it this morning, I don't see the connection from the blind:
[image]
But it still reacts to commands?

Mimiix, 2021-03-08:
The lines are indications. It is a odd situation tho. Let me check with @manup  and let him get back to you. I am thinking this bug might be involved: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261

You might want to look into Source Routing in the meanwhile.

Larsn1, 2021-03-08:
I have two Fyrtur with the same issue. They go unresponsive after some days/weeks and work normal after a short disconnect of the battery. If there is a way to get them to work without power cycle, I would be glad.

My logs look the same as fribse‘s

manup, 2021-03-08:
I have only one KADRILJ in my setup connected though a Ikea E27 light bulb with firmware version 2.3.068. Got no connection issues in months. I'm using the latest ConBee II firmware version 0x26680700 and have also updated all Ikea routers in my network to the latest version.

My KADRILJ is on firmware version 2.2.009.

Not sure if this is related but might be wort a shot to check / compare versions?

ebaauw, 2021-03-10:
My FYRTUR used to become incommunicado quite often in my production network (~50 routers; ~50 end-devices).  Removing/re-inserting the battery usually fixed the issue,  but sometimes I needed to open the network while doing that.  I haven't been able to identify the "deadly combination", but suspect that the FYRTUR doesn't play nice with one/some of the routers in my production network.

A couple of months ago, I moved the FYRTUR (and my Eurotronik Spirit TRVs, that would also become incommunicado regularly) to a second network (another RaspBee, another Pi), with 1 router (a TRÅDFRI repeater) and 10 end devices (also moved the open/close controller).  It's been stable since.

Another peculiarity of the IKEA blinds: _Open_ and _Close_ are not idempotent.  The blind stops moving when it receives a second _Open_ or _Close_ (I suppose to support stopping the movement using open/close remote).  When deCONZ doesn't receive a timely confirmation of the first _Open_ or _Close_, it re-sends the command, sometimes causing the blind to stop moving.  To mitigate this, I changed my rules to set `lift` instead of `open` (or `on`), so deCONZ will send a _Move to Lift Percentage_ instead of an _Open_ or _Close_.

fribse, 2021-03-10:
Most of my routers are IKEA, I only have one Heiman siren (had two, but the other one made my zigbee network completely unstable after 2.09.03).

Larsn1, 2021-03-10:
I have a few Ikea bulbs, 3 repeaters, Philips Hue, an Ikea outlet, INNR bulbs and a lightstrip from hue. My network was never really stable at all but somewhat usable. I switched to a conbee II newest firmware from my rasbee and use an extension cable. I still have a faster SD card laying somewhere, but almost no time after work.

Network stability and how to address problems would be much appreciated to know, but ressources on that topic are rare. In the documentation everything is working fine and is easy and you don’t need to press a button 10 times for a minute to turn on the lights.

Mimiix, 2021-03-10:
@Larsn1 Would be useful to see if you have interference issues. Could you make some logs?
To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

fribse, 2021-03-10:
> need to press a button 10 times for a minute to turn on the lights

That sounds even worse than when I had big problems. My issue was a Heiman Siren...

olemr, 2021-03-12:
I have 5 Fyrturs, 2 of them just 1m from the ConBee and I have only been able to control them once.
sort of gave up on them, but the open/close remote direct pairing works from the old WEBapp, so that saves the WAF score.
Don't use the routers that came with them. Mixture of HUE, Ikea and Namron.

closed 2021-04-15 by github-actions[bot]

froogl, 2021-05-24:
There is a new firmware for FYRTUR and KADRILJ (V-2.3.073) which addresses some issues:

- Secure OTA upgrade introduced.
- The state of blind in the App was not getting updated when the battery of the blind was removed and reinserted again. This has been fixed.
- Zigbee commands received by the Blinds during Motor FW initialisation were causing unwanted behaviour. This is now fixed.

Maybe it's now more reliable with DeCONZ.