dresden-elektronik/deconz-rest-plugin #4536: Sengled Model Z01-A19NAE26 Smart LED Tunable White A19

stevewull, 2021-03-08:
## Device

- Product name: Smart LED Tunable White 2700K-6500K
- Manufacturer: Sengled
- Model identifier: Z01-A19NAE26
- Device type :
  - Light

Note: DeCONZ recognizes this as a light and can control the brightness (See API output at bottom of this post). However, API does not have the ct attribute and the color temperature cannot be displayed or controlled. This was mentioned in issue #3045, but it was never resolved. Bulb also has a metering cluster.

## Screenshots

Node Info:
[image]

Clusters:
[image]

Basic cluster:
[image]

Identify cluster:
[image]

Groups cluster attributes:
[image]

Scenes cluster attributes:
[image]

On/Off Cluster:
[image]

Level Control Cluster:
[image]

Color Control Cluster:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Simple metering cluster:
[image]

Diagnostics cluster:
[image]

OTAU Cluster:
[image]

API:
[image]

stevewull, 2021-03-08:
On the Color Temperature cluster, I see that names of some clusters on truncated in my post. Here they are:

[image]
[image]

deviantintegral, 2021-03-09:
👋 I think this is a duplicate of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4469

Also, see my note there about being able to set the colour temperature.

stevewull, 2021-03-09:
Whoops. Sorry I missed that. Hopefully, the screenshots here provide what you were asking for.  I'll add the rest of my comments there.

Mimiix, 2021-03-09:
As this is a duplicate, I'll close it and we continue in the other one.

closed 2021-03-09 by Mimiix