dresden-elektronik/deconz-rest-plugin #4537: LIDL Silvercrest Doorbell - HG06668

knutbrecht, 2021-03-08:
## Device

- Product name: LIDL Silvercrest Doorbell 
- Manufacturer: _TZ1800_ladpngdx
- Model identifier: TS0211
- Device type : Please remove all unrelated device types. 
  - Remote
  - Switch

## Screenshots

[image]
[image]
[image]

knutbrecht, 2021-03-08:
[image]
[image]
[image]
[image]

Mimiix, 2021-03-08:
what version of deCONZ are you on?

knutbrecht, 2021-03-08:
2.09.03

Mimiix, 2021-03-08:
It is supported by https://github.com/dresden-elektronik/deconz-rest-plugin/tree/v2.10.0-beta

Also, Make sure to re-pair after updating. It would be visible in the API.

closed 2021-03-08 by Mimiix