dresden-elektronik/deconz-rest-plugin #4538: Neo NAS-AB02B0

ikke-zelf, 2021-03-08:
## Device
I'm sorry i have no access to the deconz gui, so I can not get a hold of other data formats.

    // Neo
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_d0yu2xgi'}],
        zigbeeModel: ['0yu2xgi'],
        model: 'NAS-AB02B0',
        vendor: 'Neo',
        description: 'Temperature & humidity sensor and alarm',
        fromZigbee: [fz.neo_t_h_alarm, fz.ignore_basic_report],
        toZigbee: [tz.neo_t_h_alarm],
        exposes: [
            e.temperature(), e.humidity(), exposes.binary('humidity_alarm', ea.STATE_SET, true, false), e.battery_low(),
            exposes.binary('temperature_alarm', ea.STATE_SET, true, false),
            exposes.binary('alarm', ea.STATE_SET, true, false),
            exposes.enum('melody', ea.STATE_SET, Array.from(Array(18).keys()).map((x)=>(x+1).toString())),
            exposes.numeric('duration', ea.STATE_SET).withUnit('second'),
            exposes.numeric('temperature_min', ea.STATE_SET).withUnit('°C'),
            exposes.numeric('temperature_max', ea.STATE_SET).withUnit('°C'),
            exposes.numeric('humidity_min', ea.STATE_SET).withUnit('%'),
            exposes.numeric('humidity_max', ea.STATE_SET).withUnit('%'),
            exposes.enum('volume', ea.STATE_SET, ['low', 'medium', 'high']),
            exposes.enum('power_type', ea.STATE, ['battery_full', 'battery_high', 'battery_medium', 'battery_low', 'usb']),
        ],
    },

## Screenshots

I'm sorry i have no access to the deconz gui, so I can not get a hold of other data formats.

FROMZIGBEE
 neo_t_h_alarm: {
        cluster: 'manuSpecificTuya',
        type: ['commandSetDataResponse', 'commandGetData'],
        convert: (model, msg, publish, options, meta) => {
            const dp = msg.data.dp;
            const value = tuya.getDataValue(msg.data.datatype, msg.data.data);

            switch (dp) {
            case tuya.dataPoints.neoAlarm:
                return {alarm: value};
            case tuya.dataPoints.neoUnknown2: // 0x0170 [0]
                break;
            case tuya.dataPoints.neoTempAlarm:
                return {temperature_alarm: value};
            case tuya.dataPoints.neoHumidityAlarm: // 0x0172 [0]/[1] Disable/Enable alarm by humidity
                return {humidity_alarm: value};
            case tuya.dataPoints.neoDuration: // 0x0267 [0,0,0,10] duration alarm in second
                return {duration: value};
            case tuya.dataPoints.neoTemp: // 0x0269 [0,0,0,240] temperature
                return {temperature: (value / 10).toFixed(1)};
            case tuya.dataPoints.neoHumidity: // 0x026A [0,0,0,36] humidity
                return {humidity: value};
            case tuya.dataPoints.neoMinTemp: // 0x026B [0,0,0,18] min alarm temperature
                return {temperature_min: value};
            case tuya.dataPoints.neoMaxTemp: // 0x026C [0,0,0,27] max alarm temperature
                return {temperature_max: value};
            case tuya.dataPoints.neoMinHumidity: // 0x026D [0,0,0,45] min alarm humidity
                return {humidity_min: value};
            case tuya.dataPoints.neoMaxHumidity: // 0x026E [0,0,0,80] max alarm humidity
                return {humidity_max: value};
            case tuya.dataPoints.neoPowerType: // 0x0465 [4]
                return {
                    power_type: {0: 'battery_full', 1: 'battery_high', 2: 'battery_medium', 3: 'battery_low', 4: 'usb'}[value],
                    battery_low: value === 3,
                };
            case tuya.dataPoints.neoMelody: // 0x0466 [5] Melody
                return {melody: value};
            case tuya.dataPoints.neoUnknown3: // 0x0473 [0]
                break;
            case tuya.dataPoints.neoVolume: // 0x0474 [0]/[1]/[2] Volume 0-max, 2-low
                return {volume: {2: 'low', 1: 'medium', 0: 'high'}[value]};
            default: // Unknown code
                meta.logger.warn(`Unhandled DP #${dp}: ${JSON.stringify(msg.data)}`);
            }
        },
    },
	
TOZIGBEE
neo_t_h_alarm: {
        key: [
            'alarm', 'melody', 'volume', 'duration',
            'temperature_max', 'temperature_min', 'humidity_min', 'humidity_max',
            'temperature_alarm', 'humidity_alarm',
        ],
        convertSet: async (entity, key, value, meta) => {
            switch (key) {
            case 'alarm':
                await tuya.sendDataPointBool(entity, tuya.dataPoints.neoAlarm, value);
                break;
            case 'melody':
                await tuya.sendDataPointEnum(entity, tuya.dataPoints.neoMelody, parseInt(value, 10));
                break;
            case 'volume':
                await tuya.sendDataPointEnum(
                    entity,
                    tuya.dataPoints.neoVolume,
                    {'low': 2, 'medium': 1, 'high': 0}[value]);
                break;
            case 'duration':
                await tuya.sendDataPointValue(entity, tuya.dataPoints.neoDuration, value);
                break;
            case 'temperature_max':
                await tuya.sendDataPointValue(entity, tuya.dataPoints.neoMaxTemp, value);
                break;
            case 'temperature_min':
                await tuya.sendDataPointValue(entity, tuya.dataPoints.neoMinTemp, value);
                break;
            case 'humidity_max':
                await tuya.sendDataPointValue(entity, tuya.dataPoints.neoMaxHumidity, value);
                break;
            case 'humidity_min':
                await tuya.sendDataPointValue(entity, tuya.dataPoints.neoMinHumidity, value);
                break;
            case 'temperature_alarm':
                await tuya.sendDataPointBool(entity, tuya.dataPoints.neoTempAlarm, value);
                break;
            case 'humidity_alarm':
                await tuya.sendDataPointBool(entity, tuya.dataPoints.neoHumidityAlarm, value);
                break;
            default: // Unknown key
                throw new Error(`Unhandled key ${key}`);
            }
        },
    },

Mimiix, 2021-03-08:
@SwoopX I think we need the screens?

ikke-zelf, 2021-03-08:
i'm going to try to install the vnc options for the headless deconz on my pi. If i'm not mistaken I will be able to get a hold on the deconz-gui that way, at least if I understand correctly

Mimiix, 2021-03-08:
Yes

SwoopX, 2021-03-08:
> @SwoopX I think we need the screens?
> 

Yes

ikke-zelf, 2021-03-09:
a step further, I succeeded (next time i will start with the desktop raspberry installation iso the light one...).
Now to get the resolution correct and then going to pair the device (for as far that goes)

Smanar, 2021-03-20:
Probably solved by https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4617

closed 2021-04-19 by github-actions[bot]