dresden-elektronik/deconz-rest-plugin #4539: Issue on LIDL Livarno Lux light strip

jokerigno, 2021-03-08:
## Describe the bug

I just buyed a LIDL Livarno Lux (HG06104A) and I tried to switch on colorloop functionality in Home Assistant but nothing happens. I opened an issue #47287 but it seems that in not related to the integration but to HW and Deconz.

## Steps to reproduce the behavior

  1. Turn on the light in HA
  2. Click on colorloop
  3. Nothing happens

## Expected behavior

Light change color randomly

## Screenshots

## Environment

 - Host system: Unraid
 - Running method: Marthoc Docker container 
 - Firmware version: (264A0700)
 - deCONZ version: (2.10.01)
 - Device: ConBee II 
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II - Yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

I tried to add logs. I can see them in deConz but I'm unable to paste outside VNC.
## Additional context

ebaauw, 2021-03-08:
I don't have the lightstrip to test, but the E27 colour bulb doesn't support `colorloop` (or even `hue`/`sat`).  Could you try and activate the color loop from the GUI?  If it reports `status 0xc3`, it doesn't support the command.
[image].

Doing a PR not to expose the unsupported `state` attributes.

Mimiix, 2021-03-09:
@ebaauw Took a test on my ledstrip

[image]

Ledstrip seems to not support it either.

jokerigno, 2021-03-09:
I would love to test it too but I don't know how to reach this windows. I logon in deconz app via VNC and then?

Mimiix, 2021-03-09:
Go to the Color Cluster (check the wiki on how to read/find clusters).

Then scroll a bit down.

jokerigno, 2021-03-09:
I confirm that

[image]

jokerigno, 2021-03-23:
Hi all, 

I noticed that my Model HG06104A has now only Color Temp and brightness. No more light control. Why? This is a RGB strip. Why I cannot change color anymore? Maybe I changed some value in the color cluster without intention?

Mimiix, 2021-03-24:
@jokerigno I have no clue. Best to ask on the Phoscon repo.

jokerigno, 2021-03-24:
I will but I think there's more than this.

The led strip model is know unknown (previous was HG06104A as reported in original post).

Mimiix, 2021-03-26:
@ebaauw Based on https://github.com/dresden-elektronik/phoscon-app-beta/issues/362 it now broke the ledstrip.

jokerigno, 2021-03-26:
Another thing that I noticed in HA (don't know if can help, I share it because HA developer told me that integration just copy info from deconz ) is that every time brightness is and then change color brightness is changed automatically to 219.

Moreover in HA UI the color is not reported correctly.

ebaauw, 2021-03-26:
> Based on dresden-elektronik/phoscon-app-beta#362 it now broke the ledstrip

Are you telling me that HA and even Phoscon are using `hue`/`sat` instead of `xy` to set the colour?

If not, please post screenshot of _Basic_ cluster and list the `/lights` resources of the now broken device( s).

b2un0, 2021-04-06:
can cofirm this issue, the lidl melinera stripe from #4038 are also broken since the latest update

b2un0, 2021-04-06:
The Lidl Melinera Christmas Stripe from #3716 are als broken.

The Device is broken since the latest stable release.

basic cluster:

[image]

`/lights` resources

```json
    "27": {
        "etag": "fed8c4c3652aa731b281608a962ea4da",
        "hascolor": true,
        "lastannounced": null,
        "lastseen": "2021-04-06T18:45Z",
        "manufacturername": "LIDL Livarno Lux",
        "modelid": "HG06467",
        "name": "Color dimmable light 27",
        "state": {
            "bri": 254,
            "colormode": "hs",
            "on": true,
            "reachable": true
        },
        "swversion": null,
        "type": "Color dimmable light",
        "uniqueid": "bc:33:ac:ff:fe:73:3a:4a-01"
    },
```

LeoeLeoeL, 2021-04-16:
Any good news about this bug?

Smanar, 2021-04-16:
Ha yes, some field are missing in the JSON

```
    "state": {
      "bri": 25,
      "colormode": "hs",
      "effect": "none",
      "hue": 53691,
      "on": false,
      "reachable": true,
      "sat": 141
    },
```

Do you have the last version working ?

LeoeLeoeL, 2021-04-16:
No, I stay on 2.10.4

Smanar, 2021-04-16:
So the 2.10.4 is still working or not ?

LeoeLeoeL, 2021-04-16:
2.10.4 doesn't work.
I didn't install 2.11.0 beta yet because i didn't find anything useful for me in the changelog but I can try.

Smanar, 2021-04-17:
I realy don't see the problem from the new version.

Are you able to compile the code to test ?
I don't understand how the field "hue" and "sat" can be in the Json atm.

In the file de_web_plugin.cpp arounf line 2815

    else if (isXmasLightStrip(lightNode))
    {
        lightNode->removeItem(RStateAlert);
        lightNode->removeItem(RStateX);
        lightNode->removeItem(RStateY);
    }

can you add

```
        lightNode->addItem(DataTypeUInt16, RStateHue);
        lightNode->addItem(DataTypeUInt8, RStateSat);
        lightNode->addItem(DataTypeString, RStateEffect)->setValue(RStateEffectValues[R_EFFECT_NONE]);
```

ebaauw, 2021-04-17:
This issue is about the regular lightstrip not supporting colorloop.  Please open a new issue for the Xmas lightstrip.  If you had done so in the first place, I would have seen it right away.

LeoeLeoeL, 2021-04-17:
Same bug as 2.10.4.
For information, Deconz 2.11.0 beta closed 4 times itself after few seconds. Now it works. I hope it's stable.

jokerigno, 2021-04-17:
@ebaauw why did you close this issue?

ebaauw, 2021-04-17:
Because the original problem for this issue has been addressed, see above.

jokerigno, 2021-04-17:
> Because the original problem for this issue has been addressed, see above.

But the solution broke the component.

Do you want me to open another ticket?

ebaauw, 2021-04-17:
Sorry, is this about the regular lightstrip or about the Xmas light strip?

ebaauw, 2021-04-17:
Double-checking my LIDL RGB bulb.
```
{
  "ctmax": 65279,
  "ctmin": 0,
  "etag": "85cb2941a608af15b4729881171c5ca4",
  "hascolor": true,
  "lastannounced": "2021-01-09T01:25:04Z",
  "lastseen": "2021-04-17T12:18Z",
  "manufacturername": "LIDL Livarno Lux",
  "modelid": "HG06106C",
  "name": "LIDL RGB",
  "state": {
    "alert": "none",
    "bri": 254,
    "colormode": "ct",
    "ct": 153,
    "on": true,
    "reachable": true,
    "xy": [
      0,
      0
    ]
  },
  "swversion": "0x00000050",
  "type": "Extended color light",
  "uniqueid": "58:8e:81:ff:fe:e2:cd:a0-01"
}
```
That's looking good, except I just set the colour using `xy`.  Looking at the websocket I get a notification that `colormode` is `xy`, followed by one that it is `ct`.  WTF?  The _Color Mode_ attributes 0x0008 is `xy` and the light doesn't support _Advanced Color Mode_ 0x4001.

Checking with the sniffer, the light reports _Current Hue_ and _Current Saturation_, even though it doesn't support setting these.  The values are bogus, however; when I set `xy` to [0, 0] (blue), _Current Hue_ reports 0 (red).

ebaauw, 2021-04-17:
F*ck me.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/59eee63ab3833aa2ffb4ed30808578b23732529d/de_web_plugin.cpp#L3249-L3253

I'll add a check that the reported colormode is `hs`.

jokerigno, 2021-04-17:
Thank you for working on it. Another thing that I noticed and reported in this post.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4539#issuecomment-808494660

By default brightness is 219. Everytime I change brightness color is changed to default value. Everytime i change brightness and then color brightness is changed again automatically to 219.

What am I expecting ? That if I set brigthess to 255 and then change color brightness stays at 255. And when I change color and then adjust brightness color stays the same instead of default one.

ebaauw, 2021-04-17:
By "color brightness" do you mean colour temperature (the `ct` attribute)?
I think that's the same issue: the REST API erroneously reports that the device is in Colour Temperature mode, so HA refreshes the colour from `ct` instead of from `xy`.  Like most non-Hue lights, the LIDL won't update `ct` when setting `xy`, so the color appears to be reverted.

jokerigno, 2021-04-17:
Sorry, a comma was required:
Everytime i change brightness and then color, brightness is changed again automatically to 219.

jokerigno, 2021-04-17:
> By "color brightness" do you mean colour temperature (the `ct` attribute)?
> I think that's the same issue: the REST API erroneously reports that the device is in Colour Temperature mode, so HA [...]

In your opinion this can also cause the fact that entity icon color doesn't change based on entity light color?

ebaauw, 2021-04-17:
> Everytime i change brightness and then color, brightness is changed again automatically to 219.

I'm not seeing that.  Are you sure you don't have an automation running to reset the brightness.  Do you mean brightness of the actual light or brightness as reported by the API?

> In your opinion this can also cause the fact that entity icon color doesn't change based on entity light color?

Yes, the light changes colour alright, but the API reports the last colour temperature.  This behaviour should be fixed by my commit above.

jokerigno, 2021-04-17:
No I had just setup default behaviour with light profile with brightness 255 and a specific color. In this case brightness was always reset to 255. Then I removed it and reboot HA. After that default color was warm white with brightness 219 and the issue was still there with value 219 against 255.

closed 2021-04-18 by manup

ebaauw, 2021-04-18:
Please try these changes when the next beta is released.  If it still behaves weirdly, please capture a log of the API sent and the websocket notifications received.  I’m sorry, I have no clue how to do that in HA.

LeoeLeoeL, 2021-04-23:
It seems to work again in HA. Now entity icons are in RGB according the lamp colour.
But phoscon doesn't show the colour bars.

jokerigno, 2021-04-24:
Light status is correct. But the issue with brightness restored to default every time is still present.
@LeoeLeoeL did you noticed it too?

LeoeLeoeL, 2021-04-25:
Brightness remains the same when switched off ; Even using physical switch.

LeoeLeoeL, 2021-04-25:
After 15 minutes without current, when I switch on, lamps keep their brightness but HA reports full brightness even if the correct value should be less.

jokerigno, 2021-04-26:
> Please try these changes when the next beta is released. If it still behaves weirdly, please capture a log of the API commands sent and the websocket notifications received. I’m sorry, I have no clue how to do that in HA.

@ebaauw here is the log.

Basically everytime as reported also by @LeoeLeoeL  light is reset to default settings (brightness and color)

```
2021-04-26 12:53:32 WARNING (MainThread) [homeassistant.loader] You are using a custom integration spotcast which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-26 12:53:39 WARNING (MainThread) [homeassistant.loader] You are using a custom integration variable which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-26 12:53:40 WARNING (MainThread) [homeassistant.loader] You are using a custom integration smartir which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-26 12:53:40 WARNING (MainThread) [homeassistant.loader] You are using a custom integration skodaconnect which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-26 12:53:40 WARNING (MainThread) [homeassistant.config] Package r2d2 contains invalid customize
2021-04-26 12:53:40 WARNING (MainThread) [homeassistant.config] Package videosorveglianza contains invalid customize
2021-04-26 12:53:41 WARNING (MainThread) [homeassistant.loader] You are using a custom integration blitzortung which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-26 12:53:41 WARNING (MainThread) [homeassistant.loader] You are using a custom integration garbage_collection which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-26 12:53:41 WARNING (MainThread) [homeassistant.loader] You are using a custom integration circadian_lighting which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-26 12:53:41 WARNING (MainThread) [homeassistant.loader] No 'version' key in the manifest file for custom integration 'circadian_lighting'. As of Home Assistant 2021.6, this integration will no longer be loaded. Please report this to the maintainer of 'circadian_lighting'
2021-04-26 12:53:41 WARNING (MainThread) [homeassistant.loader] You are using a custom integration frigate which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-26 12:53:41 WARNING (MainThread) [homeassistant.loader] You are using a custom integration anniversaries which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-26 12:53:41 WARNING (MainThread) [homeassistant.loader] You are using a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-26 12:53:45 DEBUG (MainThread) [pydeconz.gateway] Sending "get" "None" to "192.168.2.202 "
2021-04-26 12:53:46 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: {'config': {'UTC': '2021-04-26T10:53:46',
            'apiversion': '1.16.0',
            'backup': {'errorcode': 0, 'status': 'idle'},
            'bridgeid': '00212EFFFF04F4C4',
            'datastoreversion': '93',
            'devicename': 'ConBee II',
            'dhcp': True,
            'disablePermitJoinAutoOff': False,
            'factorynew': False,
            'fwversion': '0x264a0700',
            'gateway': '192.168.2.1',
            'internetservices': {'internet': 'connected',
                                 'remoteaccess': 'disconnected',
                                 'swupdate': 'connected',
                                 'time': 'connected'},
            'ipaddress': '192.168.2.202',
            'lightlastseeninterval': 60,
            'linkbutton': False,
            'localtime': '2021-04-26T12:53:46',
            'mac': '02:42:c0:a8:02:ca',
            'modelid': 'deCONZ',
            'name': 'Phoscon-GW',
            'netmask': '255.255.255.0',
            'networkopenduration': 180,
            'panid': 48603,
            'portalconnection': 'disconnected',
            'portalservices': False,
            'portalstate': {'communication': 'disconnected',
                            'incoming': False,
                            'outgoing': False,
                            'signedon': False},
            'proxyaddress': 'none',
            'proxyport': 0,
            'replacesbridgeid': None,
            'rfconnected': True,
            'starterkitid': '',
            'swupdate': {'checkforupdate': False,
                         'devicetypes': {'bridge': False,
                                         'lights': [],
                                         'sensors': []},
[... cut ...] ed Cucina","powerup":7,"swversion":"0214.201000EB","type":"Extended color light","uniqueid":"00:21:2e:ff:ff:05:8f:ea-0a"},"e":"changed","id":"9","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:05:8f:ea-0a"}
2021-04-26 13:02:08 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"31","r":"sensors","state":{"lastupdated":"2021-04-26T11:02:08.412","orientation":[5,8,-81],"tiltangle":7,"vibration":false,"vibrationstrength":6},"t":"event","uniqueid":"00:15:8d:00:02:b1:29:83-01-0101"}

```

ebaauw, 2021-04-26:
Please, next time **attach** a log file instead of listing large amounts of text.

The API seems to be working correctly; I don't see any discrepancies between the requested and reported brightness (except that 255 is not a valid value for `bri` and mapped to 254).  The value 219 doesn't even appear in the log.

Here HA is setting on, brightness, and colour:
```
2021-04-26 13:00:37 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True, 'xy': (0.488, 0.316), 'bri': 255}" to "192.168.2.202 /lights/11/state"
```
deCONZ accepts the command:
```
2021-04-26 13:00:37 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/11/state/on': True}},
 {'success': {'/lights/11/state/bri': 254}},
 {'success': {'/lights/11/state/xy': [0.488, 0.316]}}]
```
Then it sends a websocket notification as a result of the command:
```
2021-04-26 13:00:37 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":null,"bri":254,"colormode":"xy","ct":153,"on":true,"reachable":true,"xy":[0.488,0.316]},"t":"event","uniqueid":"84:2e:14:ff:fe:2c:43:89-01"}
```
Next, it sends a notification for `bri` 2.  This is caused by an attribute report by the lightstrip, on the _Move to Level (with On/Off)_ (2, 0) that deCONZ sends when PUTing both `on` and `bri`:
```
2021-04-26 13:00:37 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":null,"bri":2,"colormode":"xy","ct":153,"on":true,"reachable":true,"xy":[0.488,0.316]},"t":"event","uniqueid":"84:2e:14:ff:fe:2c:43:89-01"}
```
Which is followed by the attribute report for `bri` 254, after the _Move to Level_ (254, 4):
```
2021-04-26 13:00:38 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":null,"bri":254,"colormode":"xy","ct":153,"on":true,"reachable":true,"xy":[0.488,0.316]},"t":"event","uniqueid":"84:2e:14:ff:fe:2c:43:89-01"}
```
deCONZ correctly reports `bri` 254.

jokerigno, 2021-04-26:
OPS. Sorry.

And sorry for 219. I changed default value to 255.

I don't know what you see but what I see is not that correct. Basically I switched light on and then changed 1) color 2) brightness. As told in previous post every time light is reset to defaul values.

[image]

ebaauw, 2021-04-26:
From your log, deCONZ is reporting the brightness requested by HA.  If that's not what you expect, you need to check the HA side.  I do see HA sending requests to change the brightness to 255.

LeoeLeoeL, 2021-04-30:
After a power loss, Deconz and HA report a brightness of 255 even if the real brightness of the lamps is 1.

Mimiix, 2021-05-08:
@jokerigno it's something the device needs to support. So unfortunately this can't be implemented by us 😢

LeoeLeoeL, 2021-05-08:
For curiosity, I bought the Lidl bridge yesterday.
Lights don'thave the Power On option but plugs do.
A strange thing, I configured a plug to stay off after a power loss. Then I paired that plug in Deconz. The plug remembers to stay off after a power loss.

jokerigno, 2021-05-09:
> @jokerigno it's something the device needs to support. So unfortunately this can't be implemented by us 😢

I was afraid of it but I asked you just to be sure. Thank you for your feedback :)