dresden-elektronik/deconz-rest-plugin #4542: Deconz 2.09.03-raspbian-buster-stable takes >= 20 minutes to start

erikproper, 2021-03-08:
## Describe the bug

Deconz 2.09.03-raspbian-buster-stable takes >= 20 minutes to start; that is, no GUI no Webinterface for 20 minutes.

## Steps to reproduce the behavior

Happens at reboot

## Expected behavior

A bit faster would be nice.

## Screenshots

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian 
 - Firmware version: (26390500)
 - deCONZ version: (2.09.03 / 12/23/2020)
 - Device: RaspBee I 
 - Is there any other USB or serial devices connected to the host system? If so: Which?
 - A ZWave stick.

## deCONZ Logs
Attached the log file and DB file.
[deCONZ-error.log.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6103753/deCONZ-error.log.gz)
[zll.db.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6103755/zll.db.gz)

Smanar, 2021-03-08:
Hello, 
BTW how many sensor you have ?

erikproper, 2021-03-08:
Sensors: 48
Switches: 20
Lights: 43

Mimiix, 2021-03-09:
Hi!
Can you provide APS and error logs?:

To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

erikproper, 2021-03-09:
Hi Dennis,

What do you mean by APS?

The current log files (as attached) where created when using "/usr/bin/deCONZ
--dbg-info=2 --http-port=80"

What other Flags would you like to be added?

Note: Need to carefully plan restarts, as it is an operational system ....

Mimiix, 2021-03-09:
As the template suggested:

  Secondly, for versions above 2.05.87: In the deCONZ GUI: click Help > debug view. **Output of APS,ZDP, ZCL and INFO can be handy.** 
In total and to be sure: 

--dbg-error=2 --dbg-aps=2 --dbg-info=2 --dbg-zdp=2 --dbg-zcl=1

erikproper, 2021-03-09:
I've read about the options in the GUI. But ... the issue happens during
boot, *before* the GUI comes up.
Or ... does the debug view show "historical" data?

So, I guess I will need to update the deconz-gui.service script with the
extra debug flags.

erikproper, 2021-03-09:
Just started (with a modified systemd script) with:
- /usr/bin/deCONZ --dbg-error=2 --dbg-aps=2 --dbg-info=2 --dbg-zdp=2 --dbg-zcl=1 --http-port=80

This resulted in the log file you can find on: 
   https://www.dropbox.com/t/0y359K7ealZhWH9j

However, after more than an hour there was still no GUI or web interface ...

Sorry. Can't attached (if as a gzip) as it is too large ...

Mimiix, 2021-03-09:
No worries :)

Mimiix, 2021-03-09:
I've checked it, but there's no logs at all (different then the others you've provided).   What RPI are you running?

I've asked some people to check this issue with me.

erikproper, 2021-03-09:
The RPI runs on (apt-get uprgade / updated) standard pre-configured Deconz
image.

The physical RPI was actually bought from Dresden Elektronik with the
Raspbee (1) board in there.

SwoopX, 2021-03-09:
Your DB is fubar and show some kind of weird split personallity. Table `device` and `device_descriptors` seem to be ok on a brief glimpse. However, you just got tons of garbage in `nodes` and `sensors` tables.

I'd suggest the following, but please take note that you'll loose all your light and sensor IDs. This requires updating your automations but should be more convenient than resetting everything:

- Make a backup copy of the DB
- Edit the DB via sqlite3
- Issue `delete from sensors` and `delete from nodes`
- Restart deconz
- Enter light search and let it run for 10 mins
- If stuff is still missing, enter either light or sensor search as required and especially ensure for sensors/switches that you trigger their functionality

That should bring back your devices and recreate everything clean and nicely.

erikproper, 2021-03-09:
Before doing so ... would that lead to the not-returning lights/sensors to
get a new ID (i.e. as used in the REST API) when "recreating" them ...?

erikproper, 2021-03-09:
Hi,
  The "so this includes also any ID you see via API" would require knowledge about the data model ....

closed 2021-03-09 by erikproper

erikproper, 2021-03-09:
> 🤔? Did the suggestion @SwoopX <https://github.com/SwoopX> did solve the
> problem @erikproper <https://github.com/erikproper>?

Not yet.

If I understand @SwoopX correctly, then the:
- "those delete all entries in the corresponding tables, so this includes
also any ID you see via API."
implies that there is a potential impact on IDs used in the REST API.
This will have a large impact on my home automation setup.

So, I first need to find a way to manage that (and home automation is just
a hobby ....)

As such, I would have preferred a solution where it would be possible to do
some "automatic garbage collecting".

==> Note: when I use the REST API to look at my sensors/lights/groups, I
see no old nodes.
So, at least on-paper it should be possible to remove any dead-wood from
the database ... i.e. garbage collecting ...

SwoopX, 2021-03-09:
Well, what should I say? I tried to provide you with some options preventing you from resetting each and every device.

There's of course also the possibility to note down your current light/sensor IDs and try to purge the 80.000 duplicates in the tables somehow 🤷‍♂️ I fear there's no better news.

Mimiix, 2021-03-09:
@SwoopX what caused this?

erikproper, 2021-03-09:
Okay guys. It might still be relevant to figure out what happened to the
database.

Meanwhile, I think I saved my own ass. May not be a developer, but do know
a bit about databases ...

Went into "sqlite3 zll.db" and poked around. In doing so, I discovered that
the KEY of the nodes and sensors tables was not actually unique in the
table.
So, multiple rows with the same KEY .... ooops!

Then tried (see
https://stackoverflow.com/questions/18259692/how-to-recover-a-corrupt-sqlite3-database/18260642
).
This resulted in a lot of uniqueness errors ... due to the key violations.

So ... then I did the following:

Under the *assumption* that the insert commands issued by:
  sqlite3 zll.db ".recover"
reflect the order of insertion. i.e. the last recorded row is the
correct/latest one, I should be able to recover a clean database
by inverting the order of the inserts. Because of the last recorded row
would be inserted first, the key would then block the insertion of
the older double ones.

To do this, I then did:
1: sqlite3 broken.db ".recover" > data.txt
2: split data.txt into:
   - prelude.txt with the create commands
   - insert.txt with the insert commands
   - postlude.txt with the finishing commands
3: do a "tac insert.txt > reversed.txt"
5: Do "cat prelude.txt reversed.txt postlude.txt | sqlite3 new.db"

Now testing the new.db (as a fresh zll.db) ... seems to work .....

Smanar, 2021-03-09:
And you have not just a previous working backup ?

erikproper, 2021-03-09:
No. I do have backups, but I have a suspicion (also looking at the old crap
I found in the tables ...) that the database has been corrupt for a long
time.
So the older backups would not have helped me.

I did, from time to time, notice "funny" things in the sense that changes
of names of sensors or lights were gone after a reboot.

Does deCONZ check the integrity of the database, or does it "simply" trust
sqlite?
For instance, at startup time a check if the KEYs are actually unique?

Smanar, 2021-03-09:
>Does deCONZ check the integrity of the database, or does it "simply" trust
sqlite?
For instance, at startup time a check if the KEYs are actually unique?

Good question, or a possible improvement, but on normal use, it shouldn't happen  for me.

erikproper, 2021-03-09:
True, but how robust is (transaction-wise) sqlite? What happens if someone switches of the power of a raspberry when the database is being updated? Could that lead to a corrupted index? And thus possibly violations of key uniqueness.

Of course, switching off a raspberry pi (or any computer) without a proper shutdown is always a bad idea. However, ... I did have some power failures at my place (due to construction work).