dresden-elektronik/deconz-rest-plugin #4543: Adding new devices is not possible anymore

alex1082a, 2021-03-08:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi 
 - Running method: (Raspbian 
 - Firmware version: (2660700)
 - deCONZ version: (2.05.84)
 - Device: ConBee II / 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? No.

## deCONZ Logs

## Additional context

SEK-DE, 2021-03-09:
An update of your deCONZ should solve the problem i guess. 
The latest stable version is the 2.9.3., or try the beta 2.10.2 
https://github.com/dresden-elektronik/deconz-rest-plugin/releases

Mimiix, 2021-03-09:
There is no info here. Closing.

Please add logs in a new issue.