dresden-elektronik/deconz-rest-plugin #4544: SLC SmartOne S57003 - Wall 4xMono

svhelge, 2021-03-09:
The device (https://www.tlg.no/s57003-lp) is a variant of the 4 channel Suncricher remote. Other variants:
  * Namron | 4CH K8 4-gang (4512703)
  * EcoDim 
  * iCasa

## Device

- Product name: SLC SmartOne S57003 - Wall 4xMono
- Manufacturer: THE LIGHT GROUP AS
- Model identifier: S57003
- Device type : Please remove all unrelated device types. 
   - Remote
   - Switch

## Screenshots

Mimiix, 2021-03-09:
Can you add screenshots of all clusters?

svhelge, 2021-03-09:
I added screenshots of
- Power configuration cluster
- Diagnostic cluster
- Identify cluster
- ZLL commissioning cluster
anything else needed?

Mimiix, 2021-03-09:
I think this is fine. @SwoopX / @Smanar Can you check?

svhelge, 2021-03-16:
@Mimiix , anything still missing? Or can the "Incomplete issue" label be removed?

svhelge, 2021-04-07:
I created PR #4711 yesterday in order to resolve the issue.

closed 2021-04-30 by svhelge