dresden-elektronik/deconz-rest-plugin #4545: Aqara Led Bulbs not controllable after deCONZ restart

iisworks, 2021-03-09:
## Describe the bug
After a deCONZ reboot, some of my Aqara Led Bulbs (lumi.light.aqcn02) are unresponsive and there will be no dimlevel/color temp sliders visible on the right of the bulbs  in Phoscon. I think this happens to the ones that are turned off during reboot. It does respond to On/Off in deCONz vnc. This can always be solved by reading the Basic cluster in deCONZ using vnc once, the sliders will show up in Phoscon and the lights can be controlled from then on.

## Steps to reproduce the behavior
  1. All aqara lights are fine
  2. Restart deCONZ
  3. Some lights can't be controlled
  4. Observed error

## Expected behavior
Lights should be controllable

## Screenshots
[image]

[image]

[image]

## Environment
 - Syno DS218+ NAS
 - Running method: Marthoc Docker container 
 - deCONZ version: 2.10.2
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
[aqara.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6106350/aqara.log)

## Additional context

Mimiix, 2021-03-09:
06:20:56:133 0x588E81FFFED38C76 error APSDE-DATA.confirm: 0xE9 on task

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

That probably causes it. But i am not sure how to fix it. I'll ask manup to check this issue.

SwoopX, 2021-03-09:
I guess the real question here is if they properly report and if they are polled. Polling should instantly get the status of mains powered devices or within a few seconds after a deconz restart.

iisworks, 2021-03-09:
Not sure if it holds any clue, but it seems that the ones that were turned on when deCONZ restarted will still work afterwards. In any case, about half of them stop responding after a restart.

Mimiix, 2021-03-09:
@SwoopX It reminds me of the Xiaomi plug that isn't responding after areboot.

iisworks, 2021-03-09:
Small update: it turns out I was still running 2.10.1, updated the docker image to latest 2.10.2 now. Same issue though.
- All lights that were turned off are uncontrollable in Phoscon after reboot
- Somehow I could now see the Dim/WhiteTemp sliders but they did nothing
- I did notice that I could turn the light on using the on/off cluster, after that still not from Phoscon
- Only after Basic - Read the bulb becomes controllable in Phoscon

Same for all 9 Aqara bulbs.

iisworks, 2021-03-10:
BTW: I just added 2 Osram Smart+ bulbs (Dimmable Light 6 and 17) and created a group "Osram". I noticed they also act weird. The group name is gone after deCONZ reboot (called "Group 18" after a reboot instead of "Osram"), the bulbs don't react to group commands in Phoscon (one turns on, the other won't) while individually they work OK when turning on/off. So I'm wondering if something else is going on here?

iisworks, 2021-03-12:
Another update: to test I reverted to 2.7.1, but it also has the exact same issue.

Mimiix, 2021-03-12:
Already asked Swoop and Manup to check on this one.

SwoopX, 2021-03-14:
I'd need like 10 mins of logs with dbg_info=2. I'm after those lines for the light:

```
06:20:35:841 ZCL attribute report 0x00158D000520E69E for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
06:20:35:842 0x00158D000520E69E extract Xiaomi special attribute 0xFF01
06:20:35:855 	03 Device temperature 38 °C
06:20:35:856 	05 RSSI dB (?) 43 (0x002B)
06:20:35:856 	07 unknown 0 (0x0000000000000000)
06:20:35:857 	08 unknown 279 (0x0117)
06:20:35:858 	09 unknown 3844 (0x0F04)
06:20:35:859 	0a Parent NWK 0 (0x0000)
06:20:35:860 	64 lift 0 (100%)
06:20:35:860 	64 somke/gas density 0 (0x00)
06:20:35:861 	65 unknown 25 (0x19)
06:20:35:862 	66 unknown 345 (0x0159)
```

Also, I'd require a screenshot of the node info panel to know the manufacturer code.

It appears certain Xiaomi mains powered devices are not automatically polled, although they should...

iisworks, 2021-03-14:
Does this have to be a log when the error is still ongoing, or would any time be OK? How do I download a larger log file from the Docker container? I used vnc Help - Debug View to copy parts of the log before, but this only seems to work for shorter parts of logging.

wvuyk, 2021-03-15:
Looks like this one could be very much related to #4249 ? The aqara curtain motor need a read too before it reports?

Mimiix, 2021-03-16:
I think it could be related to the fix @SwoopX did.  https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4588

That one :)

iisworks, 2021-03-16:
Thanks, I will check 2.10.3 when the docker image becomes available!

wvuyk, 2021-03-16:
> I think it could be related to the fix @SwoopX did. #4588
> 
> That one :)

We will test it for #4249 , but that is a battery device? Not sure....

wvuyk, 2021-03-16:
> > I think it could be related to the fix @SwoopX did. #4588
> > That one :)
> 
> We will test it for #4249 , but that is a battery device? Not sure....

This update did not yet resolve #4249 So I am sure something else is happening here. Like I asked before, what can we do in Windows to provide more info?

iisworks, 2021-03-19:
Glad to report that https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4588 in 2.10.3 beta has fixed the issue with my Aqara bulbs! I rebooted twice now, and all 9 bulbs are availalbe, yay! Thanks a lot!

closed 2021-03-19 by iisworks