dresden-elektronik/deconz-rest-plugin #4550: Info for FLS-PP lp

jokerigno, 2021-03-09:
## Describe the question or issue you are having

I bought this some time ago to use it inside Home Assistant. I don't understand why there are two lights in HA, while I have one entity in Phoscon app.

This produce an issue: I cannot manage white level correctly because one light in HA is RGB; the other has just brightness. 

So for example if I create a light group no white level is present and brightness change level on RGB and White. So it's impossible to have full saturated color.

Please let me know if you need some other info .

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Mimiix, 2021-03-10:
@jokerigno Can you show us what the restapi exposes?

jokerigno, 2021-03-10:
Here's the full log taken from Deconz

[image]

https://pastebin.com/SLwKz2jF

Looking at the light there are two lights actually. But why?

[image]

Mimiix, 2021-03-10:
Ah that explains.  That's the channels of the device. That's the reason you see 2 devices.

I have to ask @manup or @SwoopX to explain why this is done.

manup, 2021-03-10:
On the REST-API level the FLS-PP lp is indeed presented as two lights for historical reasons to make the white channel controllable separately. In the Phoscon App merges these two two to show a slightly more dense UI.

To get a fully saturated color only the first color light should be used. The dimmable light, which is the white slider in the Phoscon App, needs to be turned off so that it doesn't desaturate the color.

jokerigno, 2021-03-13:
thank you @manup for the feedback but in my humble opinion having 2 different approach in REST-API vs Phoscon does not make any sense. Infact in Phoscon you can still controll white channel separately.

Your suggestion is much more complicated to manage in home assistant UI  and via voice control:

-  you cannot create a group light because the white light does not expose white level, just brightness and this means that changing brightness in  a group light with change both white and rgb brightness
- you cannot create template light because as told before white light does not expose white level and this produce error

The only solution is via automation but if not the best approach.

Can we align the same behavior from Phoscon to Rest-Api?

jokerigno, 2021-03-18:
Hi, 

can you share your thoughts on my proposal?

manup, 2021-03-20:
>[...]having 2 different approach in REST-API vs Phoscon does not make any sense.

Phoscon is just a visual representation, any REST API client is free to present the resources from the API as desired. For example sensors also work this way, in the API they are separated for temperature, humidity etc. but in UIs usually shown as one combined device.

I understand the complications, the problem is that there is no such thing as a real white channel in the API and also not on Zigbee level. In the API there are only color lights and color temperature lights which both don't follow a channel based representation but only have an interface for color temperature in kelvin, xy color coordinates and hue / saturation.

The very first firmware version of the FLS-PP lp didn't have a separate dimable light representing the white channel, the white level was implicitly calculated based from the selected color saturation and brightness. But then we got heaps of complains from people who wanted to control the white channel explicitly. Based on that feedback we introduced the second dimable light endpoint (the only way to implement that in a Zigbee conform way).

This way of representing RGBW has been later copied by various other controllers as well.

To be frank there is no real good solution in the bounds of Zigbee and providing full control.

> * you cannot create a group light because the white light does not expose white level, just brightness and this means that changing brightness in a group light with change both white and rgb [...]

I think this is a current limitation with what HA UI widget provide and that it would be better to have another UI widget that support such multi resource lights in a suitable way.

Small side note: The Hue API also delivers the same separation for various controllers following the FLS approach. Meaning than used with a Hue bridge the same issue exist and it would be nice to have such a widget in the repertoire for both APIs.

>Can we align the same behavior from Phoscon to Rest-Api?

I would propose to solve this on the client side, by providing a widget which represents the two resources as single light and a slider for white which just controls the brightness of the second resource.

Beside from showing a FLS-PP lp as one light the Phoscon App creates helper groups to support group control of the white channel. For each group like 'Living Room' there are 2 helper groups one for RGB part and one for white channel, making it possible to control a group of FLS there only RGB or white channel is affected.

jokerigno, 2021-03-21:
Thank you @manup for the simple and clear response. I will report this to HA hoping that something can be changed at integration level.

Mimiix, 2021-03-23:
It's funny, as i have one at home right now, mine works as expected in HA. Yes, i have 2 lights in there but the color selector does what you expect it to do.

jokerigno, 2021-03-23:
So you can manage "white" indipendenly? Can you share your code? I tried with light.group but brightness act on both lights. I tested template light but for me it does not work (set color in particular)

PS: sorry for the offtopic but I don't know how to write you.

Mimiix, 2021-03-23:
I just used the sliders on the thing. I am currently testing it before attaching it to some place in my room.

jokerigno, 2021-03-23:
Slider on the thing? 

Which slider?

Are you referring to this product?
https://zigbee.blakadder.com/dresden_elektronik_FLS-PP.html

Mimiix, 2021-03-23:
Yes. _Why would i reply if it isn't the same device :)?_

[image]

That one :)

jokerigno, 2021-03-23:
Ok but still I don't get hit. Sorry. That slide is color temperature, not white level. In my setup doesn't do anything (on the rgb entity).

Mimiix, 2021-03-23:
White level is color temp, right?

jokerigno, 2021-03-23:
Nope. White level is white level and color temp is color temp :)

Or at least for me. Have you tried it?

Let me know if you are able to set a full saturated color via HA ui controlling just one entity.

jokerigno, 2021-03-30:
Hey @Mimiix can I ask you how the white light create by the integration behave?

My light is called Dimmable light 10 and this is attributes:

`        '10': {'etag': 'f21171ac19122534cd30530fd32c4291',
               'hascolor': False,
               'lastannounced': '2021-01-14T03:31:35Z',
               'lastseen': '2021-02-26T16:43Z',
               'manufacturername': 'dresden elektronik',
               'modelid': 'FLS-PP3 White',
               'name': 'Dimmable light 10',
               'powerup': 7,
               'state': {'alert': 'none',
                         'bri': 219,
                         'on': False,
                         'reachable': True},
               'swversion': '0214.201000EB',
               'type': 'Dimmable light',
               'uniqueid': '00:21:2e:ff:ff:05:8f:ea-0b'},`

Every time I change brightness via HA I see this error:

> 2021-03-23 18:57:24 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True, 'hue': 7127, 'sat': 118, 'bri': 219}" to "192.168.2.202 /lights/10/state"
> 2021-03-23 18:57:24 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'error': {'address': '/lights/10/state',
> 'description': 'parameter, hue, not available',
> 'type': 6}},
> {'error': {'address': '/lights/10/state',
> 'description': 'parameter, sat, not available',
> 'type': 6}},
> {'success': {'/lights/10/state/on': True}},
> {'success': {'/lights/10/state/bri': 219}}]
> 2021-03-23 18:57:24 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"10","r":"lights","state":{"alert":null,"bri":219,"on":true,"reachable":true},"t":"event","uniqueid":"00:21:2e:ff:ff:05:8f:ea-0b"}
> 2021-03-23 18:57:24 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139885700299600] /lights/10/state parameter, hue, not available
> Traceback (most recent call last):
> File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 141, in handle_call_service
> await hass.services.async_call(
> File "/usr/src/homeassistant/homeassistant/core.py", line 1488, in async_call
> task.result()
> File "/usr/src/homeassistant/homeassistant/core.py", line 1523, in _execute_service
> await handler.job.target(service_call)
> File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
> await self.hass.helpers.service.entity_service_call(
> File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 642, in entity_service_call
> future.result() # pop exception if have
> File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 681, in async_request_call
> await coro
> File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 679, in _handle_entity_call
> await result
> File "/usr/src/homeassistant/homeassistant/components/light/init.py", line 235, in async_handle_light_on_service
> await light.async_turn_on(**params)
> File "/usr/src/homeassistant/homeassistant/components/deconz/light.py", line 202, in async_turn_on
> await self._device.async_set_state(data)
> File "/usr/local/lib/python3.8/site-packages/pydeconz/deconzdevice.py", line 38, in async_set_state
> await self.async_set(field, data)
> File "/usr/local/lib/python3.8/site-packages/pydeconz/api.py", line 118, in async_set
> await self._request("put", field, json=data)
> File "/usr/local/lib/python3.8/site-packages/pydeconz/gateway.py", line 109, in request
> _raise_on_error(response)
> File "/usr/local/lib/python3.8/site-packages/pydeconz/gateway.py", line 225, in _raise_on_error
> raise_error(data["error"])
> File "/usr/local/lib/python3.8/site-packages/pydeconz/errors.py", line 58, in raise_error
> raise cls("{} {}".format(error["address"], error["description"]))
> pydeconz.errors.RequestError: /lights/10/state parameter, hue, not available

Why it seems that it wants hue values?

Mimiix, 2021-03-31:
That could be related to another issue. 

I really am annoyed by the other light(the white channel), because it doesn't seem to turn off properly. No clue whats going on with that. Need to have some look with Manup on it.

jokerigno, 2021-04-01:
Yes definitely is not well managed.Let's hope that will be fixed :)

Mimiix, 2021-04-01:
@jokerigno  I spoke to @manup. Users complained that they wanted W seperate. So can't please anyone. 

He will check this weekend if the old code is still in the firmware to set the mode (so we have 1 for all , or have W as a seperate light like it is now).

jokerigno, 2021-04-01:
That would be G R E A T ! 

#fingercrossed

jokerigno, 2021-04-12:
Sorry for my insistence but I was wondering if anyone have an update on this.

jokerigno, 2021-04-19:
@Mimiix any update on this?

Mimiix, 2021-04-19:
I've asked manup.

Mimiix, 2021-05-11:
It was put in the list for next month

akafester, 2021-06-09:
@manup I believe we have talked before about an installation in Sweden. Would it be possible to have a switch or different firmware where the white channel gets calculated rather than independent?

akafester, 2021-06-09:
@manup Is it possible to get a switch or firmware where the driver could calculate the white channel rather than seperate it? 
It would be so useful in bigger installations.

Mimiix, 2021-07-01:
Wanted to reply yesterday but i forgot.

Manup gave me an old firmware. This does include the 1 channel on api. However, it stopped responding to collor changes now all together 😅  Not sure how to make it work again now. So i can't recommend that one.

He told me it is going to be a feature request so it won't be anytime soon i'm affraid.

jokerigno, 2021-07-01:
Ok no problem.  I just wrote to avoid the closure of this issue.

Mimiix, 2021-07-23:
Keeping this open

Noodleyman, 2021-08-03:
Adding a +1 to this thread, stumbled across it whilst setting up my device and having the EXACT same questions as raised in this thread.

Mimiix, 2021-09-15:
Okay, so i have some news. After i used it , it seemed broken. Spend some time with it and i decided to reset it. Now it seems to have only one channel.  Meaning that the firmware he send me _does_ work after resetting the device.

So i'll check in again after a few days and see whats going on.

Mimiix, 2021-09-29:
Seems to be working nicely here. If anyone can reach out to me so we can test together that would be wonderful.

bluemoehre, 2023-10-17:
Just found this topic due I'm facing the same issue right now in my setup. 
**Is here still being worked on?**

Other controllers from different manufacturers are just seen as a single device and somehow use another approach where in the UI is a switch to control White Color / Full Color spaces.