dresden-elektronik/deconz-rest-plugin #4555: conbee II updating via windows above 2.05.77 version 26680700 (to Beta) ?

FDirckx, 2021-03-10:
## Describe the question or issue you are having
i have no clue if im in the right spot for asking this

sorry if more ppl did ask for it i did get stuck on this part.

i have the conbee II stick and i did update it via windows to the 2.05.77 version 26680700

i want to put it to the beta versio but cant find a higher version than what i have now ConBee II | 0x26680700

i want to update it because i have some sonoff sensors that i want to use but my stick is not finding it.

Thanks Frank.

## Screenshots
[image]

 - Host system: (NAS / unraid)
 - Running method: (Unraid / Home Assistent deCONZ Add-on / Windows updater)
 - Firmware version: (0x26680700)
 - deCONZ version: (2.05.77)
 - Device: ConBee II
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Mimiix, 2021-03-10:
Firmware ! = software of deconz.

FDirckx, 2021-03-10:
Thx you made me think difrent hehe

(mijn unraid versie wou niet update door een verkeerde Repository)

Thanks Dennis

Mimiix, 2021-03-10:
Great! Good to hear that :) Geen dank 😉