dresden-elektronik/deconz-rest-plugin #4557: Lidl Silvercrest Smart Plug (UK)

evan-shaw, 2021-03-10:
## Device

- Product name: Lidl Silvercrest Smart Plug
- Manufacturer: Silvercrest
- Model identifier: Heiman  TS011F / _TZ3000_00mk2xzy  Adapter plug
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots

### Basic
As can be seen above, these are recognised as being 2 different kinds of devices. (i.e. I have 2)
Also they are recognised as lights and not switches.

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-03-11:
Missing some screens/ Differences?

evan-shaw, 2021-03-11:
[image]

The plugs, as far as I can tell, i.e. outside appearance, packaging box and labelling, are all identical, but here's the difference in Phoscon.

Mimiix, 2021-03-11:
@SwoopX You integrated it, right?

evan-shaw, 2021-03-11:
Yes they work fine. Just thought the identification was a bit weird.

ebaauw, 2021-03-11:
Probably just need an entry in:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/db673839f8cf4ec867bb9471cf79477299c5a794/de_web_plugin.cpp#L480

evan-shaw, 2021-03-11:
The first 4 devices you see in my screen capture above, should all be the same type and model.

Mimiix, 2021-03-11:
Can you try to read the Basic cluster in deCONZ of those devices?

They aren't switches, as switches are something that "trigger" something else.

ebaauw, 2021-03-11:
> The first 4 devices you see in my screen capture above, should all be the same type and model.

The third one is probably listed as "Heiman" because of the mac address prefix.  Same for the lightstrip at the bottom of the screenshot.  I don't understand where the "Adapter plug" Product comes from.

> But also, shouldn't they appear in the "Switches" list and not "Lights"?

Wired switches and plugs, used to smartify dumb devices, are exposed as `/lights` resources.  They look, smell, and feel like smart lights (from a Zigbee perspective), and are exposed like these.

Wireless switches, used to control ("trigger") smart devices, are exposed as `/sensors` resources.  They look, smell, and feel completely differently from smart lights, and are exposed differently.

evan-shaw, 2021-03-11:
> Can you try to read the Basic cluster in deCONZ of those devices?
>                                                                                 Can you tell me how to do that please?
> They aren't switches, as switches are something that "trigger" something else.

Mimiix, 2021-03-11:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

evan-shaw, 2021-03-12:
I've tried earlier this morning to get this info without success. I basically can't start the deconz-gui.
I'm running my RPi headless as it oes quite a few other things. So use RealVNC for X access.
I stop deconz - systemctl stop deconz
start deconz-gui - systemctl start deconz-gui

But the deconz-gui won't attach to my VNC X display, so fails.
I've tried a few things without success. 

_systemd[1]: deconz-gui.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
systemd[1]: deconz-gui.service: Service RestartSec=30s expired, scheduling restart.
systemd[1]: deconz-gui.service: Scheduled restart job, restart counter is at 2.
systemd[1]: Stopped deCONZ: ZigBee gateway -- GUI/REST API.
systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
deCONZ[4428]: qt.qpa.screen: QXcbConnection: Could not connect to display :0
deCONZ[4428]: Could not connect to any X display_

I even tried changing the DISPLAY value to :1 in deconz-gui.service as my X DISPLAY in VNC is :1.0, but got the same as above with a :1 instead of :0

So sorry, I cannot get the cluster info, unless there is a way of getting this from some logs?

senilio, 2021-03-12:
For getting X forwarding up and running, please take a look at https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Access-deCONZ-GUI-in-headless-setups

senilio, 2021-03-12:
Feel free to jump on to discord and ping me there, and I'll try to get you sorted.

evan-shaw, 2021-03-12:
Thanks to Senilo for the deconz-gui help.
Here's the basic cluster info for all 4 devices.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2021-03-12:
Seems to be paired incorrectly.

closed 2021-04-10 by github-actions[bot]