dresden-elektronik/deconz-rest-plugin #4559: IKEA lights: unreliable turn ON command

QedC, 2021-03-11:
## Describe the bug
Turning these on by deconz or home assistant is unreliable: the light always turns on, but it randomly stays at 1% brightness, instead of 100%. In these cases, Deconz shows in the web interface as if the light was at 100%.
The failure rate seems higher if the turn on has a transition time. For instance, when setting up a motion sensor triggering the turn on in deconz.
The lights always turn off correctly.

## Steps to reproduce the behavior
1) Turn on the light using a single command. E.g. switch or automation in deconz.

## Expected behavior
The light should turn on at the expected brightness, instead of 1%

## Screenshots
n/a

## Environment
Conbee II, Firmware 26660700
2x TRADFRI bulb E26 WS opal 1000lm
1x TRADFRI bulb E26 WS clear 950lm

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: (2.09.03)
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context
I replaced these lights with Philips hue, without changing any other configuration or location and I did not have any issue.

Mimiix, 2021-03-11:
what command do you send to the ikea light to turn it on? I haven't seen such behavior on my devices.

QedC, 2021-03-11:
Any command. It happens more often when I set an automation in deconz:

Motion sensor triggering a light, as below. You can see there is a built in transition time.

[image]

Mimiix, 2021-03-11:
No, i mean the HA command you send.

QedC, 2021-03-11:
The issue is there even without using HA at all, all contained in deconz.

I now have the automation above augmented by 2 additional standard motion sensor turn on in HA to send a total of 3 commands per motion and have a more reliable turn on command.

Mimiix, 2021-03-11:
Without anything solid, i can't try and reproduce it. I have 0 issues with Ikea lights and deCONZ. 

So if you want any help, help me to help you.

QedC, 2021-03-11:
I misunderstood the question. I was able to reproduce the issue with the service call below. But, it is about once every 10 calls. The deconz automation has the issue more frequently.

service: light.turn_on
entity: light.ingresso_lampadario
field: "/state"
data:
  'on': true
  'transitiontime': 9

I can post a screenshot of the network as well if that helps.

Mimiix, 2021-03-11:
I'll test it when i get home.

What i do remember, is that IKEA lights can't do much at the same time. That's why there's a delay. i think @ebaauw wrote something about this earlier.

QedC, 2021-03-11:
Thanks! I see and that is a shame. There are no alternative 1000lm Zigbee alternatives sold in Japan. 

I don't have a TRÅDFRI gateway to try. I am curious to see how they implemented the turn on/off.

Mimiix, 2021-03-11:
I agree with you. I checked with Robban (Kane610, dev of HA integration) and he has a fix for it in how to do it. That's why it doesn't occur as often with HA. 

I think this is the catch of having cheaper bulbs. 

Phoscon should implement this fix on their own.

timovos, 2021-03-17:
I've had similar issues with Ikea GU10 bulb. I wrote a script that changes the brightness of the bulbs after two seconds. I run this script after I turn on the lights with an automation
`kapschuur_deuren_aan:
  alias: Kapschuur deuren brightness
  sequence:
  - delay: 00:00:02
  - service: light.turn_on
    data:
      brightness: 254
    entity_id: light.kapschuur_deur_links
  - service: light.turn_on
    data:
      brightness: 254
    entity_id: light.kapschuur_deur_rechts
  mode: single`

QedC, 2021-03-17:
That is a smart way of doing it. I wish deconz had that in the integration.

With my setup, deconz + 2x Home assistant turn on commands. I got a single failure to turn on (stuck at 1%) in over 200 events. But it is quite cumbersome to implement.

ebaauw, 2021-03-17:
> I wish deconz had that in the integration.

It does, actually: https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/rules/

ebaauw, 2021-03-17:
Hm, that documentation is ancient.  You’d want condition with a `ddx` operator on `state.on` For the functionality of the above script.  Best check out the Hue documentation for that, https://developers.meethue.com/develop/hue-api/6-rules-api/.

QedC, 2021-03-18:
> > I wish deconz had that in the integration.
> 
> It does, actually: https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/rules/

Can you please clarify? Out of the box, after you pair an IKEA light it does not work properly.

ebaauw, 2021-03-18:
IKEA lights work differently from e.g. Hue lights; they have different firmwares.  Plenty of issues describing that:
- IKEA lights don’t accept a new command while the transition of the previous command is still running.  Workaround: use a `transitiontime` of 0 or send separate commands for the different attributes (effectively what the abovementioned script does);
- Not sure, but I think IKEA light allow the brightness level when turning on to be configured through the _On Level_ attribute.  Make sure this is set to 255 (for previous level) to mimic the Hue behaviour.

My remark is a reaction to the script: you can also do that in the deCONZ API, without the need for an external system for automations.

QedC, 2021-03-18:
I understand and it makes sense. 

What I am saying that IKEA lights are listed as supported in the deconz ([here](https://phoscon.de/en/conbee2/compatible)). But they don't work reliably. You have to set up an automation in HA to make them work properly and there is still a small chance that they will fail.

My point is that deconz should either implement y default the workaround on the turn on that HA has, or do not list them as compatible.

Mimiix, 2021-03-18:
> I understand and it makes sense.
> 
> What I am saying that IKEA lights are listed as supported in the deconz ([here](https://phoscon.de/en/conbee2/compatible)). But they don't work reliably. You have to set up an automation in HA to make them work properly and there is still a small chance that they will fail.
> 
> My point is that deconz should either implement by default the workaround on the turn on that HA has, or do not list them as compatible.

That's the definition of "compatible". My ikea lights work as i expect them to work, without any changes.

QedC, 2021-03-18:
I am lost here, is this a bug or not?

Mimiix, 2021-03-18:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4559#issuecomment-801670394 

In the Ikea firmware.

QedC, 2021-03-18:
Yeah, but what I can see as customer is that I set an automation in Deconz and it does not work...

Mimiix, 2021-03-18:
Because the light you use with it contains a bug.  @ebaauw provides you guidance to work around it, but in order to get it solved you need to ask IKEA.

QedC, 2021-03-18:
> I agree with you. I checked with Robban (Kane610, dev of HA integration) and he has a fix for it in how to do it. That's why it doesn't occur as often with HA.
> 
> I think this is the catch of having cheaper bulbs.
> 
> **Phoscon should implement this fix on their own.**

I might have misunderstood your previous statement then.

closed 2021-04-16 by github-actions[bot]