dresden-elektronik/deconz-rest-plugin #4560: Xfinity Keypad XHK1-UE

a1ad, 2021-03-11:
## Device

- Product name: Xfinity Keypad XHK1-UE / URC4450BC0-X-R  
- Manufacturer: Universal Electronics Inc
- Model identifier: URC4450BC0-X-R
- Device type : 
  - Switch (i think)

## Screenshots

  Required screenshots:
  - Endpoints and clusters of the node:
[image]

  - Node Info panel
[image]

### Power Configuration
[image]

### Identify
[image]

### Poll Control
[image]

### Temperature measurement
[image]

### ISA Zone
[image]

### Diagnostics
[image]

### OTAU
[image]

### IAS ACE
[image]
[image]
[image]

[Here](https://www.reddit.com/r/homeassistant/comments/k1hg4w/physical_zigbee_alarm_keypad_integrated_into_home/) is a Reddit post from someone that successfully added the keypad to zigbee2mqtt and with some more info.

Greets,

Mimiix, 2021-03-11:
@Smanar Would you like this one?

Smanar, 2021-03-11:
Hu will need to create more field.

I have searched some information about this device, I think we need to create/use as field
- config/armed = true/false
- state/alert = last action like "armed" "emergency"

But do we need state/alarm ?

IDK  it s a good idea to use this device as ZHAalarm, as he don't need to trigger alarm but more send armed/disarmed infoirmation.

And there is some comment about temperature and sensor ?

a1ad, 2021-03-11:
If it can send the 4 digit code to home assist that would be a great start. Let me know if i can help with something.

Smanar, 2021-03-12:
I think it s possible from that I m reading about it.
Do you have a linux machine to test code modification ?

a1ad, 2021-03-12:
Ofc, i have a fedora workstation or plenty of lxc servers on my proxmox host. The conbee is connected to the PI3B

Smanar, 2021-03-12:
Ok so first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you

```
sudo apt install deconz-dev
git clone --branch keypad https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

This code will do nothing, just create the entry in the API.
If you can show me the entry created to check how decone reconize the device ?
I can explain how to use the API if you need ?

a1ad, 2021-03-12:
Just on the pi or do i need a new Linux machine? I don't want to lose my ZB network of course.

Smanar, 2021-03-12:
As you want, just need a linux OS (not docker), raspbian or other.
But yes if you have your production network on the proxmox, better to use the pi, on a new network, you can just switch the conbee 10mn, time of tests, and switch it again after.

a1ad, 2021-03-12:
> Ok so first try
>
> You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
> So for you
>
> ```
> sudo apt install [...]

Yeah, i need more info about the API, compilation is ok

Smanar, 2021-03-12:
for API
long story > https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/
short story > http://IP:PORT/api/KEY/sensors

IP and PORT are the same than for phoscon
KEY is an API key, if you haven't one

`curl http://127.0.0.1:80/api/ -X POST -d '{"devicetype": "TestKey"}'`

a1ad, 2021-03-12:
There is no info in /sensors other than:
`{
  "1": {
    "config": {
      "configured": true,
      "on": true,
      "sunriseoffset": 30,
      "sunsetoffset": -30
    },
    "etag": "f306ae97bf6113b349f25bbb8354d014",
    "manufacturername": "Philips",
    "modelid": "PHDL00",
    "name": "Daylight",
    "state": {
      "dark": true,
      "daylight": false,
      "lastupdated": "2021-03-12T22:26:43.353",
      "status": 230,
      "sunrise": "2021-03-12T06:02:55",
      "sunset": "2021-03-12T17:41:17"
    },
    "swversion": "1.0",
    "type": "Daylight",
    "uniqueid": "00:21:2e:ff:ff:04:09:f0-01"
  }
}`

Smanar, 2021-03-13:
>The port was 8080 and not 80 here

If you are right, will be the same than you are using for phoscon.

>edit: I am using the beta in production (due to sonof mini ZB) and the current stable on the test server.

The code is based on the last code on official, so will be the beta too.
But you are using another machine for test ?

Can you share the logs during inclusion ? On deconz/help/log with "info" and "info_l2" ?

a1ad, 2021-03-13:
> If you are right, will be the same than you are using for phoscon.
>
>
> The code is based on the last code on official, so will be the beta too.
> But you are using another machine for test ?
>
> Can you [...]

Yes i use another VM, just to be safe.., i didnt need to include the device, when i switched the conbee to the new VM the devices where all there.

Smanar, 2021-03-13:
>i didnt need to include the device, when i switched the conbee to the new VM the devices where all there.

You need, your device are still in the zigbee network (because it will be the same network, same mac adress, same network key, ect ....) but the device are not in the API.

a1ad, 2021-03-13:
```json
  "2": {
    "config": {
      "battery": null,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "71a295870418824acd1faea34c51b15a",
    "lastseen": "2021-03-13T15:32Z",
    "manufacturername": "Universal Electronics Inc",
    "modelid": "URC4450BC0-X-R",
    "name": "Temperature 2",
    "state": {
      "lastupdated": "none",
      "temperature": null
    },
    "type": "ZHATemperature",
    "uniqueid": "00:0d:6f:00:13:4f:61:39-01-0402"
  }
```

Smanar, 2021-03-13:
Ok, nice, and no ZHAAlarm one ?
will continu the code.

a1ad, 2021-03-13:
Nope
[image]

Thanks!

Smanar, 2021-03-13:
Ok do have updated the code

```
cd deconz-rest-plugin
git pull
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Now you will have a new sensor a ZHAswitch with a field "armed"

But I have lot of question :
- does this device need ias enrollement.
- do we need to process handleIasAceClusterIndication() 

Can you try the new code, use the device to arm/disarm for exemple and show logs during the command pls ?

a1ad, 2021-03-13:
@Smanar how to obtain logs?

Smanar, 2021-03-13:
on deconz / help / log with "info" and "info_l2"

a1ad, 2021-03-13:
Do i need to reinclude the device?

Smanar, 2021-03-13:
Yep, pls.

a1ad, 2021-03-13:
I don't think its working, i only see the temp sensor.

Smanar, 2021-03-13:
I have checked the code again, nothing special, can you share the log during the inclusion ?

a1ad, 2021-03-13:
I cant include the device anymore... tried a bunch of stuff already

a1ad, 2021-03-13:
hmm the temp sensor is paired again, but no new sensor found. No logs, i was too late to enable the logs.
I can only pair the sensor when i select search lights? Does it matter?

Smanar, 2021-03-13:
I don't think there is a difference beetween sensor or light.
As the device is already included n phoscon you can trigger inclusion easily
- set phoscon in permit join.*
- read again basic attribute on deconz (cluster 0x0000, button "read", need to fill the table)

This trick will trigger an inclusion.

Enable log before setting in permit join, on the dev machine you have few devices, will be not a problem.

a1ad, 2021-03-13:
i cant find a setting in phoscon to permit join?? In deconz there is one.
So i set permit join for 255 and read the basic attr successful, but no new sensor.

the logs are full with this stuff:
`20:04:08:295 enqueue event config/localtime for /config/
20:04:08:295 poll node 80:4b:50:ff:fe:b5:65:d8-01
20:04:08:295 Poll light node On/Off light 2
20:04:08:295 Idle timer triggered
20:04:08:296 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
20:04:08:296 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
20:04:08:296 Force binding of attribute reporting for node Temperature 2
20:04:08:403 Poll APS request to 0x804B50FFFEB565D8 cluster: 0x0006 dropped, values are fresh enough
20:04:09:269 enqueue event config/localtime for /config/
20:04:09:270 poll node 84:2e:14:ff:fe:a9:44:74-01
20:04:09:270 Poll light node On/Off plug-in unit 3
20:04:09:270 Idle timer triggered
20:04:09:270 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
20:04:09:270 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
20:04:09:271 Force binding of attribute reporting for node Temperature 2
20:04:09:321 Poll APS request to 0x842E14FFFEA94474 cluster: 0x0006 dropped, values are fresh enough
20:04:10:268 poll node 80:4b:50:ff:fe:b3:4e:bb-01
20:04:10:268 Poll light node On/Off light 5
20:04:10:269 enqueue event config/localtime for /config/
20:04:10:269 Idle timer triggered
20:04:10:270 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
20:04:10:270 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
20:04:10:270 Force binding of attribute reporting for node Temperature 2
20:04:10:320 Poll APS request to 0x804B50FFFEB34EBB cluster: 0x0006 dropped, values are fresh enough
20:04:11:267 poll node 84:71:27:ff:fe:26:10:0a-01
20:04:11:267 Poll light node Range extender 7
20:04:11:269 enqueue event config/localtime for /config/
20:04:11:270 Idle timer triggered
20:04:11:270 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
20:04:11:271 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
20:04:11:271 Force binding of attribute reporting for node Temperature 2
20:04:12:162 poll node 80:4b:50:ff:fe:b7:1b:7a-01
20:04:12:163 Poll light node On/Off light 6
20:04:12:216 Poll APS request to 0x804B50FFFEB71B7A cluster: 0x0006 dropped, values are fresh enough
20:04:12:270 enqueue event config/localtime for /config/
20:04:12:271 Idle timer triggered
20:04:12:271 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
20:04:12:271 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
20:04:12:272 Force binding of attribute reporting for node Temperature 2
20:04:13:123 poll node 80:4b:50:ff:fe:b5:65:d8-01
20:04:13:123 Poll light node On/Off light 2
20:04:13:172 Poll APS request to 0x804B50FFFEB565D8 cluster: 0x0006 dropped, values are fresh enough
20:04:13:271 enqueue event config/localtime for /config/
20:04:13:271 Idle timer triggered
20:04:13:271 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
20:04:13:271 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
20:04:13:272 Force binding of attribute reporting for node Temperature 2
20:04:14:094 poll node 84:2e:14:ff:fe:a9:44:74-01
20:04:14:095 Poll light node On/Off plug-in unit 3
20:04:14:146 Poll APS request to 0x842E14FFFEA94474 cluster: 0x0006 dropped, values are fresh enough
20:04:14:270 enqueue event config/localtime for /config/
20:04:14:270 Idle timer triggered
20:04:14:271 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
20:04:14:271 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
20:04:14:271 Force binding of attribute reporting for node Temperature 2
20:04:15:092 poll node 80:4b:50:ff:fe:b3:4e:bb-01
20:04:15:092 Poll light node On/Off light 5
20:04:15:140 Poll APS request to 0x804B50FFFEB34EBB cluster: 0x0006 dropped, values are fresh enough
20:04:15:270 Daylight now: nauticalDusk, status: 220, daylight: 0, dark: 1
20:04:15:270 enqueue event config/localtime for /config/
20:04:15:271 Idle timer triggered
20:04:15:271 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
20:04:15:272 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
20:04:15:272 Force binding of attribute reporting for node Temperature 2
20:04:16:088 poll node 84:71:27:ff:fe:26:10:0a-01
20:04:16:088 Poll light node Range extender 7
20:04:16:270 enqueue event config/localtime for /config/
20:04:16:271 Idle timer triggered
20:04:16:272 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
20:04:16:272 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
20:04:16:272 Force binding of attribute reporting for node Temperature 2
20:04:16:928 poll node 80:4b:50:ff:fe:b7:1b:7a-01
20:04:16:928 Poll light node On/Off light 6
20:04:16:975 Poll APS request to 0x804B50FFFEB71B7A cluster: 0x0006 dropped, values are fresh enough
20:04:17:270 enqueue event config/localtime for /config/
20:04:17:270 Idle timer triggered
20:04:17:270 Force read attributes for node On/Off light 6
20:04:17:270 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
20:04:17:271 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
[... cut ...]
20:04:17:595 Erase task req-id: 9, type: 21 zcl seqno: 186 send time 0, profileId: 0x0104, clusterId: 0x0004
20:04:17:674 verified group capacity: 255 and group count: 1 of LightNode 0x804b50fffeb71b7a
20:04:17:675 0x804b50fffeb71b7a found group 0xFFF0
20:04:17:846 poll node 80:4b:50:ff:fe:b5:65:d8-01
20:04:17:846 Poll light node On/Off light 2
20:04:17:898 Poll APS request to 0x804B50FFFEB565D8 cluster: 0x0006 dropped, values are fresh enough
20:04:18:269 enqueue event config/localtime for /config/
20:04:18:270 Wait 1s till query finished
20:04:18:770 poll node 84:2e:14:ff:fe:a9:44:74-01`

Smanar, 2021-03-14:
```
i cant find a setting in phoscon to permit join?? In deconz there is one.
So i set permit join for 255 and read the basic attr successful, but no new sensor.
```

In phoscon it s "add new sensor/light"
If you make the permit join using directly deconz, it will not use the API, the device can't be included.

But fortunately, I have this ind of problem on an other device too, will add some debug line to your branch.

Smanar, 2021-03-14:
@a1ad have pull a special version with some more debug line, if you can try it ?

 -   set phoscon in permit join (add new sensor)
 -   read again basic attribute on deconz (cluster 0x0000, button "read", need to fill the table)

a1ad, 2021-03-14:
```json
"2": {
    "config": {
      "battery": null,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "67c497dc2c5c49ff01d58308c20bd4e2",
    "lastseen": "2021-03-14T10:52Z",
    "manufacturername": "Universal Electronics Inc",
    "modelid": "URC4450BC0-X-R",
    "name": "Temperature 2",
    "state": {
      "lastupdated": "2021-03-13T17:57:21.281",
      "temperature": 2700
    },
    "type": "ZHATemperature",
    "uniqueid": "00:0d:6f:00:13:4f:61:39-01-0402"
  },
  "3": {
    "config": {
      "armed": null,
      "battery": null,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "3e478d66fbff54ed19e14ba095ab79b5",
    "lastseen": "2021-03-14T10:52Z",
    "manufacturername": "Universal Electronics Inc",
    "mode": 1,
    "modelid": "URC4450BC0-X-R",
    "name": "Switch 3",
    "state": {
      "buttonevent": null,
      "lastupdated": "none"
    },
    "type": "ZHASwitch",
    "uniqueid": "00:0d:6f:00:13:4f:61:39-01-0501"
  }
```
Now we got the switch, again forget the logs ... i am so used to have a log file ...

Smanar, 2021-03-14:
So no need more log ^^

Information are fine, here you have the command list
```
#define CMD_ARM 0x00
#define CMD_BYPASS 0x01
#define CMD_EMERGENCY 0x02
#define CMD_FIRE 0x03
#define CMD_PANIC 0x04
#define CMD_GET_ZONE_ID_MAP 0x05
#define CMD_GET_ZONE_INFORMATION 0x06
#define CMD_GET_PANEL_STATUS 0x07
#define CMD_GET_BYPASSED_ZONE_LIST 0x08
#define CMD_GET_ZONE_STATUS 0x09
```

the one you have was CMD_GET_PANEL_STATUS , I think you will have different according the command you make on the device.
Can you share me the log when making a complete code ?
From that I have read the device send the code, and it s you that send the arm or disarm order to the device after checking.

a1ad, 2021-03-14:
Yes correct.
Its just pretty hard to obtain the logs because there is just so much excessive "noise" in there. I am going to try tho.

Can't you disable the logging for the other devices in this build?

Smanar, 2021-03-14:
It s possible but not managed by deconz, it s for that I m making special log starting by "Debug Keypad" to find them with search feature.

a1ad, 2021-03-14:
This is the only debug keypad found in the log:
Pressed a lot of digits

15:54:21:621 Debug Keypad : Address 0x000D6F00134F6139 Payload , command 0x07
15:54:21:621 [INFO] - No button map for: URC4450BC0-X-R, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: 0x07, payload: None, zclSeq: 113

Smanar, 2021-03-14:
Ok so in fact it's normal, there is no payload for this command

> 
> 8.3.2.4.6 Get Panel Status Command
> This command is used by ACE clients to request an update to the status (e.g., security system arm state) of the
> ACE server (i.e., the IAS CIE). In particular, this command is useful for battery-powered ACE clients with
> polling rates longer than the ZigBee standard check-in rate.
> 8.3.2.4.6.1 Payload Format
> There is no payload for the Get Panel Status command.
> 8.3.2.4.6.2 Ef fect on Receipt
> On receipt of this command, the ACE server responds with the status of the security system. The IAS ACE
> server SHALL generate a Get Panel Status Response command.

a1ad, 2021-03-14:
Yeah, but no log entry when pushing the buttons?

Smanar, 2021-03-14:
Yep , the keyboard is waiting for a command from deconz.
I have just making it, from reading documentation, so it s totaly experimental ^^.

I think just after that we will have more data.

Smanar, 2021-03-15:
No more return in log with last version ?

a1ad, 2021-03-15:
> No more return in log with last version ?

Ow, did you make a new version. I didn't got that. Let me check later today.

a1ad, 2021-03-15:
Relevant log files:

22:09:00:952 Debug Keypad : Address 0x000D6F00134F6139 Payload , command 0x07
22:09:00:953 Debug Keypad : Sending answer
22:09:00:953 [INFO] - No button map for: URC4450BC0-X-R, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: 0x07, payload: None, zclSeq: 114

22:09:15:966 Debug Keypad : Address 0x000D6F00134F6139 Payload , command 0x07
22:09:15:966 Debug Keypad : Sending answer
22:09:15:967 [INFO] - No button map for: URC4450BC0-X-R, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: 0x07, payload: None, zclSeq: 120

22:08:42:144 Debug Keypad : Address 0x000D6F00134F6139 Payload , command 0x07
22:08:42:145 Debug Keypad : Sending answer
22:08:42:145 enqueue event config/reachable for /sensors/2
22:08:42:146 [INFO] - No button map for: URC4450BC0-X-R, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: 0x07, payload: None, zclSeq: 110

Smanar, 2021-03-16:
Nothing more ?

Can you log but this APS ? we will have zigbee echanges like that.

>22:09:15:966 Debug Keypad : Sending answer
when you see that I m sending an answer to the device, I was hoping for an answer ....

I have find a sniff from another device https://github.com/Koenkk/zigbee2mqtt/issues/6242#issuecomment-778577448

Can you try too with the new code ? I have disabled the ZclFCDisableDefaultResponse (was false on the link)

a1ad, 2021-03-16:
Nope: only this msg when the proximity senor is activated:
22:41:40:626 Debug Keypad : Address 0x000D6F00134F6139 Payload , command 0x07
22:41:40:627 Debug Keypad : Sending answer
22:41:40:628 [INFO] - No button map for: URC4450BC0-X-R, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: 0x07, payload: None, zclSeq: 77

a1ad, 2021-03-17:
More news: you are right with the enrollmend!
When i press the 3 buttons on top, nothing happens the same log:
21:26:21:926 Debug Keypad : Address 0x000D6F00134F6139 Payload , command 0x07
21:26:21:926 Debug Keypad : Sending answer
21:26:21:926 [INFO] - No button map for: URC4450BC0-X-R, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: 0x07, payload: None, zclSeq: 60

But after i press 4 random digits we get a new entry in the log:
21:27:02:621 Debug Keypad : Address 0x000D6F00134F6139 Payload 02043131313100, command 0x00
21:27:02:622 [INFO] - No button map for: URC4450BC0-X-R, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: ARM (0x00), payload: 02043131313100, zclSeq: 62

Smanar, 2021-03-17:
Yeah ^^, I will continu tommorow but yes, I think some data are the digit.

There is a notification when you press a number, but the final notification is only when the code is finished.

But BTW you need the enrollement ? It don't work without it ?

a1ad, 2021-03-17:
> Yeah ^^, I will continu tommorow but yes, I think some data are the digit.
> 
> There is a notification when you press a number, but the final notification is only when the code is finished.
> 
> But BTW you need the enrollement ? It don't work without it ?

It started to work after the enrollment. Can i reset the enrollment to try again?

Smanar, 2021-03-18:
>It started to work after the enrollment. Can i reset the enrollment to try again?

Sure but not the priority.

Have updated the code to display more detail, and I hope the code, but I have something strange

payload: 00 04 38 38 38 38 00

00 is the ARM
the last 00 is the zone ID so it make for code 04 38 38 38 38, 38 can be the ascii value for "8", but what can be 04 ?

a1ad, 2021-03-18:
Yes! Nice work:

22:21:39:874 Debug Keypad : Address 0x000D6F00134F6139 Payload 00043131313100, command 0x00
22:21:39:874 Debug Keypad : Arm command, size 7
22:21:39:874 Debug Keypad : Arm command, Arm mode 0, code 1111, Zone id0
22:21:39:875 [INFO] - No button map for: URC4450BC0-X-R, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: ARM (0x00), payload: 00043131313100, zclSeq: 14

22:22:21:380 Debug Keypad : Address 0x000D6F00134F6139 Payload 01043535353500, command 0x00
22:22:21:380 Debug Keypad : Arm command, size 7
22:22:21:380 Debug Keypad : Arm command, Arm mode 1, code 5555, Zone id0
22:22:21:381 [INFO] - No button map for: URC4450BC0-X-R, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: ARM (0x00), payload: 01043535353500, zclSeq: 18

04 can be the zone status?

Smanar, 2021-03-18:
On my side I think 04 is the code len, but this part can be device dependent.

I m making a roolback on the code to create ZHAalarm entry, like that you will have the enrollement + the tamper field.

Can you try the new code, and show me the entry created, the JSON.

a1ad, 2021-03-18:
```json
"2": {
    "config": {
      "battery": null,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "19275a6612752d2f4502dd50aaa6fe21",
    "lastseen": "2021-03-18T17:57Z",
    "manufacturername": "Universal Electronics Inc",
    "modelid": "URC4450BC0-X-R",
    "name": "Temperature 2",
    "state": {
      "lastupdated": "2021-03-17T21:19:02.810",
      "temperature": 2200
    },
    "type": "ZHATemperature",
    "uniqueid": "00:0d:6f:00:13:4f:61:39-01-0402"
  },
  "3": {
    "config": {
      "armed": null,
      "battery": null,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "8f036646e27f684ecc52d6ad5636e360",
    "lastseen": "2021-03-18T17:57Z",
    "manufacturername": "Universal Electronics Inc",
    "mode": 1,
    "modelid": "URC4450BC0-X-R",
    "name": "URC4450BC0-X-R 3",
    "state": {
      "buttonevent": null,
      "lastupdated": "none"
    },
    "type": "ZHASwitch",
    "uniqueid": "00:0d:6f:00:13:4f:61:39-01-0501"
  }
```
Is this what you expected? The Switch node is gone now in deconz, only temp2

Smanar, 2021-03-18:
You have re make an inclusion ?
And the ZHAswitch is still on your capture ?

a1ad, 2021-03-18:
Yes still in the logs:
19:06:27:704 Debug Keypad : Address 0x000D6F00134F6139 Payload 00043131313100, command 0x00
19:06:27:704 Debug Keypad : Arm command, size 7
19:06:27:704 Debug Keypad : Arm command, Arm mode 0, code 1111, Zone id0
19:06:27:704 [INFO] - No button map for: URC4450BC0-X-R, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: ARM (0x00), payload: 00043131313100, zclSeq: 72

i tried to repair the device like you said: enable new sensors in phoscon and read basic.

Smanar, 2021-03-18:
Your machine is standard linux or docker style ?

I m talking with swoops, I think I will let the ZHAswitch and put all options on it, tamper, armed, action. Will see if your device can support that withour enrollment.

a1ad, 2021-03-18:
Wait, after reboot i got more:
    "config": {
      "armed": null,
      "battery": null,
      "enrolled": 0,
      "on": true,
      "pending": [],
      "reachable": true
    },

Smanar, 2021-03-19:
>Can i have your branch and the beta on one machine? And how do i switch between both.

You can just compile the lib, and switch them on the same deconz version.
I can too update the code to the last beta, so you can use only 1 deconz version ?

>Wait, after reboot i got more:

Hu ? So why we need a reboot ?
BTW it seem we have a mutant device with a sensor combinaison of ZHAalarm and ZHAswitch, we will keep it to test the sensor capacity.

So I have updated the code.
Have added a field state/action, it will have the last code used (just need restart, no need to re -include the device), will be visible in websocket notification
Have added the command state/armed = rue / false, to arm/disarm the device

a1ad, 2021-03-19:
Yes please, i need the beta for a device.

I just close deconz, compile, copy lib and reopen deconz, i guess thats not enough...after the restart i got more json values. Maybe browser cache, i dont know.

I ll test the new code later.

Smanar, 2021-03-19:
K, next version will be based on the last official from github

>Edit: what do you mean with: will be visible in websocket notification

If I m right you will have a websocket notification when typing a code on the keypad, a field `state/action=armed,1234,78`

a1ad, 2021-03-19:
How do i access the websocket notification??

Smanar, 2021-03-19:
Ha right, sorry, if you don't use third app, you can use just deconz logging, or a browser plugin, on my side I m using "Browser websocket client" on firefox with this kind of url `ws://192.168.1.1:8088`

a1ad, 2021-03-20:
I dont know how to access the websocket connection. I have Simple WebSocket Client for firefox, when i enter: ws://127.0.0.1:8080 i get undefined as error msg.

[234 more replies]

closed 2021-05-09 by github-actions[bot]