dresden-elektronik/deconz-rest-plugin #4561: Unable to update Deconz

marithpl, 2021-03-11:
## Describe the bug
I'm using a Deconz 2.09.01, I wish to update to 2.10.2 but I have some problems.
I've updated a docker image, run a docker container and after that I have to create a new Conbee II password.
When I connect to VNC I see all devices as hex (like 000x0 or sth like that).

What is the problem and what is the best way to make an update.

Deconz works on Synology DSM 7.0 Docker Container

## Screenshots

[image]

## Environment

 - Host system: NAS Synology
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.09.01 / 23.12.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs
```
date	stream	content
2021-03-11 13:30:38	stdout	[marthoc/deconz] VNC port: 5901
2021-03-11 13:30:38	stdout	[marthoc/deconz] Websockets port: 443
2021-03-11 13:30:38	stdout	[marthoc/deconz] Web UI port: 8081
2021-03-11 13:30:38	stdout	[marthoc/deconz] Current deCONZ version: 2.08.00
2021-03-11 13:30:38	stdout	[marthoc/deconz] Starting deCONZ...
2021-03-11 13:30:43	stdout	
2021-03-11 13:30:43	stdout	Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd :1 to connect to the VNC server.
2021-03-11 13:30:43	stdout	
2021-03-11 13:30:43	stdout	Log file is /root/.vnc/:1.log
2021-03-11 13:30:43	stdout	Starting applications specified in /etc/X11/Xvnc-session
2021-03-11 13:30:43	stdout	
2021-03-11 13:30:43	stdout	New ':1 (root)' desktop at :1 on machine 
2021-03-11 13:30:43	stdout	
2021-03-11 13:30:43	stdout	hostname: Name or service not known
2021-03-11 13:30:43	stdout	Cleaning stale x11 lock '/tmp/.X11-unix/X1'!
2021-03-11 13:30:43	stdout	Cleaning stale x11 lock '/tmp/.X1-lock'!
2021-03-11 13:30:43	stdout	Cleaning stale pidfile '/root/.vnc/:1.pid'!
2021-03-11 13:30:43	stdout	Killing Xtigervnc process ID 25... which was already dead
2021-03-11 13:30:43	stdout	hostname: Name or service not known
2021-03-11 13:30:46	stdout	14:30:46:983 HTTP Server listen on address 0.0.0.0, port: 8081, root: /usr/share/deCONZ/webapp/
2021-03-11 13:30:46	stdout	libpng warning: iCCP: known incorrect sRGB profile
2021-03-11 13:30:46	stdout	libpng warning: iCCP: known incorrect sRGB profile
2021-03-11 13:30:46	stdout	QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
2021-03-11 13:30:47	stdout	14:30:47:870 COM: /dev/ttyACM0 / serialno: , ConBee II
2021-03-11 13:30:47	stdout	14:30:47:655 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
2021-03-11 13:30:47	stdout	14:30:47:641 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
2021-03-11 13:30:47	stdout	14:30:47:638 found node plugin: libde_rest_plugin.so - REST API Plugin
2021-03-11 13:30:47	stdout	14:30:47:638 dlg action: Read binding table
2021-03-11 13:30:47	stdout	14:30:47:635 [INFO] - Button maps loaded.
2021-03-11 13:30:47	stdout	14:30:47:629 [WARNING] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
2021-03-11 13:30:47	stdout	14:30:47:627 [INFO] - Found file containing button maps. Parsing data...
2021-03-11 13:30:47	stdout	14:30:47:624 started websocket server at port 8080
2021-03-11 13:30:47	stdout	14:30:47:618 DB create temporary views
2021-03-11 13:30:47	stdout	14:30:47:618 DB cleanup
2021-03-11 13:30:47	stdout	14:30:47:617 DB PRAGMA user_version: 7
2021-03-11 13:30:47	stdout	14:30:47:617 DB file size 135168 bytes, free pages 0
2021-03-11 13:30:47	stdout	14:30:47:616 DB PRAGMA freelist_count: 0
2021-03-11 13:30:47	stdout	14:30:47:616 DB PRAGMA page_size: 4096
2021-03-11 13:30:47	stdout	14:30:47:615 DB PRAGMA page_count: 33
2021-03-11 13:30:47	stdout	14:30:47:614 DB sqlite version 3.27.2
2021-03-11 13:30:47	stdout	14:30:47:612 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
2021-03-11 13:30:47	stdout	14:30:47:612 gw run mode: docker
2021-03-11 13:30:47	stdout	14:30:47:611 parent process /bin/sh
2021-03-11 13:30:47	stdout	14:30:47:458 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
2021-03-11 13:30:47	stdout	14:30:47:186 CTRL. 3.27.214:30:47:457 COM: /dev/ttyACM0 / serialno: , ConBee II
2021-03-11 13:30:49	stdout	14:30:49:641 CTRL ANT_CTRL 0x03
2021-03-11 13:30:49	stdout	14:30:49:637 CTRL got nwk update id 0
2021-03-11 13:30:49	stdout	14:30:49:623 Current channel 15
2021-03-11 13:30:49	stdout	14:30:49:559 Device protocol version: 0x010C
2021-03-11 13:30:49	stdout	14:30:49:531 CTRL ANT_CTRL 0x03
2021-03-11 13:30:49	stdout	14:30:49:527 CTRL got nwk update id 0
2021-03-11 13:30:49	stdout	14:30:49:516 Current channel 15
2021-03-11 13:30:49	stdout	14:30:49:443 new node - ext: 0x00212effff06b12f, nwk: 0x0000
2021-03-11 13:30:49	stdout	14:30:49:437 Device protocol version: 0x010C
2021-03-11 13:30:49	stdout	14:30:49:383 unlocked max nodes: 200
2021-03-11 13:30:49	stdout	14:30:49:379 Device firmware version 0x26660700 ConBee II
2021-03-11 13:30:49	stdout	14:30:49:311 COM detected application
2021-03-11 13:30:49	stdout	(Make sure 'QTextCursor' is registered using qRegisterMetaType().)
2021-03-11 13:30:49	stdout	QObject::connect: Cannot queue arguments of type 'QTextCursor'
2021-03-11 13:30:49	stdout	(Make sure 'QTextBlock' is registered using qRegisterMetaType().)
2021-03-11 13:30:49	stdout	QObject::connect: Cannot queue arguments of type 'QTextBlock'
2021-03-11 13:30:49	stdout	14:30:49:293 COM check bootloader
2021-03-11 13:30:49	stdout	14:30:49:077 COM: /dev/ttyACM0 / serialno: , ConBee II
2021-03-11 13:30:50	stdout	14:30:50:503 ZDP discovery done in 819 ms14:30:50:585 new node - ext: 0x00158d0006793924, nwk: 0xEC5C
2021-03-11 13:30:51	stdout	14:30:51:051 ZDP discovery done in 1341 ms14:30:51:932 New websocket 192.168.68.129:58434 (state: 3) 
2021-03-11 13:30:52	stdout	14:30:52:997 Announced to internet https://phoscon.de/discover
2021-03-11 13:30:52	stdout	14:30:52:601 GW firmware version is up to date: 0x26660700
2021-03-11 13:30:52	stdout	14:30:52:601 GW firmware version: 0x26660700
2021-03-11 13:30:52	stdout	14:30:52:600 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF
2021-03-11 13:30:54	stdout	14:30:54:285 new node - ext: 0xccccccfffe5c4255, nwk: 0x2B53
2021-03-11 13:30:56	stdout	14:30:56:685 new node - ext: 0x00158d000283ec7a, nwk: 0x8604
2021-03-11 13:30:59	stdout	14:30:59:293 new node - ext: 0x00158d000522308f, nwk: 0x5F38
2021-03-11 13:31:00	stdout	14:31:00:982 New websocket 192.168.68.100:58002 (state: 3) 
2021-03-11 13:31:01	stdout	14:31:01:693 new node - ext: 0x847127fffe24ecb0, nwk: 0xE68F
2021-03-11 13:31:02	stdout	14:31:02:173 new node - ext: 0x00158d00050529a4, nwk: 0x33A0
2021-03-11 13:31:06	stdout	14:31:06:821 new node - ext: 0x00158d000633012f, nwk: 0xD717
2021-03-11 13:31:06	stdout	14:31:06:341 new node - ext: 0x00158d00028d90f5, nwk: 0xBD61
2021-03-11 13:31:09	stdout	14:31:09:701 new node - ext: 0x00158d0004ab43e4, nwk: 0x1E24
2021-03-11 13:31:09	stdout	14:31:09:221 new node - ext: 0x847127fffe24de20, nwk: 0x7938
2021-03-11 13:31:13	stdout	14:31:13:317 new node - ext: 0x000d6f00159a5f7f, nwk: 0x89FB
2021-03-11 13:31:14	stdout	14:31:14:349 ZCL attribute report 0xCCCCCCFFFE5C4255 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
2021-03-11 13:31:14	stdout	14:31:14:346 ZCL attribute report 0xCCCCCCFFFE5C4255 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
2021-03-11 13:31:17	stdout	14:31:17:453 new node - ext: 0x00158d0005445203, nwk: 0x4B12
```

## Additional context

Mimiix, 2021-03-11:
I close this issue, as this is not related to deCONZ itself. Also: you crossposted in the docker-deconz repository.

closed 2021-03-11 by Mimiix

marithpl, 2021-03-11:
So if is not Deconz or Docker-deconz related so where should I look for an answer?

Mimiix, 2021-03-11:
It is docker related :)

marithpl, 2021-03-11:
Ok i probably solved a problem.
I log in to Phoscone and import backup from old version.
Now Deconz API shows devices names.