dresden-elektronik/deconz-rest-plugin #4562: YSR-Mini-01

OnkelBuTscH, 2021-03-11:
## Device

- Product name: LED Controller YSR-Mini-01
- Manufacturer: YSRSAI
- Model identifier: ZB-CL01
- Device type : 
  - Other: LED Stripe Controller

## Screenshots

[image]
[image]
[image]
[image]

Hi,
I have a problem with the YSRSAI zigbee LED controller.
If I save the current white value as a scene, a wrong white value is set when i activate the scene.
It's the same with preset scenes night light or read etc...
Scenes with colors work properly.
Only scenes warm white - cold white are affected.
Link to the device: https://de.aliexpress.com/item/10000201716852.html?spm=a2g0o.productlist.0.0.424627b4s01Esv&algo_pvid=aa803659-f35f-4b76-9558-7279f4caaf3b&algo_expid=aa803659-f35f-4b76-9558-7279f4caaf3b-23&btsid=2100bdd816154701038204596ef3c8&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

Mimiix, 2021-03-11:
We need the screenshots from the device in deCONZ.

OnkelBuTscH, 2021-03-11:
unfortunately i have no access to deconz. 
I use the conbee 2 stick on a pi without a gui in ioBroker.
Or is there another way?

Mimiix, 2021-03-11:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Access-deCONZ-GUI-in-headless-setups

OnkelBuTscH, 2021-03-11:
I added screenshots of:

- Node Info
- Basic Cluster
- Identify Cluster
- Groups Cluster
- Scenes Cluster
- On/Off Cluster

popy2k14, 2021-03-18:
@OnkelBuTscH have the same controller but cant even pair it.
How did you pair it?

I followed the manual -> pressed over 5 seconds -> led stripe flashes but deconz cant find it.

Which version of deconz(phoscon do you have?

OnkelBuTscH, 2021-03-18:
> pressed over 5 seconds -> led stripe flashes two times and then stops but deconz cant find it.

I did the exact same thing.

> It also stops flashing when do not press "find lights" in deconz

I think it only flashes 2-3 times after pressing the button for 5 seconds that is normal. Then you can search in Phoscon.
After 10-20 seconds Phoscon found them.

The only thing I've changed is the Zigbee channel at 25.
But I don't think it has anything to do with it.?

Phoscon:
Version: 2.09.03 / 23.12.2020
Firmware: 26660700

popy2k14, 2021-03-18:
ok, thx for clearing things up, then that "flashing" behaviour should be ok.
Have read elsewhere that it flashes until it is paired.

Sadly, i have tried it often but i just does'nt join my network (pairing doesnt work).

The only difference is, you have an older conbee II firmware.
Mine are:

2.09.03 / 23.12.2020
26680700

Do you have an conbee II ?

OnkelBuTscH, 2021-03-18:
> Do you have an conbee II ?

Yes i have.

Unfortunately, I don't know what else could be the cause. :(
I have 6 led strips of it and none of them had this problem.

Do you have a hue bridge to test it?

popy2k14, 2021-03-18:
ok, so maybe the lower firmware makes the difference!?
Or mine is just defect.

Sadly no hue bridge, selled them all after the i was gone the deconz way :-)
This is also my first pairing issue with deconz.

popy2k14, 2021-03-18:
Another question, when it is not paired (like mine) and you connect it to power, does it light up white (the two white leds)?

OnkelBuTscH, 2021-03-18:
> when it is not paired (like mine) and you connect it to power, does it light up white (the two white leds)?

Yes, only warm white and cold white LED's light up if you only connects to power.

popy2k14, 2021-03-18:
Exactly the same here, just cant pair it :-(

popy2k14, 2021-03-18:
Just an idea:

is downgrading the firmware of my conbee II from 26680700 to 26660700 possible without loosing my network?
Or is it not recommended?

OnkelBuTscH, 2021-03-18:
> is downgrading the firmware of my conbee II from 26680700 to 26660700 possible without loosing my network?

Unfortunately I don't know, ^^ better ask about it in the discord.

OnkelBuTscH, 2021-03-18:
FYI, I upgraded the Firmware to 26680700.
Still no problem with pairing.
Maybe your LED Controller is damage?

popy2k14, 2021-03-19:
Tested the device on an friends hue bridge and it works without an issue. Found instantly and Rgb LEDs are working as they should.

So it seems that there incompatibility between 'my' device and deconz+conbee II.

The HUE bridge has detected it as :

Manufacturer: YSRSAI
Model: ZB-CL01
Sw: 1.0.7

How can I help to debug the issue?

popy2k14, 2021-03-20:
Hey guys, i have successfully paired it after downgrading deconz to 02.07.01.
Have raised an issue here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4615
and can provide a debug log when needed.

PS.: sorry for hijacking this thread and YES i have the same issue with scenes like @OnkelBuTscH

popy2k14, 2021-04-11:
Not solved

popy2k14, 2021-05-03:
Not solved

Smanar, 2021-05-03:
So even with the microSD and the extension cable you are not able to include new devices at all ?

popy2k14, 2021-05-03:
Sorry for confusion. Pairing works now, I have closed #4615. 

But the OP issue with scenes setting wrong white value is still a thing. Have now 3 of them, and all with the same issue.

popy2k14, 2021-05-25:
Not solved.

popy2k14, 2021-06-17:
Not solved.

popy2k14, 2021-08-24:
Bump, what is needed to get his working?

closed 2021-09-22 by github-actions[bot]