dresden-elektronik/deconz-rest-plugin #4564: SONOFF	01MINIZB goes in pairing mode alone

LeoeLeoeL, 2021-03-11:
## Describe the bug
 After some hours the device starts flashing its led and goes in pairing mode

## Steps to reproduce the behavior
 The issue happens always. I believed the device I had before (and gave back) was faulty but this one has the same problem

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian
 - Firmware version: (26680700)
 - deCONZ version: Version v2.10.2-beta (same issue with 2.9.x)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: yes -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? APC ups 

## deCONZ Logs

## Additional context

SwoopX, 2021-03-11:
That is sad to hear, but what do you expect us now to do?

LeoeLeoeL, 2021-03-12:
What could be to cause that?

Mimiix, 2021-03-12:
Without any logs or other details about your setup, it's hard to say.

LeoeLeoeL, 2021-03-12:
Logs could be very long. Any "key" to try to reduce them?
My setup is manifold.
I have Osram/Ledvance (lamps and plugs) , Ikea (lamps and extenders) , Lidl (plugs and lamps) and Vimar (covering shutters).
26/27 nodes.

Anyway, now I try to pair the Sonoff with the Hue Bridge in order to check if makes the same problem.

Zigbee bridges interfere even if at different channels.
I just added Sonoff to HUE and 4 lamps connected to Deconz switched on.

Mimiix, 2021-03-12:
Just provide them all in a pastebin :)

LeoeLeoeL, 2021-03-19:
I'm trying to save a log to show but I can't. DEBUG VIEW shows the latest 12 minutes only.
I tried to use the command "/usr/bin/deCONZ --dbg-info=2" but it opens another window of DECONZ "not connected" and the log is totally different from what DEBUG VIEW shows.

11:34:29:723 wait reconnect 8 seconds
11:34:29:727 enqueue event config/localtime for /config/
11:34:29:730 rule event /config/localtime: 11:34:28.726 -> 11:34:29.726 (1000ms)
11:34:30:723 wait reconnect 7 seconds
11:34:30:724 enqueue event config/localtime for /config/
11:34:30:727 rule event /config/localtime: 11:34:29.726 -> 11:34:30.724 (998ms)
11:34:31:224 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
11:34:31:724 wait reconnect 6 seconds
11:34:31:725 enqueue event config/localtime for /config/
11:34:31:727 rule event /config/localtime: 11:34:30.724 -> 11:34:31.725 (1001ms)
11:34:32:723 wait reconnect 5 seconds
11:34:32:724 enqueue event config/localtime for /config/
11:34:32:727 rule event /config/localtime: 11:34:31.725 -> 11:34:32.724 (999ms)
11:34:33:224 failed to reconnect to network try=7
11:34:33:723 wait reconnect 4 seconds
11:34:33:724 enqueue event config/localtime for /config/
11:34:33:727 rule event /config/localtime: 11:34:32.724 -> 11:34:33.724 (1000ms)
11:34:34:723 wait reconnect 3 seconds
11:34:34:725 enqueue event config/localtime for /config/
11:34:34:727 rule event /config/localtime: 11:34:33.724 -> 11:34:34.724 (1000ms)
11:34:35:725 wait reconnect 2 seconds
11:34:35:727 enqueue event config/localtime for /config/
11:34:35:729 rule event /config/localtime: 11:34:34.724 -> 11:34:35.727 (1003ms)
11:34:36:724 wait reconnect 1 seconds
11:34:36:803 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
11:34:36:805 COM: /dev/ttyS0 :  (0x0000/0x0000)
11:34:36:805 dev /dev/ttyS0 (/dev/ttyS0)
11:34:36:807 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
11:34:36:808 auto connect com /dev/ttyS0
11:34:36:809 enqueue event config/localtime for /config/
11:34:36:811 rule event /config/localtime: 11:34:35.727 -> 11:34:36.809 (1082ms)
11:34:36:816 COM check bootloader
11:34:37:270 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
11:34:37:272 COM: /dev/ttyS0 :  (0x0000/0x0000)
11:34:37:274 dev /dev/ttyS0 (/dev/ttyS0)
11:34:37:275 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
11:34:37:276 GW firmware update select USB device
11:34:37:723 Skip idle timer callback, too early: elapsed 914 msec
11:34:38:223 failed to reconnect to network try=8
11:34:38:723 enqueue event config/localtime for /config/
11:34:38:727 rule event /config/localtime: 11:34:36.809 -> 11:34:38.723 (1914ms)
11:34:38:827 COM timout query bootloader, assume application
11:34:38:828 Serial com connected
11:34:39:724 enqueue event config/localtime for /config/
11:34:39:727 rule event /config/localtime: 11:34:38.723 -> 11:34:39.724 (1001ms)
11:34:40:256 device state timeout ignored in state 2
11:34:40:724 enqueue event config/localtime for /config/
11:34:40:727 rule event /config/localtime: 11:34:39.724 -> 11:34:40.724 (1000ms)

Can you help me in order to print out the correct log to show here?

LeoeLeoeL, 2021-03-19:
Probably the issue happened not far from 17.03.
Before that, 4 lamps switched on alone : EKET BARS and LUCE DISIMPEGNO C1,C2,C3.
The name of ZBMINI is ON/OFF LIGHT 53
I hope I cought it in t
[Log_1.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6172986/Log_1.txt)
his log

LeoeLeoeL, 2021-03-20:
Anyway, the issue happens regularly only on DECONZ while on HUE BRIDGE is ok.
So that, I think this thread has to be modified from USER QUESTION to BUG REPORT.

Smanar, 2021-03-20:
I know an user with a similar issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2707#issuecomment-803428538

LeoeLeoeL, 2021-03-20:
[Log_2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6176391/Log_2.txt)

This log contains for sure the moment when the ZBMINI goes in pairing mode.
If there isn't anything so the issue is not logged in INFO_L2 but could be somewhere else. Where? Suggestions?

Mimiix, 2021-03-23:
This is open @LeoeLeoeL. Either way, opening a new issue because you didn't get a fast reply nor you agree with the tag is not appreciated. In the weekend, less people are around here. 

My reasoning is easy: If this was a bug, others (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4016) would have reported so. 

I've asked Manup to check this issue.

LeoeLeoeL, 2021-03-23:
I post another log for another "unpairing"
[Log_3.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6192587/Log_3.txt)

Mimiix, 2021-03-24:
Wrong button. My appologies. I've asked Manup to check this issue :)

LeoeLeoeL, 2021-03-25:
Another log.
I expect somebody to tell me if he needs more informations in the log.
[Log_4.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6208358/Log_4.txt)

Mimiix, 2021-03-26:
@LeoeLeoeL I've already forwarded, but he didn't reply yet (as you can see).

LeoeLeoeL, 2021-03-26:
It's strange nobody else suffers it.
Anyway, I put a 2nd ZBMINI in the network; Nice to know if they will be kicked out together or one per time.

LeoeLeoeL, 2021-03-27:
Now I know because this issue is so "rare".
I moved the ZBMINI from the "test position" (floor minus 1) to a room (floor 1). For info, Conbee II in the middle (floor 0).
Then I opened a new ZBMINI and put in the test position.
Well, the device in the new room didn't lose the connection until now (more than 24 hours).
The device in the test position lost the network after few hours.
This is the log when it went in pairing mode:
[Log_5.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6215996/Log_5.txt)

Today I added 2 more ZBMINI in floor 1 (now there are 3) and opened a fourth I put in the test position.
Now I'm waiting to see what happens.

Smanar, 2021-03-27:
But I don't see any "pairing mode" or permit join on your log ?

LeoeLeoeL, 2021-03-27:
Sorry I forgot to say the name of the device disconnected in Log_5 is Luce Esterna Mansarda.
Log_5 has INFO and INFO_L2 flagged only.
Should Pairing mode and permit join be there or I have to flag something else?

Smanar, 2021-03-27:
When the network enter in permit join, it s visible on "info"

[image]

LeoeLeoeL, 2021-03-27:
The network doesn't enter in permit join.
Only the device goes in paring mode.

Smanar, 2021-03-27:
I m not sure a device can enter in pairing mode alone, the whole network need to be in permit join, and the coordinator too, a device alone can do nothing except making some LED blinking.

LeoeLeoeL, 2021-03-27:
But it's happening. ZBIMINI goes in paring mode. Perhaps a device kicks it out.

Smanar, 2021-03-27:
For me it just blink.
If it go in pairing mode, it mean he was reseted to join a network, as the deconz is not in permit join, he can join it, so it will be kicked out the network.

LeoeLeoeL, 2021-03-27:
the device goes in pairing mode and needs ADD A LIGHT in Phoscon to rejoin.
This problem doesn't happend if ZBMINI joins my HUE hub.

LeoeLeoeL, 2021-03-29:
The device in the test position continues to be kicked out. 5 times since the last time I wrote here. More often during the day than in the night.
The other 3 devices (in the position I wrote before) never lost the pairing.
[Log_6.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6220786/Log_6.txt)
I think it's impossible a log doesn't log when a device is kicked out.

LeoeLeoeL, 2021-04-22:
The issue still exists.

LeoeLeoeL, 2021-05-20:
The issue still exists.
Maybe 2.12 will fix that?

LeoeLeoeL, 2021-06-13:
The issue happens in 2.12.too.
If other people with this device don't have this problem is because of a combination of devices.

LeoeLeoeL, 2021-07-06:
The problem still exists.
To solve I had to migrate most of Sonoff MiniZB to Hue bridge.

closed 2021-08-05 by github-actions[bot]