dresden-elektronik/deconz-rest-plugin #4565: Toggle Command is not working properly on some devices after update to 2.10.X beta

czm97176, 2021-03-12:
## Describe the bug

Subject: Toggle Command is not working properly on some devices after update to 2.10.X beta.

i had two window cover devices ZNCLDJ11LM which was working very smooth with deCONZ.  i was able to control the covering device from Node-Red by associating a hardware butto . However, the toggle command is not working properly after updated to beta version. it back to normal after falling back to v2.9.3 release version

## Steps to reproduce the behavior

1. Create a inject node to send true/false payload to the node
2. it will only work once, like Open-to-Close but not Close-to-open, not sure why it only works once.

  If the problem can't be reproduced and is sporadic, please provide some details
  on how often and when the issue happens.
-->

## Expected behavior

Work in both direction

## Screenshots

[image]

## Environment

 - Host system: (Raspberry)
 - Running method: (Marthoc Docker container)
 - Firmware version: (26680700)
 - deCONZ version: (2.10.2beta)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Mimiix, 2021-03-12:
This is a node-red issue, can you confirm that @zehir?

Zehir, 2021-03-12:
> This is a node-red issue, can you confirm that @Zehir?

I think not because the problem appears and disappear with the same node-red version but not the same deconz version.

Smanar, 2021-03-12:
But there is no "Toggle" command for covering ?

What is the command used by node-red ? Can be a problem is state ? eg using "open" command if the returned state is already 'opened" ?

Zehir, 2021-03-12:
> But there is no "Toggle" command for covering ?
> 
> What is the command used by node-red ? Can be a problem with state ? eg using "open" command if the returned state is already 'opened" ?

I check the code, he get the current value of `state.on` or `state.all_on` for groups and reverse it.

Smanar, 2021-03-12:
So can be the problem ?
He don't try to send a "open" command at a device already open that is sending the "close" state ?

So can need to reverse the sate, there is this kind of issue all month, have you the manufacture name to check for change ?

closed 2021-04-10 by github-actions[bot]