dresden-elektronik/deconz-rest-plugin #4566: Hue wall switch module.

ebaauw, 2021-03-12:
## Device

- Product name: Hue wall switch module
- Manufacturer: Signify Netherlands B.V. (previously known as Philips)
- Model identifier: RDM001
- Device type : Wireless (!) switch

See: https://www.philips-hue.com/en-gb/p/hue-philips-hue-wall-switch-module/8719514318045

While being positioned as an in-wall switch, this _is_ a wireless switch.
During installation, you disconnect the live and switch wires from your physical switch, and shortcut them using the included clamp, causing the outlet to be powered permanently.
You then connect the physical switch to the wall switch module, using the included wires.
The module has two inputs, supporting a single-rocker switch, a dual-rocker switch, or two single-rocker switches.
The input wires come in two lengths, presumably to connect a second single-rocker switch in another wall compartment (hope that's the correct term).

Basically, it's a device to smartify your 20th century wall switches.
This should get you some brownie points in the WAF area.

## Hue bridge resource
The resource looks familiar: 0xFC01 cluster to report the button actions.
The `capabilities` suggest it supports initial press (x000) and (short) release (x002) on each input.
However, I cannot seem to get an initial press event.

During my initial research on the Internet, I couldn't find whether the module supports momentarily switches, but looking at `config.devicemode`, which is newly introduced for this device, it does.
After setting it to `"dual pushbutton"`, I'm even seeing x001 and x003 events.
It doesn't look like the Hue app supports this setting (at least I cannot find it).
```
{
  "state": {
    "buttonevent": 2002,
    "lastupdated": "2021-03-12T11:20:30"
  },
  "swupdate": {
    "state": "noupdates",
    "lastinstall": "2021-03-12T11:19:27"
  },
  "config": {
    "on": true,
    "battery": null,
    "reachable": true,
    "devicemode": "dualrocker",
    "devicemodevalues": [
      "singlerocker",
      "singlepushbutton",
      "dualrocker",
      "dualpushbutton"
    ],
    "pending": []
  },
  "name": "Hue wall switch module 1",
  "type": "ZLLSwitch",
  "modelid": "RDM001",
  "manufacturername": "Signify Netherlands B.V.",
  "productname": "Hue wall switch module",
  "diversityid": "8ea98295-bec1-477f-bf00-94de51b30987",
  "swversion": "1.0.3",
  "uniqueid": "00:17:88:01:0b:00:05:69-01-fc00",
  "capabilities": {
    "certified": true,
    "primary": true,
    "inputs": [
      {
        "repeatintervals": [
          800
        ],
        "events": [
          {
            "buttonevent": 1000,
            "eventtype": "initial_press"
          },
          {
            "buttonevent": 1002,
            "eventtype": "short_release"
          }
        ]
      },
      {
        "repeatintervals": [
          800
        ],
        "events": [
          {
            "buttonevent": 2000,
            "eventtype": "initial_press"
          },
          {
            "buttonevent": 2002,
            "eventtype": "short_release"
          }
        ]
      }
    ]
  }
}
```

## Screenshots
The device looks very similar to the Hue smart button or the 2021 Hue dimmer.
This gives me good hopes, we'll be able to configure it in hybrid mode.

[image]

[image]

### Basic
Nothing spectacular here: it does support the 0x0031 Philips-specific _Configuration_ attribute.
Need to sniff the Hue bridge to see what they set it to.
[image]

### Power Configuration
Nothing spectacular here.  It's a 3V power cell.
[image]
Created binding to the coordinator without issue; the attribute reporting config was already set by the device (with the same values as Hue dimmer switch):
[image]

### Hue
The Hue-specific cluster works as for the Hue dimmer and smart button.
[image]

In rocker mode, the device issues an x000 followed immediately by an x002 (with a duration of 1) when connecting or disconnecting the wires.
So the module doesn't seem to differentiate between an "open" and "closed" state, it just reports state changes.

Need to sniff the Hue bridge to see how to set momentarily switch mode (they call it pushbutton), but I'm betting in this mode, it sends the regular x000, x001 per 0.8s, x003 sequence on connect (press), hold, disconnect (release).

### OTAU
Nothing spectacular here either.
The image type is 0x011c
[image]

### Client clusters
Client _On/Off_ and _Level Control_ seem to suggest the module can control (a group of) lights directly.
Need to experiment to confirm.
No clue how to set different groups for the different inputs, or even if this is possible.

Mimiix, 2021-03-12:
When the PR is ready, i'll forward it to manup.

ebaauw, 2021-03-12:
Setting `devicemode` uses `config.pending`, as this is a deep sleeper.  Once the module wakes, the Hue bridge sends a _Write Attributes_ for 0x0000/0x0034.  That's a new one, of type enum8.  The module responds with a _Write Attributes Response_ and a _Report Attributes_.  So we need to bind the _Basic_ cluster and setup attribute reporting for 0x0034.  Let's call that _Device Mode_.

The values are straightforward:

Setting | Value
--- | ---
`singlerocker` | 0x00
`singlepushbutton` | 0x01
`dualrocker` | 0x02
`dualpushbutton` | 0x03

Unfortunately not a bitmap, so you cannot set one input to rocker and the other to pushbutton.

ebaauw, 2021-03-12:
Above commits should fully support the Hue wall switch module:
```
{
  "config": {
    "battery": 100,
    "devicemode": "dualrocker",
    "on": true,
    "pending": [],
    "reachable": true
  },
  "ep": 1,
  "etag": "01173dc5b19bb0a976006eee8d0d3718",
  "lastseen": "2021-03-12T22:55Z",
  "manufacturername": "Signify Netherlands B.V.",
  "mode": 1,
  "modelid": "RDM001",
  "name": "RDM001 15",
  "state": {
    "buttonevent": 1002,
    "eventduration": 1,
    "lastupdated": "2021-03-12T22:21:20.017"
  },
  "swversion": "20210115",
  "type": "ZHASwitch",
  "uniqueid": "00:17:88:01:0b:00:05:5d-01-fc00"
}
```
When `config.devicemode` is changed through the API, it will show in `config.pending` until the value has been written to the device.  As the module is a deep sleeper, this might take up to two hours; speed this up by pressing a button to wake the device.  There should be web socket notifications for when `config.pending` is set and cleared.

snozzlebert, 2021-03-13:
Realy nice to see that @ebaauw already started implementing this. I will receive my Hue Wall Switch modules on Monday :) Thanks!

spookoly, 2021-03-13:
Very cool that this is already basically there. Ordered one as soon as it became available and it just turned up. Can't believe how fast you are @ebaauw ! Many many thanks.

Mimiix, 2021-03-13:
I'm planning to get this out in the next beta / release

ebaauw, 2021-03-14:
I think the PR is good to go.

closed 2021-03-15 by manup

ebaauw, 2021-04-02:
For completeness, the device can issue the following button events:

`devicemode` | first input | second input
--- | --- | ---
`singlerocker` | 1000, 1002 | --
`singlepushbutton` | 1000, 1001, 1002, 1003 | --
`dualrocker` | 1000, 1002 | 2000, 2002
`dualpushbutton` | 1000, 1001, 1002, 1003 | 2000, 2001, 2002, 2003

When configured as `rocker`, it will issue a x000/x002 each time you flip the rocker (to either position).  The `event duration` for the x002 is 1 (for 100ms), but `lastupdated` suggests it follows the x000 faster than that.

When configured as `pushbutton`, it will issue a x000/x002 sequence on press/release, and an x000/x001/.../x001/x003 on press/hold/release, similar to the Hue dimmer switch and the Hue smart button.  The `eventduration` increased with 8 for each x001, and they are issued pretty much every 800ms.

dworkin74, 2021-04-21:
I have added the device but it doesn't show up in the GUI.
Is that correct?

ebaauw, 2021-04-21:
If it doesn’t show in the GUI, you haven’t added it.  It could be hidden behind another node, or far away from the other nodes, making it difficult to locate.  Double check the _Node List_ panel to be sure.

dworkin74, 2021-04-21:
Sorry It is in Deconz but not in Phoscon

ebaauw, 2021-04-21:
Request support by Phoscon in the Phoson repo.

dworkin74, 2021-04-21:
I did now. Another question, I see duplicate events when pressing the button. I believe that is not what should happen. I set it to single push button.

ebaauw, 2021-04-21:
What events are you seeing where?

When set to push button, you should get a 1000 on press, and a 1002 on release; or a 1000 on press, a 1001 every 0.8s while holding, and a 1003 on release.

Short press:
```
Apr 21 19:33:53 pi5 dc_eventlog[817]: /sensors/15/state: {"buttonevent":1000,"eventduration":0,"lastupdated":"2021-04-21T17:33:53.975"}
Apr 21 19:33:54 pi5 dc_eventlog[817]: /sensors/15/state: {"buttonevent":1002,"eventduration":6,"lastupdated":"2021-04-21T17:33:54.589"}
```
Long press:
```
Apr 21 19:33:59 pi5 dc_eventlog[817]: /sensors/15/state: {"buttonevent":1000,"eventduration":0,"lastupdated":"2021-04-21T17:33:59.062"}
Apr 21 19:33:59 pi5 dc_eventlog[817]: /sensors/15/state: {"buttonevent":1001,"eventduration":8,"lastupdated":"2021-04-21T17:33:59.872"}
Apr 21 19:34:00 pi5 dc_eventlog[817]: /sensors/15/state: {"buttonevent":1001,"eventduration":16,"lastupdated":"2021-04-21T17:34:00.681"}
Apr 21 19:34:01 pi5 dc_eventlog[817]: /sensors/15/state: {"buttonevent":1001,"eventduration":24,"lastupdated":"2021-04-21T17:34:01.492"}
Apr 21 19:34:02 pi5 dc_eventlog[817]: /sensors/15/state: {"buttonevent":1001,"eventduration":32,"lastupdated":"2021-04-21T17:34:02.302"}
Apr 21 19:34:02 pi5 dc_eventlog[817]: /sensors/15/state: {"buttonevent":1003,"eventduration":35,"lastupdated":"2021-04-21T17:34:02.657"}
```

dworkin74, 2021-04-21:
It seems to send just 1 of every type instead of 2.
I had 2 wait awhile before the settings came through.
Thanks for the support

rileymd88, 2021-05-17:
Hi everyone - How did you manage to initially pair this device? I have tried clicking the following two ways without any success:

1. Press reset button once quickly and going to Phoscon -> Switches -> Add new switch -> Others
2. Holding the reset button for 10 seconds and going to Phoscon -> Switches -> Add new switch -> Others

In Phoscon it says that it did not find any new devices and I do not see any new devices within the DeCONZ map

apfelflo89, 2021-05-20:
@rileymd88 did you manage to get the module to pair?

rileymd88, 2021-05-20:
> @rileymd88 did you manage to get the module to pair?

@apfelflo89 Yes I did - It paired right away after I put one of the provided cables into the input...Not sure if this was the key or if it was just a coincidence.

TheFops, 2021-05-23:
> I'm afraid the module isn't meant for standalone use (without the Hue bridge), and simply doesn't support this, despite the client clusters. Bummer.

Is this still not possible or has it been implemented in a release?

ebaauw, 2021-05-23:
That would have to be a release of the wall switch module firmware.

TheFops, 2021-05-23:
@ebaauw OK so I'm not able to add it to deCONZ and use it blinded to my hue lights when the deCONZ server is down?

dseehof, 2021-05-25:
@rileymd88 How did you get it into Phoscon in the end? I tried all different Philips switches both in Beta and normal and also other, but nothing.

sladexos, 2021-05-25:
> @rileymd88 How did you get it into Phoscon in the end? I tried all different Philips switches both in Beta and normal and also other, but nothing.

Same here! It is very nice that we can use the wall switch module in Phoscon. But how can I get it into Phoscon. GW-Version is "2.11.05 / 9.5.2021" - After I try it 2 hours I don't get it into Phoscon... Can Anybody help?

apfelflo89, 2021-05-25:
They aren't supported by phoscon yet, they don't show up within the webguis, just within the deconz GUI on your client or within the webapi. 
So you might already paired it but it won't show. 
The LED blink while pairing so you might do a reset and then press the button/switch you connected. After success some of your lights should blink and show that the pairing succeeded 
So for me it worked with a reset and then pressing a or both bottons.

sebastien-k, 2021-07-08:
> 
> 
> > @rileymd88 did you manage to get the module to pair?
> 
> @apfelflo89 Yes I did - It paired right away after I put one of the provided cables into the input...Not sure if this was the key or if it was just a coincidence.

Hello,

Could you please be more specific and explain in detail how you managed to pair it? 
And how did you know you actually paired it ?
I'm struggling, thank you in advance !

deebeo-blip, 2021-07-21:
> Hello,
>
> Could you please be more specific and explain in detail how you managed to pair it?
> And how did you know you actually paired it ?
> I'm struggling, thank you in advance !

I'm highly interested in this as well. Question: Will it appear in the webgui once paired through deconz? I can see a lot more of these switches in my house so it would be very nice to include them via webgui like other switches.

rileymd88, 2021-07-21:
> I'm highly interested in this as well. Question: Will it appear in the webgui once paired through deconz? I can see a lot more of these switches in my house so it would be very nice to include them [...]

Hi @sebastien-k and @deebeo-blip 

I had the device sitting on my desk and then I plugged in the provided cable into the device and then starting the pairing process in Phoscon and then I pressed the button on the switch. As noted above these devices will currently NOT show up in Phoscon so you will have to check in DeCONZ itself to see if it paired successfully.

steffensammet, 2021-07-21:
Hi,

I just achieved to pair the switch (lucky me 😊). This is how I did:

1. Turn on pairing mode in Phoscon
2. On the switch, hold reset button for about 10 seconds. You will see the red light going on
3. Press a button that is connected to the switch 2 times
4. Smile 😊 

I hope this helps,
Steffen

fridward, 2021-08-24:
I am still lost:

1. Updated to phoscon 2.12.06 first. 
2. tried to pair it with switches - add new switch  - other
3. pushed reset button on the switch for 10 seconds
4. nothing is shown in the phoscon app
5. opening home assistant -one Rwl 021 is shown wit battery as the only entity ( I can not  get rid of the device in home assistant though - why ?)

....

when I  updated from 211.05 to 2.12.06 the switch was shown in phoscon, but when I removed it because I could not get it to work, I ws not able to pair it again.

fridward, 2021-08-25:
I was able to pair the wall switch again and even got to see it in phoscon, but I cannot do anything with it. different to the common hue dimmer I cannot setup a group to control a lamp. In deconz I do get strange data and in home assistant I can see the watch as RWL 121 but I do not have any entity at all (not even the battery entity anmore) :-( Actually the wall plug is in my opion the mot important product hue has to offer in years. Hopefully it will work sooner or later with conbee2.

[image]
[image]
[image]

Mimiix, 2021-08-25:
> I was able to pair the wall switch again and even got to see it in phoscon, but I cannot do anything with it. different to the common hue dimmer I cannot setup a group to control a lamp. In deconz I [...]
As @ebaauw mentioned in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4566#issuecomment-797513311

It does not support Grouping.

However, does it generate deconz_events in HA?

fridward, 2021-08-25:
> However, does it generate deconz_events in HA?

I am not sure if I do this correctly: I inserted "*" in developer tools/events/listening to and got a lot of events but no event refering to the switch

fridward, 2021-08-25:
when I hit  the switch (on/off) connected to the hue wall switch I get the following events:

Event 109 fired 10:52 AM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "rdm001_28",
        "unique_id": "00:17:88:01:0b:00:f4:e1",
        "event": 1002,
        "device_id": "c05fd4bf8839e09a97e232a373e256b6"
    },
    "origin": "LOCAL",
    "time_fired": "2021-08-25T08:52:41.756268+00:00",
    "context": {
        "id": "69af8e2476a3b4098bcbd0cbe336da4b",
        "parent_id": null,
        "user_id": null
    }
}
Event 108 fired 10:52 AM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "rdm001_28",
        "unique_id": "00:17:88:01:0b:00:f4:e1",
        "event": 1000,
        "device_id": "c05fd4bf8839e09a97e232a373e256b6"
    },
    "origin": "LOCAL",
    "time_fired": "2021-08-25T08:52:41.571140+00:00",
    "context": {
        "id": "9664c7960b21f65648afd5c9a72cb251",
        "parent_id": null,
        "user_id": null
    }
}

yves19, 2021-09-08:
No grouping, no binding despite clusters 0x0006 and 0x0008. The interest seems quite low.
It's more or less the equivalent of some IKEA scenario button launchers but is at leat twice more expensive

vladimir-aubrecht, 2021-11-19:
Just for another person who will find this thread, here is how I managed to pair it (just spent two days on it 🤦🏼‍♂️ ):

I had modules pre-paired with Philips Hue Bridge (Bridge S) and wanted to migrate them to another bridge (Bridge T).

0) Permit joining of new devices on bridge T.
1) I turned off bridge T.
2) I switched the channel on Bridge S to channel on which Bridge T is running.
3) I turned off bridge S.
4) I turned on bridge T.
5) I hold reset button on Philips Hue Wall Module. You must hold it long enough (LED must eventually go off, it's much longer then 10s ...). Make sure you hear click when pressing it.
6) When LED went off release reset button and quickly connect and disconnect white and yellow wires few times on the input 1.
7) Enjoy! 😊 

(Hold that reset button long enough!!)

spudje, 2021-12-04:
> When configured as `pushbutton`, it will issue a x000/x002 sequence on press/release, and an x000/x001/.../x001/x003 on press/hold/release, similar to the Hue dimmer switch and the Hue smart button. The `eventduration` increased with 8 for each x001, and they are issued pretty much every 800ms.

Why does it work like this? I would like to have behaviour that I only get 1 (one) event for hold (x001), just like with the IKEA Tradfri ON/OFF Switch. This continuous stream of events is unable to automate against...

leoluk, 2022-02-06:
Looks like it _does_ support controlling groups, but using a custom mechanism: https://github.com/Koenkk/zigbee2mqtt/issues/10740#issuecomment-1030935496

Managed to get mine to control a single group!

ebaauw, 2022-02-06:
Is that group hard-coded, or does it change resetting the module?  I cannot remember any specific setup by the Hue bridge, but this is certainly worth taking out the sniffer and re-examining the pairing with the Hue bridge.

leoluk, 2022-02-07:
It changes to a new random group when resetting!

ebaauw, 2022-02-07:
> Managed to get mine to control a single group!

Just to be sure: the wall switch module controls the group directly, even when zigbee2mqtt is not running?  Or does the zigbee2mqtt external converter issue the group commands?

ebaauw, 2022-02-07:
> the screenshot in the other threads shows the source NWK address as 0xd4f8, which is the device.

Yes, I saw that.  But is the MAC address in the IEEE 802.15.4 Data frame header also from the wall switch module?

I don't understand what magic zigbee2mqtt has done to get the device to send the group commands from _On/Off_.  Does it still send the commands from the FC00 cluster to the coordinator, in addition to the group commands?  When paired with deCONZ, and when paired with the Hue bridge, I only see unicast commands from the FC00 cluster.

leoluk, 2022-02-07:
Yep, the MAC is the right one. It doesn't send anything to the coordinator.

spudje, 2022-10-10:
Recently my wall switches, configured as push buttons, or deconz started to behave differently)/strangely, Most times when I (short) press I only get x000 events and not the release x002 events anymore. Only when I press a couple of times I get the x002 release.

Has the behaviour been updated and is the x002 now suppoed to be a double press? Or is something wrong on my side?