dresden-elektronik/deconz-rest-plugin #4567: Blitzwolf Smart Plug (_TZ3000_g5xawfcq): What is the update frequency?

iisworks, 2021-03-12:
## Describe the question or issue you are having
I am using Homey with the "HomeyConz App" that uses the deCONZ rest api integration, which works fine. I noticed one thing with the Blitzwolf plugs (Tuya rebrand: _TZ3000_g5xawfcq) though. The update frequency for power measurement seems low: it takes about 1-2 minutes before the power consumption (W) changes after I turn on or off a connected device for example. 

Is this by design due to polling? I was hoping for more frequent updates to power consumption (maybe 5-10s?) 
Note: when reading the "Electrical Measurement" cluster manually, I immediately see the correct actual value.

## Screenshots
Basic cluster:
[image]

Power measurement:
[image]

## Environment
 - Syno DS218+ NAS
 - Running method: Marthoc Docker container 
 - deCONZ version: 2.10.2 (same with 2.9.3 stable and 2.10.1)
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

Mimiix, 2021-03-12:
I have one, and mine does around every 30 secs. AFAIK you can't change it: IT doesn't support the normal way of changing it :(

iisworks, 2021-03-12:
I had them connected directly to Homey Zigbee in the past, and they all updated within seconds. Also tested one with the Tuya GW + app, and same there": they update immediately. 

I could definitely live with 30s max, but I'm still wondering why they all seem to take >1m here (I have 8 of them). Are they polled by deCONZ?

Mimiix, 2021-03-12:
I don't think so. I was trying to get faster updates with @SwoopX But we didn't get that.

However, @manup Has one now and maybe we can ask him to check if we can change it somehow.

iisworks, 2021-03-12:
Quick test using the rest api:
- Turn on plug
- Open GET /sensors/109 api using postman 
- Turned off plug
- Keep refreshing api GET 

**Times before and after update:**
"lastupdated": "2021-03-12T13:05:58.535",
"lastupdated": "2021-03-12T13:08:37.878",

### Start
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "fb7f3be4388e517c78b5681d48787116",
    "lastseen": "2021-03-12T13:06Z",
    "manufacturername": "_TZ3000_g5xawfcq",
    "modelid": "TS0121",
    "name": "Power 109",
    "state": {
        "current": 103,
        "lastupdated": "2021-03-12T13:05:58.535",
        "power": 18,
        "voltage": 227
    },
    "type": "ZHAPower",
    "uniqueid": "84:2e:14:ff:fe:3d:af:2a-01-0b04"
}

### End
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "4c1a9027ae8955478e174942584fa0b3",
    "lastseen": "2021-03-12T13:08Z",
    "manufacturername": "_TZ3000_g5xawfcq",
    "modelid": "TS0121",
    "name": "Power 109",
    "state": {
        "current": 0,
        "lastupdated": "2021-03-12T13:08:37.878",
        "power": 0,
        "voltage": 226
    },
    "type": "ZHAPower",
    "uniqueid": "84:2e:14:ff:fe:3d:af:2a-01-0b04"
}

iisworks, 2021-03-18:
So what I am typically seeing is nowhere near 30s, more like 2m. 
Is this something that can be looked into? Just let me know if more info needed.

iisworks, 2021-03-20:
I am pretty sure it is just not polled often enough (using 2.10.3 Beta now). 

When I look at the device in deCONZ the measured power doesn't change for minutes, but as soon as I press "Read"it updates with the correct current value (tested by switching on/off a connected bulb using the physical on/off button so I know when it's 0W or 11W).

[image]

cllatser, 2021-03-30:
I have 4 and the consumption update doesn't do it instantly either. He does it when he does the general sweep of the network. It would be nice if it was instantaneous. Thanks a lot.

manup, 2021-04-05:
From what I see in the code the reporting configuration is  300 seconds aka 5 minutes as maximum interval.
More important 1 second as minimal interval, if the value differs more than *Reportable Change* from the last report.

Value                | Min Interval | Max Interval | Reportable Change
---------------------| ----------------| -----------------| -----------------------------
RMS Voltage    | 1                 | 300               | 1
RMS Current    | 1                 | 300               | 100
Active Power    | 1                 | 300               | 1

But... the problem is that the device never sends reports for the *Electrical Measurement Cluster*.
I've checked the binding table and the reporting configuration, it's all setup correctly, but no reports.

[image]

Also interesting, the bindings for *On/Off* and *Simple Metering Clusters* seem to be hard coded, they work even when the Binding Table is completely cleared.

Since reporting for *Electrical Measurement Cluster* doesn't work at all, deCONZ can only poll the device. This is done as fast as possible — in my test setup with just one device, approx. every 3 seconds — but with increasing network size polling takes more time since this is done for each device, one after another.

How large is your network?

I don't think there is much we can do here, this needs to be fixed in the smart plug firmware.

cllatser, 2021-04-05:
Hello, I have 88 nodes. Of which 43 are Router. :(

iisworks, 2021-04-05:
Thanks for the update! I have about 60 devices, more than half of which routers (and 10 of those BW plugs). I guess it's not a big network that could explain a delay of 1-2 minutes? I would be happy with a delay of 10s or maybe even 30s.

I had it connected to Homey in the past, and it updated after 5s or so, which is similar to the official Tuya Zigbee Hub. I also have spoken to others using them with HA/Domoticz (Conbee) and they generally seem to have more frequent updates, although the yall seem to use polling as well. 

Maybe the Homey "Tuya Zigbee App repo gives a clue?
https://github.com/JohanBendz/com.tuya.zigbee

cllatser, 2021-04-09:
Hi, would there be any possibility in the future, a function via Rest API, that will simulate the cluster read click? I do not know if I explain myself. Thanks.

[image]

closed 2021-05-08 by github-actions[bot]