dresden-elektronik/deconz-rest-plugin #4572: Device Request IKEA STYRBAR

aje89, 2021-03-12:
## Device

- Product name: STYRBAR
- Manufacturer: IKEA of Sweden
- Model identifier: Remote Control N2
- Device type : Switch

## Screenshots

### Basic
[image]

### Identify
[image]

### PowerConfiguration
[image]

### Poll Control
[image]

### ZLL commissioning
[image]

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

perseoGI, 2021-03-15:
Me too, I have already buy this Styrbar switch and I'm not getting deCONZ to work with it

Mimiix, 2021-03-15:
Can you also add a screen of the node / clusters ?

aje89, 2021-03-15:
> Can you also add a screen of the node / clusters ?

Added to the Overview

ebaauw, 2021-03-15:
Looks like a new version of the five-button remote, but with the on/off button integrated in dimup and dimdown, as with the two-button remote.  We’ll need the Zigbee commands that the styrbar sends on various buttonactions.

Happy to give this a go, if I can get my hands on one.

bemardev, 2021-03-22:
Got this new switch today.
I'm trying to support it but I'm seeing some strange behaviors.
@ebaauw you seem to know well the 5-button remote and have provided some fix for symfonisk, so if you have some time and want to enlighten and help me adding this, you can contact me on discord ;)

ebaauw, 2021-03-22:
Just received mine.  Looks similar to screenshots above, but with MAC address prefix 0x680AE3, oh joy.

After pairing with deCONZ, it sends unicast commands to the coordinator (0x0000).  That's new, but at least it doesn't control all lights by default.  Makes for lousy sniffing, though.  Binding the client _On/Off_ cluster to a group works as expected.

It sends the following commands:

Button | Action | Command | Parameters
--- | --- | --- | ---
_DimUp_ | Press | _On_ |
_DimUp_ | Hold | _Move with On/Off_ | _Up_
_DimUp_ | Release | _Stop with On/Off_ |
_DimDown_ | Press | _Off_ | 
_DimDown_ | Hold | _Move_ | _Down_
_DimDown_ | Release | _Stop with On/Off_
_Left_ | Press | `0x07` | `01010d00`
_Right_ | Press | `0x07` | `00010d00`

So far, so good: exactly what you'd expect combining the two-button on/off remote with the five-button remote.

But here's the catch:

Button | Action | Command | Parameters
--- | --- | --- | ---
_Left_ | Hold | `0x09`<br>_On_<br>`0x07`<br>`0x08` | `0000`<br><br>`02000000`<br>`010d00`
_Left_ | Release | `0x09` | random != `0000`
_Right_ | Hold | `0x09`<br>_On_<br>`0x07`<br>`0x08` | `0000`<br><br>`02000000`<br>`000d00`
_Right_ | Release | `0x09` | random != `0000`

That's right: four commands on holding left or right.  Comes pretty close to a denial-of-service attack, no wonder it sends unicasts to the coordinator by default.  Maybe better not to support `config.group` for this controller, and stick to unicasts, cf. SYMFONISK controller and old TRADFRI dimmer (with dial).

We can filter out the initial `0x09` by hard-coding a check that the payload isn't 0x0000.  The `0x07` wouldn't be matched by the button map, which checks for fist payload byte of 0x00 or 0x01.  But ignoring the _On_ would be more tricky.  We would need to set a device state (like `previousDirection`) on the first `0x09` and clear it on the second `0x09`, and ignore _On_ when the state is set.  Doable, but ugly.  And pray that the messages are received by the coordinator in the order sent.

**TODO**: check whether the switch waits for an ACK before sending the next command in unicast.  If so, that would be another argument no to support `config.group`.

The rest is pretty standard: it blinks on _Identify_ (which is will receive only when awake), it reports battery percentage in full percent, like the other iKEA devices.

ebaauw, 2021-03-22:
> We can filter out the initial 0x09 by hard-coding a check that the payload isn't 0x0000. The 0x07 wouldn't be matched by the button map, which checks for fist payload byte of 0x00 or 0x01. But [...]

Seems to work when STYRBAR is bound to a group, at least in my small test network.

ebaauw, 2021-03-24:
After setting up the (imho) correct bindings in the API:
- (server) _Power Configuration_ to the coordinator;
- (client) _On/Off_, _Level Control_, and _Scenes_ to group.

The device sends the same message as groupcast _and_ as unicast to the coordinator, causing double button events (most of the time).  After removing the _Power Configuration_ binding, it only sends groupcasts.

Looking at the sniffer, the API creates the bindings for the _Power Configuration_ cluster first.  I think we've hit another brilliant firmware bug.  When I create the client cluster bindings first, and then the _Power Configuration_ cluster binding, the device only sends groupcasts.

SwoopX, 2021-03-24:
That exactly sounds like the issue being reported on the SHORTCUT. Still got that one around?

ebaauw, 2021-03-24:
Yeah seeing the same on the SHORTCUT button: both a unicast to 0x0000 and a broadcast to the group.  Paired it quite some versions ago.  I think there's a recent change in the REST API that it now generates two button events.

Here's the deCONZ log (for the SHORTCUT):
```
Mar 24 22:35:33 pi5 deCONZ[16418]: 22:35:33:286 APS-DATA.indication srcAddr: 0x4421, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -46
Mar 24 22:35:33 pi5 deCONZ[16418]: 22:35:33:288 	asdu: 016c01
Mar 24 22:35:33 pi5 deCONZ[16418]: 22:35:33:292 enqueue event event/validgroup for /sensors/10
Mar 24 22:35:33 pi5 deCONZ[16418]: 22:35:33:293 [INFO] - Button 1002 - TRADFRI SHORTCUT Button, broadcast to: 0x000B, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 108
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:294 enqueue event state/buttonevent for /sensors/10
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:295 enqueue event state/lastupdated for /sensors/10
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:304 Websocket 127.0.0.1:51300 send message: {"e":"changed","id":"10","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-03-24T21:35:33.295"},"t":"event","uniqueid":"68:0a:e2:ff:fe:34:91:ec-01-1000"} (ret = -1098005480)
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:306 Websocket 127.0.0.1:51312 send message: {"e":"changed","id":"10","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-03-24T21:35:33.295"},"t":"event","uniqueid":"68:0a:e2:ff:fe:34:91:ec-01-1000"} (ret = -1098005480)
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:309 discard sensor state push for 10: state/lastupdated (already pushed)
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:335 APS-DATA.indication srcAddr: 0x4421, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -46
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:337 	asdu: 016c01
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:341 [INFO] - Button 1002 - TRADFRI SHORTCUT Button, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 108
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:342 enqueue event state/buttonevent for /sensors/10
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:343 enqueue event state/lastupdated for /sensors/10
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:346 Websocket 127.0.0.1:51300 send message: {"e":"changed","id":"10","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-03-24T21:35:33.343"},"t":"event","uniqueid":"68:0a:e2:ff:fe:34:91:ec-01-1000"} (ret = -1098005480)
Mar 24 22:35:38 pi5 deCONZ[16418]: 22:35:33:347 Websocket 127.0.0.1:51312 send message: {"e":"changed","id":"10","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-03-24T21:35:33.343"},"t":"event","uniqueid":"68:0a:e2:ff:fe:34:91:ec-01-1000"} (ret = -1098005480)
```

ebaauw, 2021-03-24:
The x003 events appear only once, because `sensor->previousDirection` is reset on the first `APS-DATA.indication`, so the second no longer matches any entry in the button map.

Note that the unicast and the broadcast carry the same ZCL sequence number.  Already discarding duplicates for the SYMFONISK, could probably do that for the STYRBAR and the SHORTCUT as well.  I think I should check the other IKEA controllers as well: there's no client _Scenes_ cluster on the STYRBAR, and I don't see it on my remotes either.  Maybe they ditched it with the latest firmware.  Also, it seems that only one client cluster binding is enough.

ebaauw, 2021-03-24:
Discarding the second indication with the same ZCL sequence number might not be a good idea.  The unicast typically is received first (the device seems to send it first, but sometimes route discovery might cause some delay), so the groupcast gets ignored, as do the checks on existing bindings and reporting.  Now simply discard the unicast, that seems to work (also for the SHORTCUT).  Still want to do some more testing.  And double-check if the SYMFONISK has the same cause.

Note to self: check whether IKEA special scenes on the light are enabled by touchlinking the light with the remote.  The CWS came pre-paired with the STYRBAR, and still reacts to the left and right buttons.

The STYRBAR will support `buttonevent` values 1002, 1001/1003, 2002, 2001/2003, 3002, 3001/3003, 4002, and 4001/4003, but the 3001 and 4001 are issued only after holding a few second (as they're triggered by the `0x08` command).  As that also sets `sensor->previousDirection()`, the 3003 or 4003 will only be issued after the 3001 or 4001.    If you hold left or right for 2 to 3 seconds, no buttonevent is issued (too long for single press, and to short for the `0x08` command).

Zehir, 2021-03-25:
@ebaauw did you check #4265 ?
It's look like it's for this kind of issue like #4227 too.

ebaauw, 2021-03-25:
No, hadn't seen it, thanks!

The PR is a bad idea, as this will break the auto-repair of bindings and attribute reporting, see above.

In case of the IKEA STYRBAR and SHORTCUT, it's caused by the device sending both unicast and group cast messages (when the binding to the coordinator for the reporting of the battery has been made).  If I understand correctly the Tuya has a different cause: it resends the unicast (?) because it doesn't receive a _Default Response_.  Not sure what's the cause of the Aqara; need to see deCONZ logs with `APS-DATA.indication` messages.

ebaauw, 2021-03-25:
Double checked the SYMFONISK controller: exactly the same behaviour.  As soon as _Power Configuration_ is bound to the coordinator (and _On/Off_ to the group) it sends both a unicast and a group cast.  Also, the STYRBAR, SHORTCUT, and SYMFONISK only require a binding for a single client cluster.  The SYMFONISK also sends unicasts to the coordinator after reset and pairing, so the code to create the unicast binding to the coordinator isn't needed.

closed 2021-03-27 by manup

aje89, 2021-04-30:
It was working fine. But since Beta 11.0 left and right does not work anymore. (No Button events)

I have the version compiled by my own after your first changes, which is still working fine.

Mimiix, 2021-04-30:
@aje89 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4821  There's a bug report on it afaik.

aje89, 2021-04-30:
> @aje89 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4821  There's a bug report on it afaik.

Thx excuse double post

theNailz, 2021-06-20:
Any update on the Styrbar E2002 FCC ID FHO-E2002? It's not in the https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#supported-wireless-switches list, but there is this (closed) issue. In which release will this de supported? (And possibly E2001 as well?)

pattyland, 2021-07-21:
Checked today in 2.11.05 / 9.5.2021: STYRBAR remote control is not in the selection list at "Add switch" -> IKEA, when I select "Other" the STYRBAR does not show up in the list of connected switches either.

Mimiix, 2021-07-21:
@pattyland This is an Phoscon thingy :) You need to ask support there.

VentiloFred, 2021-11-21:
What news with this remote Styrbar E2002? 
I connected two of them on my Home Assistant with deCONZ v6.10.0, everything work but after trying long push left button the UP and Down doesn't work.

Mr-Hammar, 2021-12-01:
> What news with this remote Styrbar E2002? I connected two of them on my Home Assistant with deCONZ v6.10.0, everything work but after trying long push left button the UP and Down doesn't work.

Same issue for me, the only way of getting the On and Down working is to long press the left or right button again.

LetmeCheckthatforme, 2021-12-23:
I'm getting the same kind of issue. All is working fine with the Styrbar, but after a while (might did a long press) the up and down buttons arent triggering anything anymore.

aje89, 2021-12-24:
This one is closed.
There is an open bug #5516 for this issue.