dresden-elektronik/deconz-rest-plugin #4573: Unable to pair battery powered Zemismart Chain Roller Shades Driver M515EGBZTN  in v2.10.1

PhilipsGorniczy, 2021-03-12:
Hello

I am unable to properly pair Zemismart Chain Roller Shades Driver M515EGBZTN on v2.10.1 and v2.10.2
In the release notes for v2.10.1 it was marked as supported, but pairing does not complete. A new device appears in Deconz GUI, but nothing appears in Phoscon. 
I have searched and read previous issues - most people were able to pair it and see it from HomeAssistant. Mine does not show up anywhere. 
Is it supposed to work, or we just getting multiple devices under the same name, model and manufacturer and this one is not supported?

## Device

- Product name: Zemismart M515EGBZTN
- Manufacturer: _TZE200_xuzcvlku
- Model identifier: TS0601
- Device type :  Cover

## Screenshots

### Basic
[image]

###Node
[image]

Smanar, 2021-03-12:
I don't remember if the device is visible in phoscon.
Have you tried to include it 2 time, and without deleting anything between both ?

PhilipsGorniczy, 2021-03-12:
Yes, both times as a light.

Smanar, 2021-03-13:
Can you share the log during inclusion ?
You can have them in deconz (deconz/help/log with "info" and "info_l2".

To make it easy
- start logging
- in phoscon take "add new light"
- in deconz read basic attribute (cluster 0x0000, button "read", it need to fill the table)

It will trigger an inclusion.

PhilipsGorniczy, 2021-03-13:
Weird thing is now happening. I found old Conbee I and I have spare PC, so I decided to do debug on separate zigbee network.

First try was unsuccessful. After that when  I start up deconz the zemistmart driver appears even without pairing:
[image]

When I try to delete it it comes back. I tried to pair it again, but for some reason I get:
`17:16:39:595 Node 0x5C0272FFFEECC56D is known by 1 neighbors, last seen 3 s`

Full log attached. 
[zemismartmotor.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6135210/zemismartmotor.txt)

PhilipsGorniczy, 2021-03-13:
I managed to clean up the entry - I did full gateway reset. 
Then started dumping logs and tried to add the driver to deconz. 
Log from that attached:
[zemismartmotor2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6135230/zemismartmotor2.txt)

Smanar, 2021-03-13:
On your capture the device is not yet included, miss a circle on the node.

Remember making 2 inclusions

> 17:37:21:045 Tuya debug 7 : Missing manufacture name for 0x5c0272fffeecc56d
> 17:37:21:045 Tuya debug 7 : Missing manufacture name, till missing in DB.

You can have thoses logs only on first one, because on the second one the value is on the DB.

PhilipsGorniczy, 2021-03-13:
Ok, I retried.

1. I reset the gateway and restarted deconz container. 
2. I started "Add new light" (I set it for 10 MINS)
3. State after first pairing:
[image]
Cluster info got populated by then. 
4. Second pairing (light search is still active) -> after 18:10:58 from logs
5. Nothing changed.
6. Third attempt. Still nothing. In the logs:
```18:12:31:691 APS-DATA.indication srcAddr: 0x24a9, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 183, rssi: -68
18:12:31:691 	asdu: 89a9246dc5ecfeff72025c80
18:12:31:692 unknown node 0x5C0272FFFEECC56D (0x24A9), lqi: 183
18:12:31:692 0xBD2E nwk changed to 0x24A9
18:12:31:693 APS-DATA.indication from child 0x24A9
18:12:31:694 ZDP device announce: 0x5C0272FFFEECC56D, 0x24A9, 0x80
18:12:31:695 FP indication 0x0000 / 0x0013 (0x5C0272FFFEECC56D / 0x24A9)
18:12:31:695                       ...     (0x5C0272FFFEECC56D / 0xAE7A)
18:12:31:695 device announce 0x5C0272FFFEECC56D (0x24A9) mac capabilities 0x80
18:12:31:696 FP indication 0x0000 / 0x0013 (0x5C0272FFFEECC56D / 0x24A9)
18:12:31:696                       ...     (0x5C0272FFFEECC56D / 0xAE7A)
18:12:31:696 device announce 0x5C0272FFFEECC56D (0x24A9) mac capabilities 0x80
18:12:31:755 MAC Poll 0x02 0x24A9, life time: 4294967295 sec, dev timeout: 4294967295 sec
18:12:31:757 CTRL skip polling while permit join is set
```
And the same. Device did not pair fully.
Log: 
[zemismartmotor3.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6135289/zemismartmotor3.txt)

Is there some mistake in my understanding how to pair it?

Smanar, 2021-03-13:
If I add some debug line in the code, are you able to compile it to test something ? (just need a linux machine)

PhilipsGorniczy, 2021-03-13:
Yes, no problem. I have MiniPC with Ubuntu 20.04. I can run deconz without docker, straight on the OS.

Smanar, 2021-03-13:
Nice, I have asked some help (because you are the third one with similar issue), if I haven't a way, I will make you a special version for tommorow.

andys73, 2021-03-13:
Hello,
I have the exact issue with another device,  a switch from Moes / ZTS-EU3-W (3 Gang)...

It does not pair at all but I can see it in VNC. I have tried to remove it and add it several times..

I can try the special version too :)

[image]

[image]

[image]

Smanar, 2021-03-14:
@andys73 your device is a new one, pls make a new issue, will take some time, because your device use a new mac adress range and have the same cluster than TRV https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4401 so need to find a way to reconize it.

@PhilipsGorniczy I think I have find the problem
The device is battery one, but there is no sensor created for it.
So when the light part check for the device, it can use the sensor entry.

So I see 2 solutions.
- create a battery sensor, and this will be usefull.
- include the device using deconz

- on phoscon : add new light
- on deconz read basic attribute 2 time, cluster 0x0000, button read, it need to fill the table.

PhilipsGorniczy, 2021-03-14:
Thanks @Smanar !

Is there an additional step to create battery sensor? 

I did those steps:
```
To include using deconz

    on phoscon : add new light
    on deconz read basic attribute 2 time, cluster 0x0000, button read, it need to fill the table.
```
nothing changes. Still the same errors.

PhilipsGorniczy, 2021-03-14:
More details what exactly I do:

1. First pairing (reseting the device to join zigbee) - table is empty:
[image]
After clicking read first time:
[image]
After clicking read second time:
[image]
Now the second pairing (reseting the device to join zigbee). And now it gets stuck again:
```
deconz    | 14:45:59:893 skip device discovery while end devices is added
deconz    | 14:46:00:133 Node 0x5C0272FFFEECC56D is known by 1 neighbors, last seen 8 s
deconz    | 14:46:00:373 skip device discovery while end devices is added
deconz    | 14:46:00:613 Node 0x5C0272FFFEECC56D is known by 1 neighbors, last seen 8 s
```
Logs:
[zemismart_1403.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6136879/zemismart_1403.txt)

PhilipsGorniczy, 2021-03-14:
By the way, this device should be already added via this PR:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4274/files

Does it mean, that Zemismart is shipping different devices under name `_TZE200_xuzcvlku`?

Smanar, 2021-03-14:
No reset.
just set phoscon in permit join (add new light) and read basic attribute on deconz (2 time)

PhilipsGorniczy, 2021-03-14:
Tried that. Did not help. 
Logs:
[zemismart_1403_3.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6136915/zemismart_1403_3.txt)

Phoscon still empty. API also shows only the coordinator:
```
{"1":{"etag":"1349c37a58a930f2503cff2214d6d2df","hascolor":false,"lastannounced":null,"lastseen":"2021-03-14T15:03Z","manufacturername":"dresden elektronik","modelid":"ConBee","name":"Configuration tool 1","state":{"reachable":true},"swversion":null,"type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:04:09:82-01"}}
```

Sensor was not created. In sensor there is only the default day/night sensor:
```
{"1":{"config":{"configured":true,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":"eb3da059e4411b3fe693ae07095a3f20","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":false,"daylight":true,"lastupdated":"2021-03-14T15:00:47.278","status":170,"sunrise":"2021-03-14T04:57:25","sunset":"2021-03-14T16:43:57"},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:04:09:82-01"}}
```

Smanar, 2021-03-14:
Still nothing in your log, I realy don't understand, no new inclusion triggered.

I will make the battery device creation, give me 1 h

Smanar, 2021-03-14:
Can you try this modification ?

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you

```
sudo apt install deconz-dev
git clone --branch covering_7 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

With this code you will have a new sensor for battery, and the light part will be able to use it for light device creation.

PhilipsGorniczy, 2021-03-14:
Ok, so I installed xfce on the LAB box. Then I complied the code, accordingly to the instruction.
md5 after the compilation. 
`91be7fd9199f694a78f5fbde39d82b81  /usr/share/deCONZ/plugins/libde_rest_plugin.so`

I was able to start the gui via vnc and start deconz app. Phoscon is also up. 
I reset the gateway and then started add light process - for the first time it did not work. 
I changed the light search duration to 10mins and restarted the process.

I was able to get the driver added to deconz. A the beginning cluster 0x0000 was empty. I clicked read two times. 
[image]

And nothing much more happened. Still no sensors available.
Log:
[deconzlog.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6137092/deconzlog.txt)

PhilipsGorniczy, 2021-03-14:
Some more verbose debug from second attempt:
[secondattempt.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6137124/secondattempt.txt)

PhilipsGorniczy, 2021-03-14:
Actually new sensor is visible via api:
```
{"1":{"config":{"configured":false,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":null,"manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":null,"daylight":null,"lastupdated":"none","status":null},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:04:09:82-01"},"2":{"config":{"on":true,"reachable":true},"ep":1,"etag":"b841b1c9cf6eed1fba80a6cecf15fac9","lastseen":"2021-03-14T16:52Z","manufacturername":"_TZE200_xuzcvlku","modelid":"TS0601","name":"Battery 2","state":{"battery":null,"lastupdated":"none"},"type":"ZHABattery","uniqueid":"5c:02:72:ff:fe:ec:c5:6d-01-ef00"}}
```
Yay!
I will check whether I can see the device in Homeassistant.

Smanar, 2021-03-14:
When you include a tuya sensor, the code search for the manufacture name.
- in the deconz app > not working for you
- in the sensor database

As you have the battery sensor, I think it will help the API to reconize it.

PhilipsGorniczy, 2021-03-14:
Unfortunately, new sensor is visible only from deconz api via call:
`curl -X GET -i '10.0.0.125/api/KEY/sensors'
Lights endpoint does not show any new lights:
```
{"1":{"etag":"232b0008fee1d8da6aa4c6666d8ecea4","hascolor":false,"lastannounced":null,"lastseen":"2021-03-14T17:03Z","manufacturername":"dresden elektronik","modelid":"ConBee","name":"Configuration tool 1","state":{"reachable":true},"swversion":null,"type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:04:09:82-01"}}
```

Smanar, 2021-03-14:
If I m right on log starting by "Tuya debug 7" the code need to find a manufacture name.
And don't delete the battery sensor.

Smanar, 2021-03-15:
Have you try a last inclusion "add new light" ?

On previous log you had
```
17:37:21:045 Tuya debug 7 : Missing manufacture name for 0x5c0272fffeecc56d
17:37:21:045 sql exec SELECT manufacturername FROM nodes WHERE mac LIKE '5c:02:72:ff:fe:ec:c5:6d%' COLLATE NOCASE
17:37:21:045 Tuya debug 7 : Missing manufacture name, till missing in DB.
```

But now you have the battery sensor, so the API will found a manufacture name and be able to use it.

Smanar, 2021-03-19:
Have find a possible issue, can you try this branch pls

```
sudo apt install deconz-dev
git clone --branch tuya_14 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

GreyEarl, 2021-03-19:
@Smanar Today I received my Zemismart and it's  M515EGBZTN one. Can I try the above branch without loosing my current setup , or is it specifically meant for the other user?

Smanar, 2021-03-19:
The code can't make you loosing setup, it can just working or not working.
The model mean nothing for tuya, realy need the manufacture name, you can try with the last version you have, if it don't work can try the branch tuya_14, this one just improve the reconization, nothing changed on the rest of code.

GreyEarl, 2021-03-19:
@Smanar Thank you for your effort. I have compiled your fork. I tried to pair the device, but unfortunately no results in my case. No new devices seen in /lights.

Smanar, 2021-03-20:
The last patch have worked on another issue, so I have merged the 2 patchs
Now the code create the battery sensor (it have already worked)
And have better reconisation for manufacture name

```
sudo apt install deconz-dev
git clone --branch tuya_14 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Or if you have already started

```
cd deconz-rest-plugin
git checkout tuya_14 
git pull
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

GreyEarl, 2021-03-20:
@Smanar  The manufacturer is in my case: `_TZE200_xuzcvlku`
I will try your new merge.

GreyEarl, 2021-03-20:
Nothing on my end unfortunately. The most odd thing is that I still do not see the baterry sensor in /sensors, while PhilipsGorniczy does seem to have in a previous fork.

Smanar, 2021-03-20:
@GreyEarl can you share the log during pairing ? with "info" and "info_l2"

GreyEarl, 2021-03-21:
@Smanar I have to correct myself. There is now a battery sensor added. I did something wrong, usually I use deCONZ on an headless server. With some magic I am using now VNC. The logs contain a lot of noise as it is my production setup. I have included them in detail_log.txt. Here's an example of the device:

```
08:57:28:224 unknown node 0x5C0272FFFEECB8D7 (0xF5A3), lqi: 111
08:57:28:225 0x9277 nwk changed to 0xF5A3
08:57:28:226 APS-DATA.indication from child 0xF5A3
08:57:28:227 DeviceAnnce of SensorNode: 0x5C0272FFFEECB8D7 [1]
08:57:28:228 enqueue event config/reachable for /sensors/52
08:57:28:229 don't create binding for attribute reporting of sensor Battery 52
08:57:28:229 skip check bindings for client clusters (no group)
08:57:28:230 	nwk address changed 0x9277 -> 0xF5A3 [2]
08:57:28:231 enqueue event event/added for /sensors/52
08:57:28:232 FP indication 0x0000 / 0x0013 (0x5C0272FFFEECB8D7 / 0xF5A3)
08:57:28:233                       ...     (0x5C0272FFFEECB8D7 / 0x9277)
08:57:28:233 device announce 0x5C0272FFFEECB8D7 (0xF5A3) mac capabilities 0x80
08:57:28:234 FP indication 0x0000 / 0x0013 (0x5C0272FFFEECB8D7 / 0xF5A3)
08:57:28:235                       ...     (0x5C0272FFFEECB8D7 / 0x9277)
08:57:28:236 device announce 0x5C0272FFFEECB8D7 (0xF5A3) mac capabilities 0x80
08:57:28:313 Websocket 10.0.0.21:41990 send message: {"config":{"on":true,"reachable":true},"e":"changed","id":"52","r":"sensors","t":"event","uniqueid":"5c:02:72:ff:fe:ec:b8:d7-01-ef00"} (ret = 2130406040)
08:57:28:314 Websocket 10.0.0.12:59953 send message: {"config":{"on":true,"reachable":true},"e":"changed","id":"52","r":"sensors","t":"event","uniqueid":"5c:02:72:ff:fe:ec:b8:d7-01-ef00"} (ret = 2130406040)
08:57:28:316 Websocket 10.0.0.12:60058 send message: {"config":{"on":true,"reachable":true},"e":"changed","id":"52","r":"sensors","t":"event","uniqueid":"5c:02:72:ff:fe:ec:b8:d7-01-ef00"} (ret = 2130406040)
08:57:28:319 MAC poll fastEnddeviceProbe() 0x5C0272FFFEECB8D7
08:57:28:320 [4.1] Get date code
08:57:28:323 skip check bindings for client clusters (no group)
08:57:28:326 CTRL skip polling while permit join is set
08:57:28:411 don't create binding for attribute reporting of sensor Battery 52
08:57:28:412 skip check bindings for client clusters (no group)
08:57:28:414 Websocket 10.0.0.21:41990 send message: {"e":"added","id":"52","r":"sensors","sensor":{"config":{"on":true,"reachable":true},"ep":1,"etag":"809a8192ada68253714c02b583e44392","id":"52","lastseen":"2021-03-21T07:57Z","manufacturername":"_TZE200_xuzcvlku","modelid":"TS0601","name":"Battery 52","state":{"battery":null,"lastupdated":"none"},"type":"ZHABattery","uniqueid":"5c:02:72:ff:fe:ec:b8:d7-01-ef00"},"t":"event","uniqueid":"5c:02:72:ff:fe:ec:b8:d7-01-ef00"} (ret = 2130406040)
08:57:28:416 Websocket 10.0.0.12:59953 send message: {"e":"added","id":"52","r":"sensors","sensor":{"config":{"on":true,"reachable":true},"ep":1,"etag":"809a8192ada68253714c02b583e44392","id":"52","lastseen":"2021-03-21T07:57Z","manufacturername":"_TZE200_xuzcvlku","modelid":"TS0601","name":"Battery 52","state":{"battery":null,"lastupdated":"none"},"type":"ZHABattery","uniqueid":"5c:02:72:ff:fe:ec:b8:d7-01-ef00"},"t":"event","uniqueid":"5c:02:72:ff:fe:ec:b8:d7-01-ef00"} (ret = 2130406040)
08:57:28:418 Websocket 10.0.0.12:60058 send message: {"e":"added","id":"52","r":"sensors","sensor":{"config":{"on":true,"reachable":true},"ep":1,"etag":"809a8192ada68253714c02b583e44392","id":"52","lastseen":"2021-03-21T07:57Z","manufacturername":"_TZE200_xuzcvlku","modelid":"TS0601","name":"Battery 52","state":{"battery":null,"lastupdated":"none"},"type":"ZHABattery","uniqueid":"5c:02:72:ff:fe:ec:b8:d7-01-ef00"},"t":"event","uniqueid":"5c:02:72:ff:fe:ec:b8:d7-01-ef00"} (ret = 2130406040)

```
[detail_log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6177035/detail_log.txt)

Smanar, 2021-03-21:
And now it have the battery sensor, he can use it to reconize the device to create the "light" entry.
So don't delete it, and re try to include the device ("add new light"), on log (with "info" and "info _l2") you will have line with "tuya debug"

GreyEarl, 2021-03-21:
Great success!! Thank you very much Smanar! :))

After I deleted the node, and reinclude the device again:

```
10:15:35:458 Tuya debug 7 : Missing manufacture name for 0x5c0272fffeecb8d7
10:15:35:461 sql exec SELECT manufacturername FROM nodes WHERE mac LIKE '5c:02:72:ff:fe:ec:b8:d7%' COLLATE NOCASE
10:15:35:465 sql exec SELECT * FROM nodes WHERE mac='5c:02:72:ff:fe:ec:b8:d7-01' COLLATE NOCASE AND state != 'deleted'
10:15:35:468 sql exec SELECT * FROM nodes WHERE mac='0x5c0272fffeecb8d7' COLLATE NOCASE AND state != 'deleted'
10:15:35:471 sql exec SELECT * FROM nodes
10:15:35:474 LightNode 10: Smart plug 10 added
10:15:35:476 enqueue event event/added for /lights/10
10:15:35:478 enqueue event attr/lastseen for /lights/10
10:15:35:480 DB pushZdpDescriptorDb()
10:15:35:483 FP indication 0x0000 / 0x8004 (0x5C0272FFFEECB8D7 / 0xEA43)
10:15:35:486                       ...     (0x5C0272FFFEECB8D7 / 0xEA43)
10:15:35:490 ZDP indication search sensors 0x5C0272FFFEECB8D7 (0xEA43) cluster 0x8004
10:15:35:492 ZDP indication search sensors 0x5C0272FFFEECB8D7 (0xEA43) clear timeout on cluster 0x8004
10:15:35:494 read attributes of 0x5C0272FFFEECB8D7 cluster: 0x0000: [ 
10:15:35:496 0x0005 
10:15:35:498 ]
10:15:35:500 add task 25805 type 19 to 0x5C0272FFFEECB8D7 cluster 0x0000 req.id 233
10:15:35:799 MAC poll fastEnddeviceProbe() 0x5C0272FFFEECB8D7
10:15:35:801 [4.1] Get sw build id
10:15:35:804 [4.2] get basic cluster attr 0x4000 for 0x5c0272fffeecb8d7
10:15:35:807 skip check bindings for client clusters (no group)
10:15:35:809 don't create binding for attribute reporting of sensor Battery 52
```

I even see the device in Phoscon:
[image]

And in Home Assistant, this is awesome!

[image]

GreyEarl, 2021-03-21:
@Smanar 

What I discovered so far, I cannot send the up command. The up button in HA is also greyed out it seems. When I press the hardware UP button on the device, then things are going bad and device stops responding.

So far I'm already happy this device is recognized :)

Smanar, 2021-03-21:
Can come from the motor direction, as you are the only one with the issue for the moment I prefer don't change the side using the code.

On the file you are using (branch tuya_14) , on file rest_light.cpp you will find
```
        QByteArray direction = QByteArray("\x01\x00", 2);
        if (map["reverse"].toBool())
        {
            direction = QByteArray("\x01\x01", 2);
        }
```

change the code to
```
        QByteArray direction = QByteArray("\x00", 1);
        if (map["reverse"].toBool())
        {
            direction = QByteArray("\x01", 1);
        }
```

Then you will be able to use the "reverse" command.

Eg with curl

`curl -H 'Content-Type: application/json' -X PUT -d '{"reverse": true}' http://IP:PORT/api/KEY/lights/ID`

GreyEarl, 2021-03-21:
Thanks, @Smanar! Appreciate it!

@PhilipsGorniczy Let us know how it works on your end.

GreyEarl, 2021-03-22:
> Can come from the motor direction, as you are the only one with the issue for the moment I prefer don't change the side using the code.
> 
> Sorry the fix is on 3 different PR, so to make change on the version you have ATM (branch tuya_14) , on file rest_light.cpp you will find
> 
> ```
>         QByteArray direction = QByteArray("\x01\x00", 2);
>         if (map["reverse"].toBool())
>         {
>             direction = QByteArray("\x01\x01", 2);
>         }
> ```
> 
> change the code to
> 
> ```
>         QByteArray direction = QByteArray("\x00", 1);
>         if (map["reverse"].toBool())
>         {
>             direction = QByteArray("\x01", 1);
>         }
> ```
> 
> Then you will be able to use the "reverse" command.
> 
> Eg with curl
> 
> `curl -H 'Content-Type: application/json' -X PUT -d '{"reverse": true}' http://IP:PORT/api/KEY/lights/ID`

Excuted you suggestion, however the UP button in HA is still greyed out. It seems others had similar issues with other blinds and they had to create a binding, like this: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121#issuecomment-524617659

Could it be the device needs something like this as well?

Also, I've tested the device further. And sometimes, after a period of inactivity, the devices does not respond on input. This is what I then see in the logs:

```
22:25:18:272 delay sending request 114 ep: 0x01 cluster 0xEF00 to 0x5c0272fffeecb8d7 onAir 2
22:25:18:274 delay sending request 117 dt 0 ms to 0x5C0272FFFEECB8D7, ep: 0x01 cluster: 0xEF00 onAir: 2
22:25:18:276 delay sending request 117 ep: 0x01 cluster 0xEF00 to 0x5c0272fffeecb8d7 onAir 2
22:25:18:551 delay sending request 99 dt 0 ms to 0x5C0272FFFEECB8D7, ep: 0x01 cluster: 0xEF00 onAir: 2
22:25:18:553 delay sending request 99 ep: 0x01 cluster 0xEF00 to 0x5c0272fffeecb8d7 onAir 2
22:25:18:555 delay sending request 103 dt 0 ms to 0x5C0272FFFEECB8D7, ep: 0x01 cluster: 0xEF00 onAir: 2
22:25:18:560 delay sending request 103 ep: 0x01 cluster 0xEF00 to 0x5c0272fffeecb8d7 onAir 2
22:25:18:562 delay sending request 107 dt 0 ms to 0x5C0272FFFEECB8D7, ep: 0x01 cluster: 0xEF00 onAir: 2
22:25:18:565 delay sending request 107 ep: 0x01 cluster 0xEF00 to 0x5c0272fffeecb8d7 onAir 2
22:25:18:567 delay sending request 111 dt 0 ms to 0x5C0272FFFEECB8D7, ep: 0x01 cluster: 0xEF00 onAir: 2
22:25:18:570 delay sending request 111 ep: 0x01 cluster 0xEF00 to 0x5c0272fffeecb8d7 onAir 2
22:25:18:573 delay sending request 114 dt 0 ms to 0x5C0272FFFEECB8D7, ep: 0x01 cluster: 0xEF00 onAir: 2
22:25:18:576 delay sending request 114 ep: 0x01 cluster 0xEF00 to 0x5c0272fffeecb8d7 onAir 2
22:25:18:578 delay sending request 117 dt 0 ms to 0x5C0272FFFEECB8D7, ep: 0x01 cluster: 0xEF00 onAir: 2
22:25:18:580 delay sending request 117 ep: 0x01 cluster 0xEF00 to 0x5c0272fffeecb8d7 onAir 2
```

GreyEarl, 2021-03-22:
@Smanar please scratch the greyed-out button in HA. After some investigation it seems an HA Lovelace issue.

jscheurwater, 2021-03-22:
@GreyEarl Can I contact you to ask you some questions on how you integrated the Zemismart into Phoscon?

GreyEarl, 2021-03-22:
> @GreyEarl Can I contact you to ask you some questions on how you integrated the Zemismart into Phoscon?

Sure, though I did not do anything special for Phoscon.

Smanar, 2021-03-22:
The problem in HA, the grayed button can come from deconz too, on some device the side is reversed, depend of mounting.
HA disable the button according to the return state, so if the device return "open", HA disable the "up" button, for the user use only "close", but if it s reversed you can imagine what happen.

Easy way to be sure, jusr look the device JSON you need to have 

```
shutter closed
on= true
bri = 255
open = false
lift = 100%
```

Or all reversed, no mix.

jscheurwater, 2021-03-23:
Can somebody tell me how to compile the code (to add the battery sensor) inside the Home Assistant environment?
I have a Ubuntu environment ready, but I have my conbee 2 connected to my Raspberry Pi with Home Assistant.

This means I can't test it on my Ubuntu.

Mimiix, 2021-03-23:
That is something to ask in the HA support channels/info channels. We don't maintain the addon/OS.

GreyEarl, 2021-03-23:
> The problem in HA, the grayed button can come from deconz too, on some device the side is reversed, depend of mounting.
> HA disable the button according to the return state, so if the device return [...]

You are right by the way. In HA everything is reversed. For example, the open command is 'close' and vica versa. I work around that for the moment.

Smanar, 2021-03-23:
And the command "reverse" using the api haven't worked ?
There is too on option in HA ( or in all other third app) to reverse covering direction, but it reverse ALL filed in same time.

GreyEarl, 2021-03-25:
> And the command "reverse" using the api haven't worked ?
> There is too on option in HA ( or in all other third app) to reverse covering direction, but it reverse ALL filed in same time. It s like reverse the wiring.

The reverse command did not work initially, but now it does, yes, thank you! Apparantly I missed something in the process.

Regarding the unresponsiveness of the device after a while. shall I create a new issue?

jscheurwater, 2021-03-25:
When will the updated code be merged into the DeCONZ stable version?

Smanar, 2021-03-26:
> The reverse command did not work initially, but now it does, yes, thank you! Apparantly I missed something in the process.

The code have moved a lot since the start

> Regarding the unresponsiveness of the device after a while. shall I create a new and seperate issue for that?

I think it s better, not to have answer (I relay don't have idea on my side) but to find other users with the same issue.

@jscheurwater the last stable have just some days old, so at minimum in 1 month, but will be soon in a beta one

jscheurwater, 2021-03-26:
Thanks for the info @Smanar I already have plans to edit my setup so I can also use your custom code in my environment.

Smanar, 2021-03-26:
I realy don't know date
The previous PR is alredy in milestone > https://github.com/dresden-elektronik/deconz-rest-plugin/milestone/7
But you need too the one on ds82

closed 2021-03-27 by manup

Meatballs1, 2021-04-20:
Is it likely these can/would report their opening %? I'm currently having no luck setting the Limits manually on the device, and wondering how @GreyEarl is getting on. 

Currently just use an automation with a delay to open/close.

GreyEarl, 2021-04-20:
> Is it likely these can/would report their opening %? I'm currently having no luck setting the Limits manually on the device, and wondering how @GreyEarl is getting on.
> 
> Currently just use an automation with a delay to open/close.

I can set a position from 0 to 100% in Home Assistant. Though I do have set the limits first. Try this. This is better written than the manual that came with the device:

[image]

Meatballs1, 2021-04-20:
Ah yes, those are much clearer thanks!

I think the actual problem was I was misreading the 'Delete Limit' instructions because the second step is described as **'_Keep_** pressing the motor [up key]....' so I was pressing pressing the `set key` (that is an upwards press) rather than the `up key`.

Meatballs1, 2021-04-20:
Just need to work out if HA will set a % position with the slow motor speed now...

mhin76, 2021-04-27:
Hi all. I have a similar issue with my Zigbee. I had a nice network with several Xiaomi temp and door sensors (and others). after suddenly loosing one door sensor and not able to add it again (it worked for several days but trying to add it I only get in deconz v2.10.4 the short 0x ID but nothing in Phoscon) I resetted the gateway. now I cannot add anymore any of the Xiaomi temp nor door sensor. all other lights, motion sensor, plugs work. I tried differet Conbee firmwares. no luck. will try the approach written above and hope it will work

Smanar, 2021-04-27:
> I have a similar issue with my Zigbee

"Similar", this issue is about Chain roller Shades ...

If you can't add new device, check if the firwmare number is visible on phoscon, try with an USB extension usb cordon and check the deconz logs, but pls create a new issue for that.

[25 more replies]