dresden-elektronik/deconz-rest-plugin #4575: D-Link Motion Sensor

jazza1234567, 2021-03-13:
## Device
Product name: Smart Motion Sensor
Manufacturer: D-Link
Model identifier: DCH-b122
Device type : Motion Sensor

## Screenshots

### Node Info
[image]

### Basic Info
[image]

### Power Configuration
[image]

### Identify
[image]

### Poll Control
[image]

### IAS Zone
[image]
[image]

### Diagnostics
[image]

### unknown
[image]

### OTAU 
[image]

jazza1234567, 2021-03-17:
hey @SwoopX  just wanted to ask if I did something wrong on this request,

I noticed you did a pull request for my post that was immediately after this one, so just not sure if I messed this one up!

The model numbers are really similar too in case it looked like a double post!

Cheers

SwoopX, 2021-03-17:
Nothing wrong at all. I just had a PR for the other device ready since several months (as is has been previously requested), but the outstanding questions were never answered or information provided respectively.

Before adding this device, I need the pending one to get merged.

closed 2021-04-23 by manup

NW27, 2021-11-13:
Hi,
I have some of these sensors. Any idea when they will be available in the app?
Thanks Neil.