dresden-elektronik/deconz-rest-plugin #4576: D-Link Smart Door/Window Sensor

jazza1234567, 2021-03-13:
## Device
Product name: Smart Door/Window Sensor
Manufacturer: D-Link
Model identifier: DCH-B112
Device type : Sensor

## Screenshots

### Node Info
[image]
[image]

### Basic Info
[image]

### Power Configuration
[image]

### Identify
[image]

### Poll Control
[image]

### IAS Zone
[image]
[image]

### Diagnostics
[image]

### OTAU 
[image]

jazza1234567, 2021-04-04:
This issue is still pending. If i can contribute anymore data to help I am
more than happy to do so!

closed 2021-04-18 by manup

NW27, 2021-11-13:
Hi,
I have some of these sensors. Any idea when they will be available in the app?
Thanks Neil.