dresden-elektronik/deconz-rest-plugin #4578: Moving from RaspBee to RaspBee II

det-box, 2021-03-13:
## Describe the question or issue you are having
Hello, what is the best way for me to make a move from
Raspberry3 with RaspBee to a
Raspberry4 with RaspBee II.
I made a backup on the Raspberry3 in the Phoscon and read it on the Raspberry4 in the Phoscon.
All Node are so far dareden but not with the GateWay (RaspBee II). mThere are no lines between GateWay and Node.
What am I possibly doing wrong?

I am grateful for any help.
Thanks

## Screenshots
RaspBee I
[image]

RaspBee II
[image]

## Environment

[image]

 - Host system: (Raspberry Pi )
 - Running method: (Raspbian /  deCONZ-gui)
 - Firmware version: (26690700)
 - deCONZ version: (2.09.03)
 - Device: (RaspBee I -> RaspBee II)

 -

## deCONZ Logs

## Additional context

Smanar, 2021-03-13:
Your procedure seem good.
Can be a problem direclty with the raspbee2, can you take a look on deconz log, on deconz/help/log with all "info" and "error" ?

SwoopX, 2021-03-13:
How about a backup/restore via Phoscon?

Wild guess: you did so and deconz crashed while importing the backup?

det-box, 2021-03-13:
here is a log, all INFO and Error checked

`18:05:26:319 Poll APS confirm 235 status: 0xE1
18:05:26:319 	 drop item attr/modelid
18:05:26:319 	 drop item attr/swversion
18:05:26:320 0x84182600000D716E error APSDE-DATA.confirm: 0xE1 on task
18:05:26:320 Erase task req-id: 235, type: 19 zcl seqno: 119 send time 1, profileId: 0x0104, clusterId: 0x0006
18:05:26:724 enqueue event config/localtime for /config/
18:05:26:725 Wait 1s till query finished
18:05:26:726 rule event /config/localtime: 18:05:25.723 -> 18:05:26.724 (1001ms)
18:05:26:900 poll node 84:18:26:00:00:0d:71:6e-03
18:05:26:901 Poll light node Stube Dose3
18:05:26:952 read attributes of 0x84182600000D716E cluster: 0x0006: [ 
18:05:26:953 0x0000 
18:05:26:953 ]
18:05:26:953 add task 201 type 19 to 0x84182600000D716E cluster 0x0006 req.id 243
18:05:26:954 Poll APS request 243 to 0x84182600000D716E cluster: 0x0006
18:05:27:191 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
18:05:27:224 Daylight now: goldenHour2, status: 180, daylight: 0, dark: 0
18:05:27:723 enqueue event config/localtime for /config/
18:05:27:724 Idle timer triggered
18:05:27:724 Force read attributes for ZHAOpenClose SensorNode Wohnzimmer Fenster
18:05:27:726 rule event /config/localtime: 18:05:26.724 -> 18:05:27.723 (999ms)
18:05:28:117 Poll APS confirm 243 status: 0xE1
18:05:28:118 	 drop item attr/modelid
18:05:28:118 	 drop item attr/swversion
18:05:28:119 0x84182600000D716E error APSDE-DATA.confirm: 0xE1 on task
18:05:28:119 Erase task req-id: 243, type: 19 zcl seqno: 120 send time 1, profileId: 0x0104, clusterId: 0x0006
18:05:28:120 unhandled APS-DATA.confirm id: 243 status 0xE1
18:05:28:724 enqueue event config/localtime for /config/
18:05:28:725 Wait 1s till query finished
18:05:28:725 poll node 84:18:26:00:00:0d:71:6e-03
18:05:28:726 Poll light node Stube Dose3
18:05:28:730 rule event /config/localtime: 18:05:27.723 -> 18:05:28.724 (1001ms)
18:05:28:774 read attributes of 0x84182600000D716E cluster: 0x0006: [ 
18:05:28:775 0x0000 
18:05:28:776 ]
18:05:28:776 add task 210 type 19 to 0x84182600000D716E cluster 0x0006 req.id 254
18:05:28:777 Poll APS request 254 to 0x84182600000D716E cluster: 0x0006
18:05:29:724 enqueue event config/localtime for /config/
18:05:29:724 Idle timer triggered
18:05:29:724 Force read attributes for ZHAPressure SensorNode Wohnzimmer Wetter
18:05:29:726 rule event /config/localtime: 18:05:28.724 -> 18:05:29.724 (1000ms)
18:05:29:958 Poll APS confirm 254 status: 0xE1
18:05:29:959 	 drop item attr/modelid
18:05:29:959 	 drop item attr/swversion
18:05:29:959 0x84182600000D716E error APSDE-DATA.confirm: 0xE1 on task
18:05:29:960 Erase task req-id: 254, type: 19 zcl seqno: 121 send time 1, profileId: 0x0104, clusterId: 0x0006
18:05:29:960 unhandled APS-DATA.confirm id: 254 status 0xE1
18:05:30:129 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
18:05:30:682 poll node 84:18:26:00:00:0d:71:6e-03
18:05:30:683 Poll light node Stube Dose3
18:05:30:724 enqueue event config/localtime for /config/
18:05:30:724 Wait 1s till query finished
18:05:30:733 rule event /config/localtime: 18:05:29.724 -> 18:05:30.724 (1000ms)
18:05:30:734 read attributes of 0x84182600000D716E cluster: 0x0006: [ 
18:05:30:736 0x0000 
18:05:30:737 ]
18:05:30:738 add task 219 type 19 to 0x84182600000D716E cluster 0x0006 req.id 9
18:05:30:739 Poll APS request 9 to 0x84182600000D716E cluster: 0x0006
18:05:31:723 enqueue event config/localtime for /config/
18:05:31:724 Idle timer triggered
18:05:31:725 Force read attributes for ZHATemperature SensorNode Bad Wetter
18:05:31:742 rule event /config/localtime: 18:05:30.724 -> 18:05:31.723 (999ms)
18:05:31:914 Poll APS confirm 9 status: 0xE1
18:05:31:915 	 drop item attr/modelid
18:05:31:915 	 drop item attr/swversion
18:05:31:915 0x84182600000D716E error APSDE-DATA.confirm: 0xE1 on task
18:05:31:916 Erase task req-id: 9, type: 19 zcl seqno: 122 send time 1, profileId: 0x0104, clusterId: 0x0006
18:05:31:916 unhandled APS-DATA.confirm id: 9 status 0xE1
[... 359 lines cut ...]
18:06:06:827 0x84182600000D716E error APSDE-DATA.confirm: 0xE1 on task
18:06:06:827 Erase task req-id: 192, type: 19 zcl seqno: 141 send time 1, profileId: 0x0104, clusterId: 0x0006
18:06:06:828 unhandled APS-DATA.confirm id: 192 status 0xE1
18:06:07:224 Daylight now: goldenHour2, status: 180, daylight: 0, dark: 0
18:06:07:424 poll node 84:18:26:00:00:0d:71:6e-03
18:06:07:424 Poll light node Stube Dose3
18:06:07:475 read attributes of 0x84182600000D716E cluster: 0x0006: [ 
18:06:07:475 0x0000 
18:06:07:476 ]
18:06:07:476 add task 386 type 19 to 0x84182600000D716E cluster 0x0006 req.id 202
18:06:07:477 Poll APS request 202 to 0x84182600000D716E cluster: 0x0006
18:06:07:523 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
18:06:07:724 enqueue event config/localtime for /config/
18:06:07:725 sql exec SELECT conf FROM zbconf ORDER BY rowid desc limit 1
18:06:07:726 Idle timer triggered
18:06:07:737 rule event /config/localtime: 18:06:06.724 -> 18:06:07.724 (1000ms)
18:06:08:649 0x84182600000D716E error APSDE-DATA.confirm: 0xE1 on `task`
`

> How about a backup/restore via Phoscon?
> 
> Wild guess: you did so and deconz crashed while importing the backup?

Backup/Restore was done with Phoscon
After the restore Deconz quit and was automatically restarted.

SwoopX, 2021-03-13:
That was actually an application crash and no restart. This led to your correct network settings not being loaded.

Please update to the latest beta (2.10.02), which has a fix for that. Afterwards, re-import your backup.

det-box, 2021-03-14:
Thank you,
With version 2.10.02 and import backup it works.

closed 2021-03-14 by det-box

00lex, 2021-03-16:
oh wait! 
is it correct that it is now possible to change the deconz hardware without having to manually reconnect all devices?

I upgraded to raspbee2 but a stick is meanwhile better for my setup. but im scared to pair all 60 devices again

Smanar, 2021-03-16:
It possible since the starting.
Go in phoscon, gateway, make a backup, change the hardware and make a restore.

00lex, 2021-03-16:
when I switched to raspbee2 I had problems with the devices and had to set up everything again manual. 
somewhere on git I read it does not work yet. but okay I will try it. thanks!

so that no misunderstandings arise. 
i can switch from raspbee2 to conbee2 and only need to restore a backup?

Smanar, 2021-03-17:
In therory yes. It work for Raspbee to conbee 2, haven't found guys that have tried Raspbee2 to Conbee 2, but I don't see reason for it not work. And yes I remember there was issue for Raspbee 2, but it seem they are solved now.

If someone have already try it ?