dresden-elektronik/deconz-rest-plugin #4584: Aqara wall switch not working on the first press

ahocquet, 2021-03-14:
## Describe the bug

I have simple and double Aqara wall switches. For a couple of days, I’ve noticed that I had to press two times the button in order to catch the event. After I’ve pressed the button for the second time, it works for all the next presses as well. But the problem will come back if I don’t use the wall switch for at least 5 minutes.

My Conbee II is in my living room, 3 meters away from one of my wall switch that has the problem. I also have some hue lights paired.

As you can see in the screenshots below, some of my wall switches don't have green lines.

## Screenshots
[image]

## Environment

 - Host system: Intel NuC
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: 26660700
 - deCONZ version: 2.09.03 / 12/23/2020
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## Additional context

Smanar, 2021-03-14:
Hello, nothing to see with your issue, but you have added the router after all your sensors ?

SwoopX, 2021-03-14:
For my taste, you heavily scratch on the maximum number of devices which can be directly handled by the coordinator, but probably already exceeded that (32, but presumably less).

More routers are recommendable.

ahocquet, 2021-03-14:
Thanks guys for your inputs!

It seems that my main issue is about not having routers. But I have 3 philips hue attached and on aqara main powered which are orange/yellow colored on deconz, meaning they can act as a router. Or am I wrong? 

How can a attach my wall switches to some routers instead of the conbee II coordinator ?

@Smanar : battery level is goo,d I'm at 100% and just changed the battery.

b2un0, 2021-03-15:
> How can a attach my wall switches to some routers instead of the conbee II coordinator ?

reset and pair again near the hue bulbs.

i notices similar problems and this helps me a lot

Smanar, 2021-03-15:
It s the origin of my question ^^.
Device always prefer direct connexion, but seriously 0 device using router on 30 ....
Xiaomi have a weak with mesh, they never change their parent, so can use the b2un0 tips, or just short press the button, with that, it will be fatser, you will loose nothing and make the device able to search  a new way.

ahocquet, 2021-03-15:
Thank you guys, I've re-paired some of my wall switches and the issue is now gone. I guess you were right smanar, there were too many endpoint devices connecting directly to the coordinator!

ahocquet, 2021-03-16:
I am looking at my direct connections to my coordinator right now, and it seems there are still too many devices connecting directly, even though I've repaired a lot of them. These devices are Aqara products, my Philips motion sensor connects to the hue lights routers.

Here's a screen shot:

[image]

Should I try to fix this ?

Smanar, 2021-03-16:
You still have the issue ?
You have enought routers for me, for an end device can use  them. They always prefer direct connexion, so if after a a re-inclusion you have again a direct connexion keep it, don't loose your time to force the device use a router.
When a new sensor will be not able to make adirect connexion, it will use a router itself for me.

But your mesh is realy funny, it s like you have all your sensors in the  same room than the conbee, and all bulbs in an other one ^^

ahocquet, 2021-03-18:
@Smanar, no I don't have the issue anymore. I'm just concerned regarding my mesh because even though I'm repairing some of my end devices, they still prefer direct connection most of them.

Anyway, issue seems to be gone for now!

Smanar, 2021-03-18:
>I'm repairing some of my end devices, they still prefer direct connection most of them.

Yep, same for everybody, you can only "cheat" to avoid that, not a solution, but on a normal network, I hope new sensor will take router naturally  if there is not more direct connxion possible.

closed 2021-04-16 by github-actions[bot]