dresden-elektronik/deconz-rest-plugin #4586: Devices start disconnection after few hours

chofstede, 2021-03-14:
## Describe the bug

Multiple devices loose connection after a few hours of deCONZ Operation.
After a re-start of the deconz service, all devices are working fine. Then they
loose connection. This was observed with multiple devices (IKEA and Philips)

When sending GROUP (Broadcast) Messages, they still can be toggled. 
But not directly on device level.

## Steps to reproduce the behavior

Start deCONZ. Wait ~3-4 Hours. Some devices then are not reachable anymore.

## Screenshots

[image]
Device "Stehlampe" is not reachable anymore. 

[image]
Here you can see, after re-start of deCONZ it was stable for a while. (Red) and then became unstable (grey)

## Environment

deCONZ Version 2.09.03
Firmware Version 26660700

 - Host system: Raspberry Pi 4
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.09.03
 - Device:  ConBee II
 - Do you use an USB extension cable: yes. 1m
 - Is there any other USB or serial devices connected to the host system? If so: JeeLink USB

## deCONZ Logs

**16:37:59:146 0x00178801047377C3 error APSDE-DATA.confirm: 0xE9 on task
16:37:59:147 max transmit errors for node 0x00178801047377C3, last seen by neighbors 14 s**
16:37:59:696 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
16:38:01:694 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
16:38:02:146 0x00178801047377C3 error APSDE-DATA.confirm: 0xE9 on task
16:38:02:147 max transmit errors for node 0x00178801047377C3, last seen by neighbors 17 s
16:38:02:513 [INFO] - Button 1002 - TRADFRI motion sensor, broadcast to: 0xD23E, endpoint: 0x01, cluster: ONOFF (0x0006), action: On with timed off, payload: 0058020000, zclSeq: 73
16:38:03:694 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
16:38:05:144 0x00178801047377C3 error APSDE-DATA.confirm: 0xE9 on task
16:38:05:144 max transmit errors for node 0x00178801047377C3, last seen by neighbors 20 s
16:38:05:543 [INFO] - Button 1002 - TRADFRI motion sensor, broadcast to: 0xD23E, endpoint: 0x01, cluster: ONOFF (0x0006), action: On with timed off, payload: 0058020000, zclSeq: 74
16:38:05:695 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
16:38:07:694 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
16:38:08:140 0x00178801047377C3 error APSDE-DATA.confirm: 0xE9 on task
16:38:08:141 max transmit errors for node 0x00178801047377C3, last seen by neighbors 23 s
16:38:09:695 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
16:38:11:146 0x00178801047377C3 error APSDE-DATA.confirm: 0xE9 on task
16:38:11:147 max transmit errors for node 0x00178801047377C3, last seen by neighbors 26 s
16:38:11:306 0xEE2C seems to be a zombie recv errors 10
16:38:11:307 LightNode removed 0x00178801047377c3
16:38:11:307 Node zombie state changed 0x00178801047377c3
16:38:11:696 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
16:38:11:710 Current channel 25
16:38:11:714 CTRL got nwk update id 3

easybeat, 2021-03-16:
> [image]
> Here you can see, after re-start of deCONZ it was stable for a while. (Red) and then became unstable (grey)
> 

Hi

this is a very good information. How did you manage to get this data from deconz?

I face a lot of problems with disconnected devices and would like to get alerted when one is disconnected for a longer time.

Kind regards
Beat

Mimiix, 2021-03-16:
@easybeat That's from Home-Assistant. The integration provides that info. However, it could also be this bug : https://github.com/home-assistant/core/issues/37372

chofstede, 2021-03-16:
Seems to be a actual connectivity problem. Even though the setup is all in one small Appartement and not even 7 meters between all the devices, the problem was fixed when adding an IKEA Tradfri ZigBee Repeater plug [(](https://www.ikea.com/de/de/p/tradfri-signalverstaerker-10400408/)

Since the repeater was put into the mesh, the network seems much more stable and all bulbs can be reached fine. Working for 2 
 whole days now.

Sorry for the inconvinience, I'll close that issue.

regards,

Chris

closed 2021-03-16 by chofstede